Nossiac/mtk-openwrt-feeds #3: MT7610 not working

DavideFioravanti, 2017-11-29:
I have a device with MT7628+MT7610, but only MT7628 is working.
If I try to load the mt7610.ko module I get this:
```
root@LEDE:/# insmod /lib/modules/4.4.102/mt7610.ko
[ 1190.260669] mt7610: Unknown symbol _raw_spin_lock_irqsave (err 0)
[ 1190.272815] mt7610: Unknown symbol __per_cpu_offset (err 0)
[ 1190.283921] mt7610: Unknown symbol _raw_spin_unlock_bh (err 0)
[ 1190.295606] mt7610: Unknown symbol _raw_spin_unlock_irqrestore (err 0)
[ 1190.308598] mt7610: Unknown symbol del_timer_sync (err 0)
[ 1190.319390] mt7610: Unknown symbol _raw_spin_lock_bh (err 0)
failed to insert /lib/modules/4.4.102/mt7610.ko
```

I compiled only:
Application -> uci2dat
Drivers -> mt7610
Drivers -> mt7628
Misc -> mtk-luci-plugin

(I don't know what flash, nvram and libnvram do)

Nossiac, 2017-11-29:
That may happen, because I cannot find a 7610 model around. 
I'll look into it later.

DavideFioravanti, 2017-11-29:
First of all, thank you for your hard work. :)
I have this router AUKEY WF-R7, but the board inside says WL-WN575A2.
Tell me if you need more info to investigate the problem.

Thank you again.

Nossiac, 2017-12-05:
@DavideFioravanti , I just found a rooter with 7610, the driver is working.
The error you run into was not about the driver. It seems that some basic kernel APIs were not exported. Did you made some changes to the kernel or kconfig?

DavideFioravanti, 2017-12-05:
I didn't change the source code at all, I just edited the configuration in menuconfig.
I compiled the image with these settings:
```
Target System (MediaTek Ralink MIPS)
Subtarget (MT7628 based boards)
Target Profile (Wavlink WL-WN575A3)
```
with these additions:
ramdisk image (because I can boot it without flashing)
Luci
Nano
MTK Properties (I included every option in this section)

Then I just deselected the kmod-mt76 package and it automatically removed all the other kmod in the Wireless Drivers section (kmod-cfg80211 and kmod-mac80211 too)

If you can build an initramfs (ramdisk) image for my device I can test it.

Thank you

DavideFioravanti, 2018-01-04:
Solved. It was the SMP problem https://github.com/Nossiac/mtk-openwrt-feeds/issues/8#issuecomment-354508613.
Thank you! :) 👍

closed 2018-01-04 by DavideFioravanti

hybriz, 2018-03-29:
@DavideFioravanti could you post how you got LEDE to work on that AUKEY router? which image did you use as base?

DavideFioravanti, 2018-04-11:
@hybriz I compiled using the WL-WN575A3 profile. They should differ just for the 5ghz wireless chip