Nossiac/mtk-openwrt-feeds #4: Mt7628 + lede 17.04.1 : Instant kernel panic

pparent76, 2017-12-06:
When inserting the module on lede 17.04.1 I get instant kernel panic.  I've tested with both lede-17.01.4-ramips-mt7628-mt7628-squashfs-sysupgrade.bin  and lede-17.01.4-ramips-mt7628-wl-wn575a3-squashfs-sysupgrade.bin images.

> root@LEDE:/# insmod /tmp/mt7628-linux-4.4.92.ko 
> [  211.877383] mt7628: module license 'unspecified' taints kernel.
> [  211.889138] Disabling lock debugging due to kernel taint
> [  211.947108] 
> [  211.947108] 
> [  211.947108] === pAd = c05e0000, size = 1292832 ===
> [  211.947108] 
> [  211.965573] <-- RTMPAllocTxRxRingMemory, Status=0, ErrorValue=0x
> [  211.978719] <-- RTMPAllocAdapterBlock, Status=0
> [  211.987721] RtmpChipOpsHook(492): Not support for HIF_MT yet!
> [  211.999099] mt7628_init()-->
> [  212.004799] mt7628_init(FW(8a00), HW(8a01), CHIPID(7628))
> [  212.015492] e2.bin mt7628_init(1135)::(2), pChipCap->fw_len(63888)
> [  212.027732] mt_bcn_buf_init(218): Not support for HIF_MT yet!
> [  212.039109] <--mt7628_init()
> [  212.046984] CPU 0 Unable to handle kernel paging request at virtual address 0000003c, epc == 806dd3f8, ra == 806dd3e8
> [  212.068055] Oops[#1]:
> [  212.072553] CPU: 0 PID: 1008 Comm: insmod Tainted: P                4.4.92 #0
> [  212.086679] task: 818bc000 ti: 80cb2000 task.ti: 80cb2000
> [  212.097358] $ 0   : 00000000 00000061 0000003c ffffffff
> [  212.107714] $ 4   : 80371f80 00000003 00000001 00000000
> [  212.118061] $ 8   : 00000004 c0000000 00000000 ffff0000
> [  212.128409] $12   : 00000000 00000013 00000000 00000000
> [  212.138756] $16   : 00000000 80f1c000 c071d000 80740000
> [  212.149105] $20   : 80370000 80370000 80370000 802fe334
> [  212.159454] $24   : 00000000 80042fd4                  
> [  212.169802] $28   : 80cb2000 80cb3c90 00000000 806dd3e8
> [  212.180152] Hi    : 002afae3
> [  212.185839] Lo    : c2f83c8f
> [  212.191542] epc   : 806dd3f8 0x806dd3f8 [mt7628@80600000+0x145540]
> [  212.203776] ra    : 806dd3e8 0x806dd3e8 [mt7628@80600000+0x145540]
> [  212.215999] Status: 1100e403	KERNEL EXL IE 
> [  212.224284] Cause : 00800008 (ExcCode 02)
> [  212.232210] BadVA : 0000003c
> [  212.237900] PrId  : 00019655 (MIPS 24KEc)
> [  212.245824] Modules linked in: mt7628(P+) pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt mt76x2e mt7603e ledtrig_usbport mt76 mac80211 cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables leds_gpio ohci_platform ohci_hcd ehci_platform ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
> [  212.376737] Process insmod (pid: 1008, threadinfo=80cb2000, task=818bc000, tls=77517d48)
> [  212.392746] Stack : 80f1c000 00000000 80740000 80f1c000 00000100 80f1c000 80740000 80360000
> 	  00000000 807461b0 80f27b80 80200020 00000004 8003f270 8182e840 80f22e00
> 	  c05e0000 806de760 806de5c0 806debd4 806de8f0 806de670 00000000 00000000
> 	  00000000 00000000 00000100 00000000 00000000 00000000 00000000 00000000
> 	  00000000 319fd7f0 80746000 80360000 80e55980 8000d85c 81871400 00200020
> 	  ...
> [  212.463130] Call Trace:[<807461b0>] 0x807461b0 [mt7628@80600000+0x145540]
> [  212.476600] [<80200020>] 0x80200020
> [  212.483496] [<8003f270>] 0x8003f270
> [  212.490401] [<806de760>] 0x806de760 [mt7628@80600000+0x145540]
> [  212.501946] [<806de5c0>] 0x806de5c0 [mt7628@80600000+0x145540]
> [  212.513490] [<806debd4>] 0x806debd4 [mt7628@80600000+0x145540]
> [  212.525034] [<806de8f0>] 0x806de8f0 [mt7628@80600000+0x145540]
> [  212.536580] [<806de670>] 0x806de670 [mt7628@80600000+0x145540]
> [  212.548126] [<80746000>] 0x80746000 [mt7628@80600000+0x145540]
> [  212.559671] [<8000d85c>] 0x8000d85c
> [  212.566570] [<80036284>] 0x80036284
> [  212.573465] [<800a53d4>] 0x800a53d4
> [  212.580371] [<80033c94>] 0x80033c94
> [  212.587270] [<800a51c0>] 0x800a51c0
> [  212.594174] [<80066cfc>] 0x80066cfc
> [  212.601069] [<8009d620>] 0x8009d620
> [  212.607970] [<800681ac>] 0x800681ac
> [  212.614866] [<800663a0>] 0x800663a0
> [  212.621767] [<800683c0>] 0x800683c0
> [  212.628665] [<8006845c>] 0x8006845c
> [  212.635568] [<8000686c>] 0x8000686c
> [  212.642465] [<80008c98>] 0x80008c98
> [  212.649360] 
> [  212.652294] 
> Code: 8e22023c  2403ffff  2442003c <c0440000> 7c640004  e0440000  1080fffc  00000000  8e624fd4 
> [  212.672006] ---[ end trace 4041321e5efab636 ]---
> [  212.683792] Fatal exception: panic in 5 seconds
> [  217.696385] Kernel panic - not syncing: Fatal exception
> [  217.708642] Rebooting in 3 seconds..

As I mentioned on my email We are ready to pay you if you can help with this problem. Contact me at pparent@comminter.com

Nossiac, 2017-12-07:
plz try the new built module.

pparent76, 2017-12-08:
Sadly I get exactly the same result (tested on official lede-17.01.4-ramips-mt7628-mt7628-squashfs-sysupgrade.bin )

```
root@LEDE:/# insmod /tmp/mt7628-linux-4.4.92.ko 
[   48.706971] mt7628: module license 'unspecified' taints kernel.
[   48.718724] Disabling lock debugging due to kernel taint
[   48.776316] 
[   48.776316] 
[   48.776316] === pAd = c05a9000, size = 1292832 ===
[   48.776316] 
[   48.794820] <-- RTMPAllocTxRxRingMemory, Status=0, ErrorValue=0x
[   48.807967] <-- RTMPAllocAdapterBlock, Status=0
[   48.816966] RtmpChipOpsHook(492): Not support for HIF_MT yet!
[   48.828345] mt7628_init()-->
[   48.834043] mt7628_init(FW(8a00), HW(8a01), CHIPID(7628))
[   48.844736] e2.bin mt7628_init(1135)::(2), pChipCap->fw_len(63888)
[   48.856973] mt_bcn_buf_init(218): Not support for HIF_MT yet!
[   48.868347] <--mt7628_init()
[   48.876112] CPU 0 Unable to handle kernel paging request at virtual address 0000003c, epc == 806dd3f8, ra == 806dd3e8
[   48.897207] Oops[#1]:
[   48.901706] CPU: 0 PID: 973 Comm: insmod Tainted: P                4.4.92 #0
[   48.915660] task: 819c6d48 ti: 80e72000 task.ti: 80e72000
[   48.926338] $ 0   : 00000000 00000061 0000003c ffffffff
[   48.936693] $ 4   : 80371f80 00000003 00000001 00000000
[   48.947039] $ 8   : 00000004 c0000000 00000000 ffff0000
[   48.957388] $12   : 00000000 00000013 00000000 00000000
[   48.967735] $16   : 00000000 80d40000 c06e6000 80740000
[   48.978085] $20   : 80370000 80370000 80370000 802fe334
[   48.988434] $24   : 00000000 80042fd4                  
[   48.998783] $28   : 80e72000 80e73c90 00000000 806dd3e8
[   49.009135] Hi    : 003c3c30
[   49.014823] Lo    : c89c1cc0
[   49.020528] epc   : 806dd3f8 0x806dd3f8 [mt7628@80600000+0x145540]
[   49.032762] ra    : 806dd3e8 0x806dd3e8 [mt7628@80600000+0x145540]
[   49.044986] Status: 1100a403	KERNEL EXL IE 
[   49.053272] Cause : 00800008 (ExcCode 02)
[   49.061197] BadVA : 0000003c
[   49.066886] PrId  : 00019655 (MIPS 24KEc)
[   49.074809] Modules linked in: mt7628(P+) pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt ledtrig_usbport mac80211 cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables leds_gpio ohci_platform ohci_hcd ehci_platform ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
[   49.202097] Process insmod (pid: 973, threadinfo=80e72000, task=819c6d48, tls=77413d48)
[   49.217933] Stack : 80d40000 00000000 80740000 80d40000 00000100 80d40000 80740000 80360000
	  00000000 807461b0 80e4f500 802fa080 00000004 8003f270 8182e840 80d24400
	  c05a9000 806de760 806de5c0 806debd4 806de8f0 806de670 00000000 00000000
	  00000000 00000000 00000100 00000000 00000000 00000000 00000000 00000000
	  00000000 78490a1d 80746000 80360000 80e4f080 8000d85c 81871400 800454fc
	  ...
[   49.288317] Call Trace:[<807461b0>] 0x807461b0 [mt7628@80600000+0x145540]
[   49.301786] [<8003f270>] 0x8003f270
[   49.308691] [<806de760>] 0x806de760 [mt7628@80600000+0x145540]
[   49.320236] [<806de5c0>] 0x806de5c0 [mt7628@80600000+0x145540]
[   49.331779] [<806debd4>] 0x806debd4 [mt7628@80600000+0x145540]
[   49.343323] [<806de8f0>] 0x806de8f0 [mt7628@80600000+0x145540]
[   49.354867] [<806de670>] 0x806de670 [mt7628@80600000+0x145540]
[   49.366413] [<80746000>] 0x80746000 [mt7628@80600000+0x145540]
[   49.377956] [<8000d85c>] 0x8000d85c
[   49.384853] [<800454fc>] 0x800454fc
[   49.391747] [<80036284>] 0x80036284
[   49.398648] [<800341a4>] 0x800341a4
[   49.405548] [<80033c94>] 0x80033c94
[   49.412462] [<80066cfc>] 0x80066cfc
[   49.419358] [<8009d620>] 0x8009d620
[   49.426258] [<800681ac>] 0x800681ac
[   49.433154] [<800663a0>] 0x800663a0
[   49.440054] [<800683c0>] 0x800683c0
[   49.446953] [<8006845c>] 0x8006845c
[   49.453856] [<8000686c>] 0x8000686c
[   49.460753] [<80008c98>] 0x80008c98
[   49.467649] 
[   49.470583] 
Code: 8e22023c  2403ffff  2442003c <c0440000> 7c640004  e0440000  1080fffc  00000000  8e624fd4 
[   49.490365] ---[ end trace 2661ba975bf54475 ]---
[   49.502087] Fatal exception: panic in 5 seconds
```

Of course if you want/need mt7628 devices to test we can send you some for free.

Thanks a lot for everything.

Nossiac, 2017-12-13:
I replied your email.
I have tried the module on 2 different sets, both were working properly.

pparent76, 2017-12-13:
Thanks a lot for your answer.
Sorry your mail went to spam.

**Can you explain how you make it work, on official lede realese?**

**What I did:**
*Download official SDK for 17.01.4, and  build mt7628_1_mipsel_24kc.ipk as described in your repo.
*Download official lede-17.01.4-ramips-mt7628-mt7628-squashfs-sysupgrade.bin and flash it to a mt7628nn device (ZBT zbt-3326)
*Remove /lib/modules/4.4.92/mt76.ko /lib/modules/4.4.92/mt7603e.ko  /lib/modules/4.4.92/mt76x2e.ko from device (open-source driver) and reboot.
*After reboot a through insmod mt7628.ko => instant kernel panic

Here is the md5sum:
```
root@LEDE:/# md5sum  /lib/modules/4.4.92/mt7628.ko 
 e756518f05cba28b0d9d4c3c7f624d30  /lib/modules/4.4.92/mt7628.ko
```

Here is the crash Once again:

```
root@LEDE:/# insmod mt7628.ko
[  133.853144] mt7628: module license 'unspecified' taints kernel.
[  133.864898] Disabling lock debugging due to kernel taint
[  133.923106] 
[  133.923106] 
[  133.923106] === pAd = c0a8e000, size = 1292832 ===
[  133.923106] 
[  133.941578] <-- RTMPAllocTxRxRingMemory, Status=0, ErrorValue=0x
[  133.954719] <-- RTMPAllocAdapterBlock, Status=0
[  133.963722] RtmpChipOpsHook(492): Not support for HIF_MT yet!
[  133.975098] mt7628_init()-->
[  133.980804] mt7628_init(FW(8a00), HW(8a01), CHIPID(7628))
[  133.991496] e2.bin mt7628_init(1135)::(2), pChipCap->fw_len(63888)
[  134.003735] mt_bcn_buf_init(218): Not support for HIF_MT yet!
[  134.015114] <--mt7628_init()
[  134.023068] CPU 0 Unable to handle kernel paging request at virtual address 0000003c, epc == 806dd3f8, ra == 806dd3e8
[  134.044161] Oops[#1]:
[  134.048660] CPU: 0 PID: 907 Comm: insmod Tainted: P                4.4.92 #0
[  134.062613] task: 819c4f18 ti: 80e2a000 task.ti: 80e2a000
[  134.073291] $ 0   : 00000000 00000061 0000003c ffffffff
[  134.083647] $ 4   : 80371f80 00000003 00000001 00000000
[  134.093995] $ 8   : 00000004 c0000000 00000000 ffff0000
[  134.104341] $12   : 00000000 ffffff80 00000000 00000000
[  134.114688] $16   : 00000000 80ca5800 c0bcb000 80740000
[  134.125035] $20   : 80370000 80370000 80370000 802fe334
[  134.135383] $24   : 00000000 80042fd4                  
[  134.145732] $28   : 80e2a000 80e2bc90 00000000 806dd3e8
[  134.156083] Hi    : 0068ae54
[  134.161769] Lo    : 5d08be8e
[  134.167474] epc   : 806dd3f8 0x806dd3f8 [mt7628@80600000+0x145540]
[  134.179707] ra    : 806dd3e8 0x806dd3e8 [mt7628@80600000+0x145540]
[  134.191932] Status: 1100a403	KERNEL EXL IE 
[  134.200215] Cause : 00800008 (ExcCode 02)
[  134.208141] BadVA : 0000003c
[  134.213831] PrId  : 00019655 (MIPS 24KEc)
[  134.221755] Modules linked in: mt7628(P+) pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt ledtrig_usbport mac80211 cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables leds_gpio ohci_platform ohci_hcd ehci_platform ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
[  134.349044] Process insmod (pid: 907, threadinfo=80e2a000, task=819c4f18, tls=77efbd48)
[  134.364881] Stack : 80ca5800 00000000 80740000 80ca5800 00000100 80ca5800 80740000 80360000
	  00000000 807461b0 80c16980 802fa080 00000004 8003f270 8182e840 80e2ce00
	  c0a8e000 806de760 806de5c0 806debd4 806de8f0 806de670 00000000 00000000
	  00000000 00000000 00000100 00000000 00000000 00000000 00000000 00000000
	  00000000 ea53aa38 80746000 80360000 80c16a80 8000d85c 81871400 800454fc
	  ...
[  134.435264] Call Trace:[<807461b0>] 0x807461b0 [mt7628@80600000+0x145540]
[  134.448731] [<8003f270>] 0x8003f270
[  134.455638] [<806de760>] 0x806de760 [mt7628@80600000+0x145540]
[  134.467181] [<806de5c0>] 0x806de5c0 [mt7628@80600000+0x145540]
[  134.478725] [<806debd4>] 0x806debd4 [mt7628@80600000+0x145540]
[  134.490270] [<806de8f0>] 0x806de8f0 [mt7628@80600000+0x145540]
[  134.501815] [<806de670>] 0x806de670 [mt7628@80600000+0x145540]
[  134.513360] [<80746000>] 0x80746000 [mt7628@80600000+0x145540]
[  134.524901] [<8000d85c>] 0x8000d85c
[  134.531799] [<800454fc>] 0x800454fc
[  134.538695] [<80036284>] 0x80036284
[  134.545594] [<800341a4>] 0x800341a4
[  134.552494] [<80033c94>] 0x80033c94
[  134.559410] [<80066cfc>] 0x80066cfc
[  134.566305] [<8009d620>] 0x8009d620
[  134.573205] [<800681ac>] 0x800681ac
[  134.580101] [<800663a0>] 0x800663a0
[  134.587003] [<800683c0>] 0x800683c0
[  134.593901] [<8006845c>] 0x8006845c
[  134.600803] [<8000686c>] 0x8000686c
[  134.607700] [<80008c98>] 0x80008c98
[  134.614596] 
[  134.617530] 
Code: 8e22023c  2403ffff  2442003c <c0440000> 7c640004  e0440000  1080fffc  00000000  8e624fd4 
[  134.637234] ---[ end trace 479d863e2b989f49 ]---
[  134.648937] Fatal exception: panic in 5 seconds
```
Thanks a lot.

Nossiac, 2017-12-30:
an updated version of 7628.ko has been committed, you can rebuilt with this repo.
note that lede-17.01 has upgraded its kernel to 4.4.108, you'd better update your code first:
```
git checkout lede-17.01
git pull origin lede-17.01
```

http://nossiac.com/download/mtk-wifi-ko/mt7628-for-mt7628-linux-4.4.108.ko

pparent76, 2017-12-30:
Thanks a lot for your efforts!

But why do you make build for 17.01-git that constantly changes kernel, rather than stable realeses (e.g 17.01.4) that remains accessible across time? It makes your repo difficult to use on the long run.

Pierre.

Nossiac, 2018-01-02:
yes, you are right. i'm setting up a autobuild server to build drivers for specific tags/branches.

pparent76, 2018-01-02:
Thanks a lot!

pparent76, 2018-01-02:
This bug seems solved with kernel 4.4.108.

closed 2018-01-02 by pparent76