Nossiac/mtk-openwrt-feeds #5: 求最新的内核4.9.67驱动

zhaibin18, 2017-12-11:
从github上拖下来的lede的内核已经是4.9.67了。

Nossiac, 2017-12-13:
Sorry, I'm a little busy these days.
Will come back on this project next week, I think.

MiltonLai, 2017-12-22:
The kernel is 4.9.70 now, guys...

Nossiac, 2017-12-26:
It was planned to follow the stable release only, that is lede-17.01.
Since the ko is too big for a git rpo, I'm gonna refactor it a little bit. hopefully , can be done this week.

recyard, 2017-12-27:
Hope there are 4.9 kernel modules someday, since my router model (Netgear R6220) only has support in LEDE master(development) branch with 4.9 kernel.

zhaibin18, 2017-12-30:
I used a Phicomm K2P router only works with LEDE master branch with 4.9 kernel，Hope there is 4.9 kernel support, Many thanks!

chd1998, 2018-01-02:
Waiting for the new support of 4.9.*, Thanx a lot for your effort!

chd1998, 2018-01-02:
Can anybody help me with these errors?  Thanx in advance...
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-autoreboot:
 * 	libuci-lua (>= 2018-01-01) * 
 * opkg_install_cmd: Cannot install package luci-app-autoreboot.
package/Makefile:64: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/chendong/lede/lede'
package/Makefile:106: recipe for target '/home/chendong/lede/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install' failed
make[1]: *** [/home/chendong/lede/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/chendong/lede/lede'
/home/chendong/lede/lede/include/toplevel.mk:207: recipe for target 'world' failed
make: *** [world] Error 2

Nossiac, 2018-01-02:
some of the drivers were developed with linux 2.6, they are too old to build with kernel 4.9, a lot of code needs to be rewritten. also it seems lede-master adopted a more strict compiling policy, which produced more errors than lede-17.01. that is to say, it will take longer than I expected, hope that won't fail this project, 😆 😆

MaFeSa, 2018-01-02:
> hope that won't fail this project
😱
We hope not 🙏

Vonger, 2018-01-15:
Hi, I find your github on google, it save me a lot of time do the script👍 
I success compile it for Linux 4.9, just a lot of tricky. 
please update your ko to this mt7628.ko: 
md5: 31fd66c1606abc14c0fdf6eb941f4807
Feel free to email me if any question :)

Nossiac, 2018-01-16:
@Vonger thanks for your work! what's the exact kernel version?
I didn't get my hands on 4.9 support for the last 2 weeks. is it possible to send me those patches you made for 4.9? That way the automation server will build and post the ko with the latest kernel everyday.

Vonger, 2018-01-17:
It will work for all Linux 4.9 version, actually it is not necessary to make for every linux subversion unless its interface changed. Currently my patches have to manually run, something wrong with the Makefile, please send me your patch for 4.4 so I can combine them and send back to you.

barhom, 2018-02-09:
Seems LEDE is preparing to move to kernel 4.14 soon. Just a heads up before a lot of work is done on 4.9.

Nossiac, 2018-02-09:
Yes, I read the maillist from time to time.

My plan is to make these modules stable in kernel 4.4 first. After posting this repo, I got a lot of feedback from github as well as the community. There are still a lot of work pending there. I schedule the tasks by some sort of priority.

For newer kernel, I'm expecting the new official release, like openwrt-18.0X. There would be serveral new modules in the mainstream kernel which is contributted by MediaTek. Hopefully they are going to be available in kernel 4.14. That's awesome.

Vonger, 2018-02-09:
I will fork into my github and make it support 4.14 later :)

Nossiac, 2018-02-09:
@Vonger Good! That would be nice if you can share the important fixes you made to the source.

Nossiac, 2018-02-27:
I'm testing the driver on 4.9.x, latest module will be uploaded sooner this week.

closed 2018-02-27 by Nossiac

BGFWDev, 2018-02-27:
It would be great if MediaTek could skip 4.9 and go to the 4.14 release.  It has significant advantages. 

Thanks,

Chris

Nossiac, 2018-02-27:
already built the driver with latest openwrt trunk (linux-4.9.70), but not verified yet (short of hand).
If anyone want to have a try, you can find the ko here: [http://nossiac.com/download/mtk-wifi-ko/](http://nossiac.com/download/mtk-wifi-ko/)

BGFWDev, 2018-02-27:
Any plans for MT7615 and MT7612 on MT7623?

-Chris