Nossiac/mtk-openwrt-feeds #6: auto reboot when insmod ko files

reusu, 2017-12-14:
LEDE 17.01.4

build mt7628_1_mipsel_24kc.ipk
remove kmod-mt76
insmod /lib/modules/4.4.92/mt7628.ko
then ssh has no output and auto reboot

reusu, 2017-12-14:
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.020953] mt7628: module license 'unspecified' taints kernel.
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.026975] Disabling lock debugging due to kernel taint
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.089905] 
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.089905] 
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.089905] === pAd = c0c07000, size = 1292832 ===
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.089905] 
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.099469] <-- RTMPAllocTxRxRingMemory, Status=0, ErrorValue=0x
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.106795] <-- RTMPAllocAdapterBlock, Status=0
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.111438] RtmpChipOpsHook(492): Not support for HIF_MT yet!
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.117263] mt7628_init()-->
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.120211] mt7628_init(FW(8a00), HW(8a01), CHIPID(7628))
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.125685] e2.bin mt7628_init(1135)::(2), pChipCap->fw_len(63888)
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.131961] mt_bcn_buf_init(218): Not support for HIF_MT yet!
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.137784] <--mt7628_init()
Thu Dec 14 16:37:43 2017 kern.alert kernel: [  279.196967] CPU 0 Unable to handle kernel paging request at virtual address 0000003c, epc == 860dd3f8, ra == 860dd3e8
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.207807] Oops[#1]:
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.210122] CPU: 0 PID: 1843 Comm: insmod Tainted: P                4.4.92 #0
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.217359] task: 86910a10 ti: 868cc000 task.ti: 868cc000
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.222829] $ 0   : 00000000 00000061 0000003c ffffffff
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.228145] $ 4   : 80371f80 00000003 00000001 00000000
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.233456] $ 8   : 00000004 c0000000 00000000 ffff0000
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.238768] $12   : 00000000 00000013 00000000 00000000
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.244080] $16   : 00000000 8686d000 c0d44000 86140000
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.249389] $20   : 80370000 80370000 80370000 802fe31c
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.254695] $24   : 00000000 80042fd4                  
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.260003] $28   : 868cc000 868cdc90 00000000 860dd3e8
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.265317] Hi    : 0029f680
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.268233] Lo    : 724fcc3a
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.271165] epc   : 860dd3f8 0x860dd3f8 [mt7628@86000000+0x145540]
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.277434] ra    : 860dd3e8 0x860dd3e8 [mt7628@86000000+0x145540]
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.283695] Status: 1100a403	KERNEL EXL IE 
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.287950] Cause : 00800008 (ExcCode 02)
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.292010] BadVA : 0000003c
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.294929] PrId  : 00019655 (MIPS 24KEc)
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.298989] Modules linked in: mt7628(P+) pppoe ppp_async iptable_nat ip6table_nat pppox ppp_generic nf_nat_ipv6 nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xtThu Dec 14 16:37:43 2017 kern.warn kernel: [  279.440605] Process insmod (pid: 1843, threadinfo=868cc000, task=86910a10, tls=77ac1d48)
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.448802] Stack : 8686d000 00000000 86140000 8686d000 00000100 8686d000 86140000 80360000
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.448802] 	  00000000 861ff1b0 868f3300 802fa068 00000004 8003f270 87c2de30 872ce800
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.448802] 	  c0c07000 860de760 860de5c0 860debd4 860de8f0 860de670 00000000 00000000
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.448802] 	  00000000 00000000 00000100 00000000 00000000 00000000 00000000 00000000
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.448802] 	  00000000 b73d3d89 861ff000 80360000 868f3780 8000d85c 87c7b400 800454fc
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.448802] 	  ...
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.484915] Call Trace:[<861ff1b0>] 0x861ff1b0 [mt7628@86000000+0x145540]
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.491860] [<8003f270>] 0x8003f270
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.495423] [<860de760>] 0x860de760 [mt7628@86000000+0x145540]
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.501341] [<860de5c0>] 0x860de5c0 [mt7628@86000000+0x145540]
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.507256] [<860debd4>] 0x860debd4 [mt7628@86000000+0x145540]
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.513173] [<860de8f0>] 0x860de8f0 [mt7628@86000000+0x145540]
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.519091] [<860de670>] 0x860de670 [mt7628@86000000+0x145540]
[... cut ...]
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.569941] [<8000686c>] 0x8000686c
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.573479] [<80008c98>] 0x80008c98
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.577016] 
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.578524] 
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.578524] Code: 8e22023c  2403ffff  2442003c <c0440000> 7c640004  e0440000  1080fffc  00000000  8e624fd4 
Thu Dec 14 16:37:43 2017 kern.warn kernel: [  279.588654] ---[ end trace 316f63844c5a3197 ]---

reusu, 2017-12-14:
CPU | MTK 7628NN @580Mhz
-- | --
Memory/Storage | DDR2 128MB/ FLASH 16MB

pparent76, 2017-12-18:
same as me!

Nossiac, 2017-12-30:
http://nossiac.com/download/mtk-wifi-ko/mt7628-for-mt7628-linux-4.4.108.ko
plz note that the kernel for lede-17.01 is 4.4.108 now.

closed 2018-01-12 by Nossiac