Nossiac/mtk-openwrt-feeds #7: Need for mt7603 and mt7612 drivers

MiltonLai, 2017-12-22:
Hi Nossiac, thanks for these drivers! 
As mt7621 becomes more popular, there are quite a few boards are using mt7603 and mt7612. Do you have a schedule to finish the drivers for 7603 and 7612?

Nossiac, 2017-12-26:
will be ready in a couple of days.

Nossiac, 2017-12-27:
i have uploaded the ko for 7603 and 7612, try clone the repo again.

MiltonLai, 2017-12-28:
@Nossiac Sorry I didn't figure out how to use it... I built the new image and run it, when I click the "Enable" button under MTK->WiFi Configuration, the popup shows "Please restart manually to create this interface", but after I restart the device nothing happened.

Another question: In "make menuconfig", if the drivers under MTK Properties->Drivers are selected, should I unselect the kmod-mt76* drivers under Kernel Modules->Wireless Drivers?

MiltonLai, 2017-12-28:
Hmm... I rebuilt the image without kmod-mt76* ( only kmod-cfg80211 and kmod-mac80211 are selected). It seems works this time.
The Wireless menu is gone under Network, 
Under MTK->WiFi Configuration, both mt7612 and mt7603 are enabled after click the Reload button.
Now the client device can see and connect to default wifi-mt7612, but wifi-mt7603 is still missing.
The Mode field is blank, is it the cause?
[image]
[image]
In the Raw Configuration, WirelessMode=9

Nossiac, 2018-01-02:
> wifi-mt7603 is still missing.
mt7603 did not broadcasting its ssid? What's the output of "iwconfig"?
I'll look into the UI issue later.

MiltonLai, 2018-01-03:
The system log:
```
Wed Jan  3 09:11:09 2018 daemon.info odhcpd[673]: Using a RA lifetime of 0 seconds on br-lan
Wed Jan  3 09:11:44 2018 user.debug : search_dev_and_profile_orig: mt7612=/etc/wireless/mt7612/mt7612.dat
Wed Jan  3 09:11:44 2018 user.debug : search_dev_and_profile_orig: mt7603=/etc/wireless/mt7603/mt7603.dat
Wed Jan  3 09:11:44 2018 user.debug : debug
Wed Jan  3 09:11:44 2018 user.debug : load /etc/wireless/mt7612/mt7612.dat
Wed Jan  3 09:11:44 2018 user.debug : loading profile/etc/wireless/mt7612/mt7612.dat
Wed Jan  3 09:11:44 2018 user.warn : skip comment line #The word of "Default" must not be removed
Wed Jan  3 09:11:44 2018 user.warn : skip line without '=' Default
Wed Jan  3 09:11:44 2018 user.warn : skip repeated keyGreenAP=0
Wed Jan  3 09:11:44 2018 user.warn : skip line without '=' 
Wed Jan  3 09:11:44 2018 user.debug : setup_vifs
Wed Jan  3 09:11:44 2018 user.debug : debug
Wed Jan  3 09:11:44 2018 user.debug : load /etc/wireless/mt7603/mt7603.dat
Wed Jan  3 09:11:44 2018 user.debug : loading profile/etc/wireless/mt7603/mt7603.dat
Wed Jan  3 09:11:44 2018 user.warn : skip comment line # Generated by mtkwifi.lua
Wed Jan  3 09:11:44 2018 user.warn : skip line without '=' Default
Wed Jan  3 09:11:44 2018 user.warn : skip line without '=' 
Wed Jan  3 09:11:44 2018 user.debug : setup_vifs
Wed Jan  3 09:11:44 2018 user.debug : setup_vifs
Wed Jan  3 09:11:59 2018 user.info : call mt7612_reload(mt7612)
Wed Jan  3 09:11:59 2018 user.debug : mt7612_reload called!
Wed Jan  3 09:11:59 2018 user.debug : mt7612_down called!
Wed Jan  3 09:11:59 2018 daemon.err rmmod: module is not loaded
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.359375] register rt2860
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.369558] 
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.369558] 
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.369558] === pAd = c0701000, size = 2104664 ===
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.369558] 
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.379365] <-- RTMPAllocTxRxRingMemory, Status=0
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.384878] <-- RTMPAllocAdapterBlock, Status=0
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.389392] pAd->CSRBaseAddress =0xc0600000, csr_addr=0xc0600000!
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.395486] device_id =0x7612
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.398449] ==>rlt_wlan_chip_onoff(): OnOff:1, Reset= 1, pAd->WlanFunCtrl:0x0, Reg-WlanFunCtrl=0x20a
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.408087] chip_id1=, chip_id2=, pAd->MACVersion=0x76623000
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.413738] chip_id1=0x0, chip_id2=0x0, pAd->MACVersion=0x76623000
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.419887] RtmpChipOpsEepromHook::e2p_type=0, inf_Type=5
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.425278] RtmpEepromGetDefault::e2p_dafault=1
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.429788] NVM is EFUSE mode
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.432759] get_dev_name_prefix(): dev_idx = 1, dev_name_prefix=rai
Wed Jan  3 09:12:00 2018 user.debug : mt7612_up called!
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.446022] build time = 
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.448674] 20141115060606a
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.451646] rom patch for E3 IC
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.454796] 
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.456318] platform = 
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.458793] ALPS
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.460681] hw/sw version = 
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.463705] ??
Wed Jan  3 09:12:00 2018 kern.warn kernel: [ 1632.465635] patch version = 
Wed Jan  3 09:12:01 2018 daemon.notice netifd: Network device 'rai0' link is up
Wed Jan  3 09:12:01 2018 user.debug : search_dev_and_profile_orig: mt7612=/etc/wireless/mt7612/mt7612.dat
Wed Jan  3 09:12:01 2018 user.debug : search_dev_and_profile_orig: mt7603=/etc/wireless/mt7603/mt7603.dat
Wed Jan  3 09:12:01 2018 user.debug : debug
Wed Jan  3 09:12:01 2018 user.debug : load /etc/wireless/mt7612/mt7612.dat
Wed Jan  3 09:12:01 2018 user.debug : loading profile/etc/wireless/mt7612/mt7612.dat
Wed Jan  3 09:12:01 2018 user.warn : skip comment line #The word of "Default" must not be removed
Wed Jan  3 09:12:01 2018 user.warn : skip line without '=' Default
Wed Jan  3 09:12:01 2018 user.warn : skip repeated keyGreenAP=0
Wed Jan  3 09:12:01 2018 user.warn : skip line without '=' 
Wed Jan  3 09:12:01 2018 user.debug : setup_vifs
Wed Jan  3 09:12:01 2018 user.debug : debug
Wed Jan  3 09:12:01 2018 user.debug : load /etc/wireless/mt7603/mt7603.dat
Wed Jan  3 09:12:01 2018 user.debug : loading profile/etc/wireless/mt7603/mt7603.dat
Wed Jan  3 09:12:01 2018 user.warn : skip comment line # Generated by mtkwifi.lua
Wed Jan  3 09:12:01 2018 user.warn : skip line without '=' Default
Wed Jan  3 09:12:01 2018 user.warn : skip line without '=' 
Wed Jan  3 09:12:01 2018 user.debug : setup_vifs
Wed Jan  3 09:12:01 2018 user.debug : setup_vifs
[... 86 lines cut ...]
Wed Jan  3 09:12:12 2018 user.warn : skip line without '=' Default
Wed Jan  3 09:12:12 2018 user.warn : skip line without '=' 
Wed Jan  3 09:12:12 2018 user.debug : setup_vifs
Wed Jan  3 09:12:12 2018 daemon.err uhttpd[735]: sh: iwconfig: not found
[... cut ...]
Wed Jan  3 09:12:12 2018 user.debug : loading profile/etc/wireless/mt7603/mt7603.dat
Wed Jan  3 09:12:12 2018 user.warn : skip comment line # Generated by mtkwifi.lua
Wed Jan  3 09:12:12 2018 user.warn : skip line without '=' Default
Wed Jan  3 09:12:12 2018 user.warn : skip line without '=' 
Wed Jan  3 09:12:12 2018 user.debug : setup_vifs
Wed Jan  3 09:12:12 2018 daemon.err uhttpd[735]: sh: iwconfig: not found
```
It seems iwconfig was not included in the default image. I checked the items in "make menuconfig" and found nothing about iwconfig

Nossiac, 2018-01-04:
@MiltonLai 
After @DavideFioravanti 's commit, You'll be able to configure 7603's wireless mode properly.

laincat, 2018-01-04:
Need for MT7602?

http://ftp.mqmaker.com/WiTi/Docs/Hardware/
http://ftp.mqmaker.com/WiTi/Docs/Hardware/MT7602E.pdf
http://ftp.mqmaker.com/WiTi/Docs/Hardware/MT7612E_Data_Sheet.pdf

Nossiac, 2018-01-04:
@MiltonLai I do find a similar issue with current 7603 driver today.... hmm... interesting...

HeadLessHUN, 2018-01-18:
Hi again!

I've successfully build the image, flashed. i can load both 7612.ko,7603.ko. And i can enable the interfaces. in iwconfig it showed up al the four interface apcli0,ra0,rai0,apclii0 and has informations. but the client cannot see the wireless network...

For the record i should enable all of the tools for it to work? I didn't check the nvram tools just the driver, uci2dat,luci pages

`#
# MTK Properties
#

#
# Applications
#
# CONFIG_PACKAGE_flash is not set
# CONFIG_PACKAGE_reg is not set
CONFIG_PACKAGE_uci2dat=y

#
# Drivers
#
# CONFIG_PACKAGE_mt7602 is not set
CONFIG_PACKAGE_mt7603=y
# CONFIG_PACKAGE_mt7610 is not set
CONFIG_PACKAGE_mt7612=y
# CONFIG_PACKAGE_mt7615 is not set
# CONFIG_PACKAGE_mt7620 is not set
# CONFIG_PACKAGE_mt7628 is not set
CONFIG_PACKAGE_mt76x2=y

#
# Libraries
#
# CONFIG_PACKAGE_libnvram is not set

#
# Misc
#
CONFIG_PACKAGE_mtk-luci-plugin=y
CONFIG_LUCI_APP_MTKWIFI=y
CONFIG_LUCI_APP_WEBCONSOLE=y`

MaFeSa, 2018-02-06:
>  i can load both 7612.ko,7603.ko. And i can enable the interfaces. in iwconfig it showed up al the four interface apcli0,ra0,rai0,apclii0 and has informations. but the client cannot see the wireless network...

Same as for me...
Where am I doing wrong?

Nossiac, 2018-02-10:
Fixed an issue that may cause 7603 tx failure.
An updated version of mt7603.ko has been posted to : http://nossiac.com/download/mtk-wifi-ko/

MaFeSa, 2018-02-10:
With LuCI I have always "Please restart manually to create this interface."

lfiebach, 2018-02-10:
loading 7612 causes :

```
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.792900] register rt2860
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.803069]
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.803069]
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.803069] === pAd = c0781000, size = 2104664 ===
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.803069]
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.812854] <-- RTMPAllocTxRxRingMemory, Status=0
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.818349] <-- RTMPAllocAdapterBlock, Status=0
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.822903] pAd->CSRBaseAddress =0xc0680000, csr_addr=0xc0680000!
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.829111] device_id =0x7662
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.832133] ==>rlt_wlan_chip_onoff(): OnOff:1, Reset= 1, pAd->WlanFunCtrl:0x0, Reg-WlanFunCtrl=0x20a
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.841896] chip_id1=, chip_id2=, pAd->MACVersion=0x76623000
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.847537] chip_id1=0x0, chip_id2=0x0, pAd->MACVersion=0x76623000
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.853714] RtmpChipOpsEepromHook::e2p_type=0, inf_Type=5
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.859287] RtmpEepromGetDefault::e2p_dafault=1
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.863894] NVM is EFUSE mode
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.866881] get_dev_name_prefix(): dev_idx = 1, dev_name_prefix=rai
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.879972] build time =
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.882644] 20141115060606a
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.885485] rom patch for E3 IC
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.888625]
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.890176] platform =
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.892904] ALPS
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.894931] hw/sw version =
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.897830] ��
Sat Feb 10 16:01:09 2018 kern.warn kernel: [  723.899817] patch version =

```

lfiebach, 2018-02-10:
Shown error happens if i bind mt7612 to an interface !

Nossiac, 2018-02-10:
@MaFeSa that was true, it could be more friendly.   Clicking the "reload" button or execute "mtkwifi restart" may fix it.

MaFeSa, 2018-02-10:
I have tried many times without success, it worked only one time and after a reboot there is again the problem. 
Thanks

Nossiac, 2018-02-11:
@MaFeSa a little confusing for me. 
Are you referring to "wireless not detected" issue, or the new comment from you, "enable button does not work"?
which chip do u use, 7603 or 7612?

MaFeSa, 2018-02-11:
@Nossiac sorry I was wrong, after reading this I understood
"If you want the driver to run automatically, the easiest hack would be overwriting /sbin/wifi with /sbin/mtkwifi from mtk-luci-plugin. If you are a pro hacker, you can write your own rcS script."

I have the same error of @lfiebach when do "modprobe mt7612.ko"

Device Xiaomi 3G with 4.4.112 kernel

Thanks

lfiebach, 2018-02-11:
With mt7603 speed is below 1mbit.
OpenWrt Drivers > 20mbit.

mt7612 result in described earlier error and freezes system.

i also did download the last driver versions but results into the same error.

P.S.: feed does not use latest version of the binarys ?

regards

Nossiac, 2018-02-12:
This thread has gone off the track. After going through the thread, I got 2 issues:
1. 7603 performance is too poor.
2. 7612 causes system freeze.
I'll look into them after Chinese New Year.

lfiebach, 2018-02-12:
thx, should i open a fresh ticket for this ?

slthomason, 2018-02-15:
On the WE1326 (same as WG3526)
I just built the latest and I get:

root@OpenWrt:~# mtkwifi restart
module is not loaded
failed to mmap /lib/modules/4.9.77/mt7603.ko
failed to mmap /lib/modules/4.9.77/mt7612.ko
module is not loaded
failed to mmap /lib/modules/4.9.77/mt7603.ko
failed to mmap /lib/modules/4.9.77/mt7612.ko

I show that I have these packages:
mt7603 - 1
mt76x2 - 1

What other packages do I need?

MaFeSa, 2018-02-15:
@slthomason 
For now only kernel 4.4.xxx (Lede 17.01 branch) works.

slthomason, 2018-02-15:
Does that include a backport for the kracks fixed?

Nossiac, 2018-02-23:
@slthomason  @Lu-Fi 

Just back to work after the Chinese New Year.
As I said, this issue has go off the track too far, I'm going to close this.

For 7612 system freezing problem, I still cannot reproduce it with my sets. Please open a new issue, and I need your HW info and console logs.

Also, there is a short trouble-shooting guide that may help. [https://github.com/Nossiac/mtk-openwrt-feeds/blob/master/about-mtk-wifi.md#troubleshooting](https://github.com/Nossiac/mtk-openwrt-feeds/blob/master/about-mtk-wifi.md#troubleshooting)

closed 2018-02-23 by Nossiac

slthomason, 2018-02-23:
> about-mtk-wifi.md#troubleshooting

Ok - I am building and testing today.

Just to clarify - does this include the 17.01.4 branch?

Spencer

Nossiac, 2018-02-23:
@slthomason yes. BTW, 17.01.4 is a tag, not  a branch.

MaFeSa, 2018-02-28:
@Nossiac 
modprobe mt7612.ko causes system freeze also with latest commit.

Thanks