Nossiac/mtk-openwrt-feeds #8: Kernel Panic on device config

alangregory, 2017-12-27:
Testing with TpLink WR840n-v4 (https://wikidevi.com/wiki/TP-LINK_TL-WR840N_v4).
Modules built with Lede stable:
`Linux ap-dev 4.4.102 #0 Sun Nov 26 14:10:36 2017 mips GNU/Linux`
Module loaded:
```
root@ap-dev:~# lsmod | grep mt76
cfg80211              233784  2 mt76,mac80211
mac80211              414818  1 mt76
mt76                   16335  0 
mt7628               1398711  0 
```
Interface shows on ifconfig:
```
root@ap-dev:~# ifconfig -a ra0
ra0       Link encap:Ethernet  HWaddr 00:00:00:00:00:00  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:6 
```
But a teste with "ifconfig ra0 up" produces a kernel panic. Does it need some specific setup before bringing it up?
```
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.521345] TX_BCN DESC a297e000 size = 320
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.525635] RX[0] DESC a32a1000 size = 1024
Tue Dec 26 23:15:38 2017 kern.alert kernel: [ 3609.530005] CPU 0 Unable to handle kernel paging request at virtual address 000010c0, epc == 8002018c, ra == 828e170c
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.540778] Oops[#1]:
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.543095] CPU: 0 PID: 1030 Comm: ifconfig Tainted: P                4.4.102 #0
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.550596] task: 83919420 ti: 83324000 task.ti: 83324000
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.556066] $ 0   : 00000000 00000001 000010c0 ffffffe0
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.561387] $ 4   : 00001fa0 00000f00 00000f00 00000002
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.566700] $ 8   : 00000020 80008c90 802f0000 30316132
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.572012] $12   : 00000000 ffffff80 00000000 00000000
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.577324] $16   : 82000020 000000c0 803b0000 c0392334
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.582635] $20   : 00000000 00000000 00000000 c0392334
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.587946] $24   : 00000002 800200e0                  
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.593259] $28   : 83324000 83325be8 60000000 828e170c
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.598575] Hi    : 00000000
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.601494] Lo    : ec4e4000
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.604441] epc   : 8002018c r4k_dma_cache_inv+0xac/0xc8
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.610266] ra    : 828e170c linux_pci_map_single+0x88/0xb4 [mt7628]
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.616708] Status: 1100e403	KERNEL EXL IE 
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.620962] Cause : 00800008 (ExcCode 02)
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.625025] BadVA : 000010c0
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.627945] PrId  : 00019655 (MIPS 24KEc)
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.632007] Modules linked in: iptable_mangle iptable_filter ipt_REJECT ip_tables xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG x_tables nf_reject_ipv4 nf_log_ipv4 nf_log_common mt7628(P) mt76 mac80211 cfg80211 compat tun leds_gpio gpio_button_hotplug
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.657961] Process ifconfig (pid: 1030, threadinfo=83324000, task=83919420, tls=772f5d48)
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.666334] Stack : 00000001 00000000 00000000 801ee5f0 60000000 828e170c 00000000 c0392334
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.666334] 	  c0391000 00000f00 838f8540 00000f00 c0392338 828e17bc 8291641c 83325c54
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.666334] 	  00000001 80048a30 0000ff01 8007ede8 c0391000 c0392334 a32a1000 c0391064
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.666334] 	  00000f00 8289d66c 8327b3c8 00000000 a32a1000 00000400 c0392334 c0392338
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.666334] 	  a32a1400 00000000 0000000f 8291641c 82916370 82916438 82940000 c0391000
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.666334] 	  ...
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.702463] Call Trace:
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.704961] [<8002018c>] r4k_dma_cache_inv+0xac/0xc8
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.710418] [<828e170c>] linux_pci_map_single+0x88/0xb4 [mt7628]
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.717339] [<828e17bc>] RTMP_AllocateRxPacketBuffer+0x84/0xac [mt7628]
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.724884] [<8289d66c>] RTMPInitTxRxRingMemory+0x3e4/0x518 [mt7628]
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.732163] [<828500d0>] hif_sys_init+0x84/0x198 [mt7628]
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.738470] [<82852a28>] rt28xx_init+0x7c/0x664 [mt7628]
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.744687] [<828def9c>] rt28xx_open+0xbc/0x128 [mt7628]
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.750911] [<82866c74>] RTMP_COM_IoctlHandle+0x4e0/0x9ec [mt7628]
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.758012] [<828de7d4>] MainVirtualIF_open+0x74/0x100 [mt7628]
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.764441] [<801fcbf0>] __dev_open+0xe0/0x148
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.768956] [<801fcef4>] __dev_change_flags+0xb8/0x174
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.774168] [<801fcfd8>] dev_change_flags+0x28/0x70
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.779131] [<8025e684>] devinet_ioctl+0x224/0x6dc
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.784000] [<801e216c>] sock_ioctl+0x298/0x2f8
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.788604] [<800b9850>] do_vfs_ioctl+0x4fc/0x5b4
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.793377] [<800b9958>] SyS_ioctl+0x50/0x98
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.797719] [<8000686c>] syscall_common+0x30/0x54
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.802486] 
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.803994] 
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.803994] Code: 2484ffff  00852021  00832024 <bc510000> 1482fffe  24420020  0000000f  8fbf0014  03e00008 
Tue Dec 26 23:15:38 2017 kern.warn kernel: [ 3609.814139] ---[ end trace fc35030c65c4779c ]---

```

MaFeSa, 2017-12-27:
Do git pull and recompile againg
There is a kernel bump to 4.4.108

alangregory, 2017-12-28:
Bumped stable to last commit, actually is 4.4.107.
This time module failed to load, going to investigate.
```
Thu Dec 28 04:29:35 2017 kern.warn kernel: [  141.190128] mt7628: module license 'unspecified' taints kernel.
Thu Dec 28 04:29:35 2017 kern.warn kernel: [  141.196150] Disabling lock debugging due to kernel taint
Thu Dec 28 04:29:35 2017 kern.warn kernel: [  141.227466] mt7628: Unknown symbol _raw_spin_lock_irqsave (err 0)
Thu Dec 28 04:29:35 2017 kern.warn kernel: [  141.233712] mt7628: Unknown symbol __per_cpu_offset (err 0)
Thu Dec 28 04:29:36 2017 kern.warn kernel: [  141.239385] mt7628: Unknown symbol _raw_spin_lock (err 0)
Thu Dec 28 04:29:36 2017 kern.warn kernel: [  141.244901] mt7628: Unknown symbol _raw_spin_lock_irq (err 0)
Thu Dec 28 04:29:36 2017 kern.warn kernel: [  141.250772] mt7628: Unknown symbol _raw_spin_unlock_bh (err 0)
Thu Dec 28 04:29:36 2017 kern.warn kernel: [  141.256790] mt7628: Unknown symbol _raw_spin_unlock_irqrestore (err 0)
Thu Dec 28 04:29:36 2017 kern.warn kernel: [  141.263463] mt7628: Unknown symbol del_timer_sync (err 0)
Thu Dec 28 04:29:36 2017 kern.warn kernel: [  141.269018] mt7628: Unknown symbol _raw_spin_lock_bh (err 0)

```

BertLammers, 2017-12-28:
@alangregory Having the same issue (will close the duplicated issue). I use a LinkitSmart device (mt7688 soc). 

Using linux kernel 4.4.107.

```
insmod mt7628.ko
failed to insert /lib/modules/4.4.107/mt7628.ko
```

Logread:
```
Thu Dec 28 17:32:12 2017 kern.warn kernel: [  845.146521] mt7628: module license 'unspecified' taints kernel.
Thu Dec 28 17:32:12 2017 kern.warn kernel: [  845.158280] Disabling lock debugging due to kernel taint
Thu Dec 28 17:32:12 2017 kern.warn kernel: [  845.193685] mt7628: Unknown symbol _raw_spin_lock_irqsave (err 0)
Thu Dec 28 17:32:12 2017 kern.warn kernel: [  845.205838] mt7628: Unknown symbol __per_cpu_offset (err 0)
Thu Dec 28 17:32:12 2017 kern.warn kernel: [  845.216933] mt7628: Unknown symbol _raw_spin_lock (err 0)
Thu Dec 28 17:32:12 2017 kern.warn kernel: [  845.227689] mt7628: Unknown symbol _raw_spin_lock_irq (err 0)
Thu Dec 28 17:32:12 2017 kern.warn kernel: [  845.239133] mt7628: Unknown symbol _raw_spin_unlock_bh (err 0)
Thu Dec 28 17:32:12 2017 kern.warn kernel: [  845.250830] mt7628: Unknown symbol _raw_spin_unlock_irqrestore (err 0)
Thu Dec 28 17:32:12 2017 kern.warn kernel: [  845.263834] mt7628: Unknown symbol del_timer_sync (err 0)
Thu Dec 28 17:32:12 2017 kern.warn kernel: [  845.274650] mt7628: Unknown symbol _raw_spin_lock_bh (err 0)
```

alangregory, 2017-12-29:
Looks like driver was built with SMP support, but device doesn't run with smp on kernel. @Nossiac can you confirm ?

Nossiac, 2017-12-29:
I used an universal kernel configuration (which is based on mt7621, with SMP enabled) for all drivers. That did saved me a lot of time and disk space. But drivers like 7612 can be used with either SMP or non-SMP enabled. Sorry that I forgot about this in the beginning.

I'm gonna push a quick fix by this weekend (there's a lot of work to do before the new year). Will reconfigure the automation server to start a new build for every platform (7621, 7620, 7628).

Nossiac, 2017-12-30:
http://nossiac.com/download/mtk-wifi-ko/mt7628-for-mt7628-linux-4.4.108.ko 
how about this one?

alangregory, 2017-12-30:
No problem, thanks for your effort. I'll do a new build test after new year.

alangregory, 2018-01-02:
@Nossiac ,It worked now, no error on insmod. About the kernel panic, I deactivated all mt76 driver components on build time and it worked.

closed 2018-01-03 by Nossiac