Nossiac/mtk-openwrt-feeds #9: insmod mt7628.ko failed

BertLammers, 2017-12-28:
Using linux kernel 4.4.107.

`insmod mt7628.ko
failed to insert /lib/modules/4.4.107/mt7628.ko`

Logread:
`Thu Dec 28 17:32:12 2017 kern.warn kernel: [  845.146521] mt7628: module license 'unspecified' taints kernel.
Thu Dec 28 17:32:12 2017 kern.warn kernel: [  845.158280] Disabling lock debugging due to kernel taint
Thu Dec 28 17:32:12 2017 kern.warn kernel: [  845.193685] mt7628: Unknown symbol _raw_spin_lock_irqsave (err 0)
Thu Dec 28 17:32:12 2017 kern.warn kernel: [  845.205838] mt7628: Unknown symbol __per_cpu_offset (err 0)
Thu Dec 28 17:32:12 2017 kern.warn kernel: [  845.216933] mt7628: Unknown symbol _raw_spin_lock (err 0)
Thu Dec 28 17:32:12 2017 kern.warn kernel: [  845.227689] mt7628: Unknown symbol _raw_spin_lock_irq (err 0)
Thu Dec 28 17:32:12 2017 kern.warn kernel: [  845.239133] mt7628: Unknown symbol _raw_spin_unlock_bh (err 0)
Thu Dec 28 17:32:12 2017 kern.warn kernel: [  845.250830] mt7628: Unknown symbol _raw_spin_unlock_irqrestore (err 0)
Thu Dec 28 17:32:12 2017 kern.warn kernel: [  845.263834] mt7628: Unknown symbol del_timer_sync (err 0)
Thu Dec 28 17:32:12 2017 kern.warn kernel: [  845.274650] mt7628: Unknown symbol _raw_spin_lock_bh (err 0)`

alangregory, 2017-12-28:
Hello @BertLammers , i'm having the same problem, could you please info about the device you're testing to #8  so it doesn't get duplicated on issues?

BertLammers, 2017-12-28:
Duplicated

closed 2017-12-28 by BertLammers