Nossiac/mtk-openwrt-feeds #11: mt7615-for-mt7621: weak 2.4g signal, no 5g signal and wrong mac address

hanwckf, 2018-01-03:
Phicomm K2P, mt7615 in DBDC mode. Build with lede-17.01 git branch and no kernel panic.
I can not search for its 5ghz signal but there is a very weak 2.4g signal ( about -80dbm ).I connect to this 2.4g signal, it works fine.
Mac address is wrong, it gives "02:00:00:10:00:00" to 2.4g(rax0) and "00:00:00:00:00:00" to 5g(ra0).
Config file is "mt7615.3.2G.dat" and "mt7615.3.5G.dat"
```shell
root@LEDE:~# mtkwifi reload
module is not loaded
root@LEDE:~# dmesg
[    0.000000] Linux version 4.4.108 (root@hanwckf-Aspire-VN7-592G) (gcc version 5.4.0 (LEDE GCC 5.4.0 r3799-ffbbcc9) ) #0 SMP Tue Jan 2 15:58:13 2018
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Phicomm K2P
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 08000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000] free_area_init_node: node 0, pgdat 80463000, node_mem_map 81000000
[    0.000000]   Normal zone: 256 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] PERCPU: Embedded 10 pages/cpu @8110c000 s8512 r8192 d24256 u40960
[    0.000000] pcpu-alloc: s8512 r8192 d24256 u40960 alloc=10*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Writing ErrCtl register=000280c0
[    0.000000] Readback ErrCtl register=000280c0
[    0.000000] Memory: 124576K/131072K available (3586K kernel code, 197K rwdata, 832K rodata, 236K init, 251K bss, 6496K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS:256
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.015467] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.080692] pid_max: default: 32768 minimum: 301
[    0.089966] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.102992] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    8.513025] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    8.513035] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    8.513045] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    8.513197] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.227672] Synchronize counters for CPU 1: done.
[    8.255162] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    8.255169] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    8.255175] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    8.255253] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.347747] Synchronize counters for CPU 2: done.
[    8.374002] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    8.374008] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    8.374014] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    8.374097] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.467295] Synchronize counters for CPU 3: done.
[    0.476708] Brought up 4 CPUs
[    0.485935] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[... 737 lines cut ...]
nf_nat_ipv4             3825  1 iptable_nat
nf_nat_masquerade_ipv4    1468  1 ipt_MASQUERADE
nf_nat_redirect         1019  1 xt_REDIRECT
nf_reject_ipv4          2147  1 ipt_REJECT
nf_reject_ipv6          2504  1 ip6t_REJECT
ppp_async               6592  0 
ppp_generic            20610  3 pppoe,ppp_async,pppox
pppoe                   8336  0 
pppox                   1418  1 pppoe
slhc                    4475  1 ppp_generic
x_tables               11783 22 ipt_REJECT,ipt_MASQUERADE,xt_time,xt_tcpudp,xt_state,xt_nat,xt_multiport,xt_mark,xt_mac,xt_limit,xt_conntrack,xt_comment,xt_TCPMSS,xt_REDIRECT,xt_LOG,iptable_mangle,iptable_filter,ip_tables,ip6t_REJECT,ip6table_mangle,ip6table_filter,ip6_tables
xt_LOG                   816  0 
xt_REDIRECT              720  0 
xt_TCPMSS               2832  2 
xt_comment               560125 
xt_conntrack            2256 14 
xt_limit                1120 20 
xt_mac                   688  0 
xt_mark                  720  0 
xt_multiport            1296  0 
xt_nat                  1200  0 
xt_state                 752  0 
xt_tcpudp               1840 10 
xt_time                 1680  0 
```

Nossiac, 2018-01-03:
Probably the driver had some trouble finding its eeporm data.

It usually resides in the "factory" partition on your flash, with a specific offset. It can also reside in the eFuse storage of mt7615 chips. A little bit tricky.

It may help to clarify the problem if you can locate the eeprom data.

hanwckf, 2018-01-03:
@Nossiac 
thx, I set ```E2pAccessMode=2``` in dat config files, and copy my eeprom to ```/lib/firmware/mt7615.eeprom.bin```
The 2.4g and 5g wireless seems to work fine, and it shows mac address correctly.
Then I launch iperf to test 5ghz wireless performance, after a while ( 230 seconds ), the router reboot automatically, the kernel log after ```mtkwifi reload``` is
```shell
root@LEDE:/# mtkwifi reload
[  348.709400] WifiSysClose(), wdev idx = 0
[  348.755024] WifiSysGetBssInfoState(): BssInfoIdx 0 not found!!!
[  348.766825] WifiSysUpdateBssInfoState(): BssInfoIdx 0 not found!!!
[  348.780313] br-lan: port 2(ra0) entered disabled state
[  348.840655] device ra0 left promiscuous mode
[  348.849367] br-lan: port 2(ra0) entered disabled state
[  348.889209] WifiSysClose(), wdev idx = 1
[  348.895971] WifiSysGetBssInfoState(): BssInfoIdx 1 not found!!!
[  348.895979] WifiSysUpdateBssInfoState(): BssInfoIdx 1 not found!!!
[  348.921707] br-lan: port 3(rax0) entered disabled state
[  348.982756] device rax0 left promiscuous mode
[  348.991656] br-lan: port 3(rax0) entered disabled state
[  349.002727] IPv6: ADDRCONF(NETDEV_UP): rax0: link is not ready
[  349.043322] WifiSysClose(), wdev idx = 2
[  349.051707] br-lan: port 4(apcli0) entered disabled state
[  349.112888] device apcli0 left promiscuous mode
[  349.122139] br-lan: port 4(apcli0) entered disabled state
[  349.133563] IPv6: ADDRCONF(NETDEV_UP): apcli0: link is not ready
[  349.160877] WifiSysClose(), wdev idx = 3
[  349.169078] MtAsicSetPiggyBack(802): Not support for HIF_MT yet!
[  349.181077] RTMPSetLEDStatus: before AndesLedEnhanceOP , status=0, LED_CMD=0!
[  349.195365] AndesLedEnhanceOP: Success!
[  349.203029] WifiSysClose(), wdev idx = 0
[  349.219229] MtAsicSetPiggyBack(802): Not support for HIF_MT yet!
[  349.231197] RTMPSetLEDStatus: before AndesLedEnhanceOP , status=0, LED_CMD=0!
[  349.245486] AndesLedEnhanceOP: Success!
[  349.253138] WifiSysClose(), wdev idx = 1
[  349.262479] kill LoopBackTxTask task failed!
[  349.271117] RTMPSetLEDStatus: before AndesLedEnhanceOP , status=4, LED_CMD=1!
[  349.285379] AndesLedEnhanceOP: Success!
[  349.298046] AndesRestartCheck: Current TOP_MISC2(0x7)
[  349.351503] CmdReStartDLRsp: Status Success!, Status(0)
[  349.361937] EventExtEventHandler: Unknown Ext Event(6f)
[  349.372387] AndesRestartCheck:  TOP_MISC2(1)
[  349.380907] RTMPSetLEDStatus: before AndesLedEnhanceOP , status=2, LED_CMD=1!
[  349.395137] AndesLedEnhanceOP: Success!
[  349.403036] RT28xxPciAsicRadioOff(): Not support for HIF_MT yet!
[  349.415056] tx_kickout_fail_count = 0
[  349.422357] tx_timeout_fail_count = 0
[  349.429664] rx_receive_fail_count = 0
[  349.436938] alloc_cmd_msg = 2096
[  349.443369] free_cmd_msg = 2097
[  349.451574] cut_through_token_list_destroy(): 8641d888,8641d888
[  349.464205] cut_through_token_list_destroy(): 8641d898,8641d898
[  349.481602] FwOwn()::Set Fw Own
[  349.491931] RTMP_AllTimerListRelease: Size=0
[  349.500511] FwOwn()::Return since already in Fw Own...
[  349.510789] <---HwCtrlThread
[  349.516997] br-lan: port 5(apclix0) entered disabled state
[  349.578160] device apclix0 left promiscuous mode
[  349.587538] br-lan: port 5(apclix0) entered disabled state
[  349.599109] IPv6: ADDRCONF(NETDEV_UP): apclix0: link is not ready
[  349.787179] RtmpOSNetDevDetach(): RtmpOSNetDeviceDetach(), dev->name=ra0!
[  349.858068] In remove, pci probe count=0
[  349.866040] Unregister PCI device driver
[  350.056111] register mt_drv
[  350.074593]
[  350.074593]
[  350.074593] === pAd = c1281000, size = 3629168 ===
[  350.074593]
[  350.094751] <-- RTMPAllocAdapterBlock, Status=0
[  350.103833] get_wdev_by_idx: invalid idx(0)
[  350.112162] pAd->PciHif.CSRBaseAddress =0xc1180000, csr_addr=0xc1180000!
[  350.125505] get_wdev_by_idx: invalid idx(0)
[... 558 lines cut ...]
[  601.649549] Modules linked in: mt7615(P) pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_itu_t crc_ccitt ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables eeprom_93cx6 leds_gpio gpio_button_hotplug [last unloaded: mt7615]
[  601.764358] Process swapper/0 (pid: 0, threadinfo=80454000, task=80463d78, tls=00000000)
[  601.780439] Stack : 00000001 00000000 87c0de54 84c6c848 c15b1000 00000001 00000000 00000000
          85c47200 c129f1d0 40800400 000075b5 84f00000 84f06a74 01800000 00000000
          00000000 00010000 00000000 00000000 00000000 00000000 00000000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          ...
[  601.851135] Call Trace:
[  601.856271] [<84d3863c>] FastPathDequeBh+0x1bc/0x8fc [mt7615]
[  601.868001] [<800302dc>] tasklet_hi_action+0x144/0x19c
[  601.878216] [<8002fbcc>] __do_softirq+0x294/0x2e0
[  601.887566] [<8002feb4>] irq_exit+0x78/0x94
[  601.895883] [<801e5f8c>] plat_irq_dispatch+0xb4/0xdc
[  601.905754] [<80005988>] except_vec_vi_end+0xb8/0xc4
[  601.915642] [<80013680>] r4k_wait_irqoff+0x18/0x20
[  601.925186] [<8005efd4>] cpu_startup_entry+0x17c/0x1e4
[  601.935423] [<80485c08>] start_kernel+0x4a4/0x4c4
[  601.944772]
[  601.947719]
Code: 24020001  a7a20046  8e02000c <8c420054> a7a20048  8e02000c  afa20064  92020008  a3a20039
[  601.967766] ---[ end trace 609676f0e79b0222 ]---
[  601.992186] Kernel panic - not syncing: Fatal exception in interrupt
[  602.018239] Rebooting in 3 seconds..
```

Nossiac, 2018-01-12:
close the original issue. trace the new issue in #13 .

closed 2018-01-12 by Nossiac