Nossiac/mtk-openwrt-feeds #13: MT7615 driver mystically caused router reboot

maxwellchar, 2018-01-04:
Router: LEDE-1701, MT7621+MT7615
When I download some large files with 5G wifi , sometimes the wifi crash, sometimes the router reboot mysteriously.
It may cause kernel panic:
softirq: huh, entered softirq 0 HI 80030198 with preempt_count 00000100, exited with fffffe00?

Nossiac, 2018-01-10:
steady reproducing steps are appreciated.

maxwellchar, 2018-01-11:
I copyed my eeprom to /lib/firmware/mt7615.eeprom.bin, and got data files from stock firmware, 
then copyed them to /etc/wireless/mt7615/mt7615-sku.dat, /etc/wireless/mt7615/mt7615-sku-bf.dat.
Rebooted router and reloaded wifi driver. I just renamed the wifi ssid. 
When downloading some files with wifi connection, the router auto reboot.
It may be the same problem with #11 , @hanwckf he got auto reboot too.