Nossiac/mtk-openwrt-feeds #14: MediaTek MT7602EN

build000, 2018-01-06:
If possible add MediaTek MT7602EN drivers ? - this requires to https://wikidevi.com/wiki/MQmaker_WiTi_Board (OpenWrt/Lede).

Nossiac, 2018-01-10:
AFAK, mt7602 actually shares driver with mt7612, before it works, a few configuration options need to be updated first. An init version can be uploaded this week.

Nossiac, 2018-01-12:
https://github.com/Nossiac/mtk-openwrt-feeds/commit/fe0c9c797f10e75453c4ca468a08f1557ef76ae8   added 7602 support ( it actually shares driver with mt7612).
The ko module will be refreshed in a couple of hours.

build000, 2018-01-12:
Thx.
I will test how soon it will be possible and possibly I will answer whether and what are the problems under WiTi.

closed 2018-02-23 by Nossiac