Nossiac/mtk-openwrt-feeds #15: Driver failed to load

sanitariu, 2018-01-12:
root@OpenWrt:/# modprobe mt7628.ko
failed to mmap /lib/modules/4.9.76/mt7628.ko

Linux version 4.9.76

Nossiac, 2018-01-13:
the ko was built with kernel 4.4.x from lede-17.01.
you are using openwrt trunk which uses kernel 4.9.x, which is not supported for the time being.

sanitariu, 2018-01-14:
Thanks !
Can you tell me which revision to checkout exactly ?

Nossiac, 2018-01-14:
there are several tags from lede-1701 branch, currently I built them all:

```
lede-1701.latest --> linux-4.4.108  
lede-1701.04     --> linux-4.4.50  
lede-1701.03     --> linux-4.4.61  
lede-1701.02     --> linux-4.4.71  
lede-1701.01     --> linux-4.4.89  
lede-1701.00     --> linux-4.4.92  

```

closed 2018-01-14 by Nossiac

sanitariu, 2018-01-15:
Thanks for the help bit after merging lede+openwrt this is impossible.
Do you have plans to update for newer kernel ?