Nossiac/mtk-openwrt-feeds #16: About MT7620 Antenna's INTERNAL mode and EXTERNAL mode

chobitsos, 2018-01-16:
Hello dear developer，This driver works very well！My only question is: I did not find the antenna mode choice when compiling the firmware.

Usually in the old driver which has the following mode selection
`CONFIG_INTERNAL_PA_INTERNAL_LNA=y
 CONFIG_INTERNAL_PA_EXTERNAL_LNA is not set
 CONFIG_EXTERNAL_PA_EXTERNAL_LNA is not set
CONFIG_RT2860V2_AP_RTMP_INTERNAL_TX_ALC is not set
CONFIG_RT2860V2_AP_RTMP_TEMPERATURE_COMPENSATION is not set`

Does your current driver support these modes?

Nossiac, 2018-01-17:
The driver was built with "CONFIG_INTERNAL_PA_INTERNAL_LNA=y". If you want to customize the driver as you wish, I suggest that you build it on your own.
According to the NDA, I'm not going to publish the source code here, but I believe you will be able to find it somewhere else, even on github, :-)

chobitsos, 2018-01-17:
i just need CONFIG_INTERNAL_PA_INTERNAL_LNA = y :-)

closed 2018-01-17 by chobitsos

namoaton, 2018-05-09:
hello. Where to get precompiled mt712 driver with enabled external LNA and PA?

kennylam777, 2018-05-29:
Well, I know there is an NDA, so I am not asking for the source code.

I am asking for the building method instead :)

Driver version: mtk-openwrt-sdk-20170518-1443366e
LEDE version: lede-v17.04.4
Target module: mt7620 on linux-4.4.92

What I did was
1. Copy "mtk-openwrt-sdk-20170518-1443366e/package/ramips"  to "lede/packge/ramips"
2. Copy "mtk-openwrt-sdk-20170518-1443366e/dl/mt7620-p4rev-120998.tar.bz2" to "lede/dl/mt7620-p4rev-120998.tar.bz2"
3. make menuconfig, select a mt7620 board , enable the mt7620 module in "MTK Properties"
4. make package/mt7620/compile

And then I got follow error message.

``
builder@6daa6f0816ca:/data/lede$ make package/mt7620/compile V=s
make[1]: Entering directory '/data/lede'
make[2]: Entering directory '/data/mtk-openwrt-feeds/drivers/mt7620'
. /data/lede/include/shell.sh; bzcat /data/lede/dl/mt7620-p4rev-120998.tar.bz2 | tar -C /data/lede/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7620/mt7620-p4rev-120998/ -xf - 
[ ! -d ./src/ ] || cp -fpR ./src/* /data/lede/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7620/mt7620-p4rev-120998

Applying ./patches/001-fix-kernel-warning.patch using plaintext: 
patching file src/common/cmm_data_pci.c
patching file src/include/os/rt_drv.h
patching file src/include/os/rt_linux.h
patching file src/include/rtmp.h
patching file src/os/linux/rt_profile.c
patching file src/os/linux/rt_rbus_pci_drv.c

Applying ./patches/002-fix-insmod-fail.patch using plaintext: 
patching file src/os/linux/rt_rbus_pci_drv.c
Hunk #1 succeeded at 776 (offset 3 lines).

Applying ./patches/003-support-hwnat.patch using plaintext: 
patching file build/Makefile

Applying ./patches/025-wpa-key-mismatch.patch using plaintext: 
patching file src/ap/ap.c
patching file src/ap/ap_cfg.c
patching file src/common/cmm_profile.c
patching file src/include/rtmp.h
Hunk #1 succeeded at 1467 (offset -8 lines).
touch /data/lede/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7620/mt7620-p4rev-120998/.prepared_23bcf3a489b01103da9ceb3877dde7c4
rm -f /data/lede/staging_dir/target-mipsel_24kc_musl-1.1.16/stamp/.mt7620_installed
(cd /data/lede/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7620/mt7620-p4rev-120998/./; if [ -x ./configure ]; then find /data/lede/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7620/mt7620-p4rev-120998/ -name config.guess | xargs -r chmod u+w; find /data/lede/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7620/mt7620-p4rev-120998/ -name config.guess | xargs -r -n1 cp --remove-destination /data/lede/scripts/config.guess; find /data/lede/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7620/mt7620-p4rev-120998/ -name config.sub | xargs -r chmod u+w; find /data/lede/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7620/mt7620-p4rev-120998/ -name config.sub | xargs -r -n1 cp --remove-destination /data/lede/scripts/config.sub; AR="mipsel-openwrt-linux-musl-gcc-ar" AS="mipsel-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap /data/lede/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7620/mt7620-p4rev-120998:mt7620-p4rev-120998 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mipsel-openwrt-linux-musl-ld NM="mipsel-openwrt-linux-musl-gcc-nm" CC="mipsel-openwrt-linux-musl-gcc" GCC="mipsel-openwrt-linux-musl-gcc" CXX="mipsel-openwrt-linux-musl-g++" RANLIB="mipsel-openwrt-linux-musl-gcc-ranlib" STRIP=mipsel-openwrt-linux-musl-strip OBJCOPY=mipsel-openwrt-linux-musl-objcopy OBJDUMP=mipsel-openwrt-linux-musl-objdump SIZE=mipsel-openwrt-linux-musl-size CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /data/lede/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7620/mt7620-p4rev-120998:mt7620-p4rev-120998 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /data/lede/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7620/mt7620-p4rev-120998:mt7620-p4rev-120998 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/data/lede/staging_dir/target-mipsel_24kc_musl-1.1.16/usr/include -I/data/lede/staging_dir/target-mipsel_24kc_musl-1.1.16/include -I/data/lede/staging_dir/toolchain-mipsel_24kc_gcc-5.4.0_musl-1.1.16/usr/include -I/data/lede/staging_dir/toolchain-mipsel_24kc_gcc-5.4.0_musl-1.1.16/include/fortify -I/data/lede/staging_dir/toolchain-mipsel_24kc_gcc-5.4.0_musl-1.1.16/include " LDFLAGS="-L/data/lede/staging_dir/target-mipsel_24kc_musl-1.1.16/usr/lib -L/data/lede/staging_dir/target-mipsel_24kc_musl-1.1.16/lib -L/data/lede/staging_dir/toolchain-mipsel_24kc_gcc-5.4.0_musl-1.1.16/usr/lib -L/data/lede/staging_dir/toolchain-mipsel_24kc_gcc-5.4.0_musl-1.1.16/lib -znow -zrelro "   ./configure --target=mipsel-openwrt-linux --host=mipsel-openwrt-linux --build=x86_64-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc  [... cut ...] 518-1443366e/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7620/mt7620-p4rev-120998/build//../src/common/crypt_md5.o /data/mtk-openwrt-sdk-20170518-1443366e/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7620/mt7620-p4rev-120998/build//../src/common/crypt_md5.c
......
``

It seems there was problem in LEDE after the line of 
``
make -f scripts/Makefile.build obj=/***/linux-ramips_mt7620/mt7620-p4rev-120998/build/
```
and the module compiling was just skipped.

Nossiac, 2018-05-29:
@kennylam777 
according to your log, you should take a close look at 2 places:
1. the overall build command was written in the `Build/Compile` section of openwrt's package makefile.
2. the actual source code makefile was mt7620-p4rev-120998/build/Makefile.

and possible reasons:
1. you gave a wrong path to openwrt's "Build/Compile".
2. the driver did not configured properly so that the target was skipped/ignored. (make menuconfig)

Pentiumluyu, 2018-05-30:
@kennylam777  @Nossiac Under the Openwrt build system, you need to build Linux before building the module. The Openwrt wiki actually states that. [How to Build a Single Package](https://wiki.openwrt.org/doc/howtobuild/single.package#kernel_modules)
```
make menuconfig
make target/linux/compile
make package/mt7620/compile
```