Nossiac/mtk-openwrt-feeds #17: Stable kernel bumped to 4.4.110

HeadLessHUN, 2018-01-17:
Hi!
Yesterday i tried to compile an fw with your wifi drivers for xiaomi miwifi 3g 7603/7612 for 7621, with 2 days old tree but i noticed that ko files are 0 bytes... I've found out there is no version for the latest stable kernel which is 4.4.110. Tried to replace with 4.4.108 version it can be loaded but not working mt7603 seem to be working but can't found ssid. 7612 reboot after load and refresh on luci. Mt76* packages removed from the build. So could you upgrade to the latest kernel? I'd appreciate that.

Nossiac, 2018-01-17:
autobuild started. the new kernel module should be ready in a few hours.
as for 7603, there's an known issue in its tx path. I'll let you know if I make any progress.

kysonlok, 2018-01-17:
> autobuild started. the new kernel module should be ready in a few hours.
> as for 7603, there's an known issue in its tx path. I'll let you know if I
> make any progress.

Hi Nossias,

How do you do about autobuild? Using buildbot or other framework? Could you
share the autobuild source code?

Nossiac, 2018-01-17:
jenkins might be the most popular open source automation service. What I'm using is a hand made script here: [https://github.com/Nossiac/simpletask](https://github.com/Nossiac/simpletask) 
not very powerful yet, but feasible enough to handle tasks like that.

kysonlok, 2018-01-17:
OK, thanks!

HeadLessHUN, 2018-01-17:
tx path issue means that it fail sometime or it can't connect?

thanks for the short reaction.

In theoretical can i test i on a live system without rebuild?
Install the luci pages, uc2dat, etc...
download the new kmod from your site. place it in /lib/modules/4.4.110/mt*.ko
```
wget -O /lib/modules/4.4.110/mt7612.ko "http://119.23.148.191:8888/mtk-wifi-ko/mt7612-for-mt7621-linux-4.4.110.ko"
wget -O /lib/modules/4.4.110/mt7603.ko "http://119.23.148.191:8888/mtk-wifi-ko/mt7603-for-mt7621-linux-4.4.110.ko"
```
then 

```
rmmod mt76
rmmod mt76x2e
rmmod mt7603e
insmod mt7612
insmod mt7603
```
it should be working, shouldn't it?

closed 2018-01-18 by HeadLessHUN