Nossiac/mtk-openwrt-feeds #18: MT7615 for kernel 3.10...

MrFishchev, 2018-01-17:
Hi, could you tell me please, how can I compile mt_wifi (chip 7615) for linux kernel 3.10?
I have already tried everything, that I know.
I have a board with MT7621 cpu and two MT7615N chips for wi-fi (2Ghz and 5Ghz).
I see that your code uses precompile .ko files, but how I can compile the same for only 3.10 kernel?

Thanks.

Nossiac, 2018-01-17:
have u got the source code? with the source code, it's pretty easy to build 7615 with 3.10 kernel, because mediatek already did that.

but why? seems kernel 3.10 is dropped by openwrt/lede long ago.

MrFishchev, 2018-01-17:
I found sources of driver (https://github.com/dt99/mt7615) but I can't build them for MT7621:
I tried this:
1. In mt_wifi_embedded Makefile I chose MT7621 platform and changed LINUX_SRC & CROSS_COMPILE, but I got the errors

> mipsel-openwrt-linux-gcc: error: unrecognized command line option '-m64'
> mipsel-openwrt-linux-gcc: error: unrecognized command line option '-mno-red-zone'
> mipsel-openwrt-linux-gcc: error: unrecognized command line option '-mcmodel=kernel'
> mipsel-openwrt-linux-gcc: error: unrecognized command line option '-maccumulate-outgoing-args'

2. I tried build it for x86 ubuntu (my sistem). Success.

I need 3.10 because I have other drivers for the kernel 3.10.

Nossiac, 2018-01-17:
are you going to use the driver with openwrt? to build the driver with openwrt, the best choice is to write a kmod makefile, openwrt's build scripts will help with all the cross-compiling configurations.

if you need to build it manually, how about this:
`make -C <openwrt-kernel-path> ARCH=mips CROSS_COMPILE=mipsel-openwrt-linux-musl- SUBDIRS=<driver-source-dir>/mt_wifi_ap CONFIG_XXXXX=y CONFIG_YYYY=y modules`

closed 2018-02-09 by MrFishchev