Nossiac/mtk-openwrt-feeds #20: Using with Linkit 7688

nicupavel, 2018-01-22:
Hi,
Is this driver supposed to work with linkit 7688 platform? (http://wiki.seeedstudio.com/wiki/LinkIt_Smart_7688)

I downloaded mt7628-for-mt7628-linux but doesn't really work. It does create the interfaces correctly but you can't connect to AP and scanning the network with iwinfo ra0 scan always reports 0 results.

For sure it doesn't look into the correct flash partition to get the MAC address as it comes with 00:00:00:00:00:xx.
The flash partition on Linkit with MAC settings is called "factory".

```
[   19.077478] TX_BCN DESC a71dc000 size = 320
[   19.085871] RX[0] DESC a71df000 size = 1024
[   19.094378] RX[1] DESC a71e0000 size = 1024
[   19.169167] E2pAccessMode=0
[   19.177126] AndesSendCmdMsg: Could not send in band command due to diable fRTMP_ADAPTER_MCU_SEND_IN_BAND_CMD
[   19.199759] APSDCapable[0]=0
[   19.205475] APSDCapable[1]=0
[   19.211168] APSDCapable[2]=0
[   19.216871] APSDCapable[3]=0
[   19.222564] APSDCapable[4]=0
[   19.228266] APSDCapable[5]=0
[   19.233959] APSDCapable[6]=0
[   19.239661] APSDCapable[7]=0
[   19.245366] APSDCapable[8]=0
[   19.251059] APSDCapable[9]=0
[   19.256769] APSDCapable[10]=0
[   19.262634] APSDCapable[11]=0
[   19.268505] APSDCapable[12]=0
[   19.274370] APSDCapable[13]=0
[   19.280244] APSDCapable[14]=0
[   19.286114] APSDCapable[15]=0
[   19.291974] default ApCliAPSDCapable[0]=0
[   19.493267] Key1Str is Invalid key length(0) or Type(0)
[   19.504008] Key2Str is Invalid key length(0) or Type(0)
[   19.514760] Key3Str is Invalid key length(0) or Type(0)
[   19.525507] Key4Str is Invalid key length(0) or Type(0)
[   19.595679] RTMPSetDefaultChannel() : default channel to 1 
[   19.606786] load fw image from fw_header_image
[   19.615590] AndesMTLoadFwMethod1(2182)::pChipCap->fw_len(63888)
[   19.627314] FW Version:20151201
[   19.633900] FW Build Date:20151201183641
[   19.762177] CmdAddressLenReq:(ret = 0)
[   19.770438] CmdFwStartReq: override = 1, address = 1048576
[   19.781500] CmdStartDLRsp: WiFI FW Download Success
[   19.793578] MtAsicDMASchedulerInit(): DMA Scheduler Mode=0(LMAC)
[   19.805553] efuse_probe: efuse = 10000012
[   19.813486] RtmpChipOpsEepromHook::e2p_type=0, inf_Type=4
[   19.824179] RtmpEepromGetDefault::e2p_dafault=2
[   19.833158] RtmpChipOpsEepromHook: E2P type(2), E2pAccessMode = 2, E2P default = 2
[   19.848153] NVM is FLASH mode
[   19.854074] 1. Phy Mode = 6
[   19.904670] RtmpOSFileOpen(): Error 2 opening /etc_ro/wlan/MT7628_EEPROM_20140317.bin
[   19.920383] The EEPROM in Flash is wrong, use default
[   19.930397] proprietary driver not support flash write, will write on ated.
[   20.110762] Country Region from e2p = ffff
[   20.119016] tssi_1_target_pwr_g_band = 39
[   20.126997] 2. Phy Mode = 6
[   20.132837] 3. Phy Mode = 6
[   20.138388] NICInitPwrPinCfg(11): Not support for HIF_MT yet!
[   20.149776] NICInitializeAsic(651): Not support rtmp_mac_sys_reset () for HIF_MT yet!
[   20.165297] mt_mac_init()-->
[   20.170989] MtAsicInitMac()-->
[   20.177678] CmdSlotTimeSet:(ret = 0)
[   20.185041] mt7628_init_mac_cr()-->
[   20.191954] MtAsicSetMacMaxLen(1276): Set the Max RxPktLen=1024!
[   20.203900] <--mt_mac_init()
[   20.209761]  WTBL Segment 1 info:
[   20.216329]          MemBaseAddr/FID:0x28000/0
[   20.223915]          EntrySize/Cnt:32/128
[   20.230650]  WTBL Segment 2 info:
[   20.237209]          MemBaseAddr/FID:0x40000/0
[   20.244804]          EntrySize/Cnt:64/128
[   20.251526]  WTBL Segment 3 info:
[   20.258088]          MemBaseAddr/FID:0x42000/64
[   20.265867]          EntrySize/Cnt:64/128
[   20.272589]  WTBL Segment 4 info:
[   20.279152]          MemBaseAddr/FID:0x44000/128
[   20.287093]          EntrySize/Cnt:32/128
[   20.293896] AntCfgInit(2958): Not support for HIF_MT yet!
[   20.304683] MCS Set = 00 00 00 00 00
[   20.311755] MtAsicSetChBusyStat(861): Not support for HIF_MT yet!
[   20.323881] [PMF]ap_pmf_init:: apidx=0, MFPC=0, MFPR=0, SHA256=0
[   20.335785] RTMPSetPhyMode(): channel out of range, use first ch=0
[   20.348096] MtAsicSetRalinkBurstMode(3048): Not support for HIF_MT yet!
[   20.361203] MtAsicSetPiggyBack(796): Not support for HIF_MT yet!
[   20.388519] CmdChannelSwitch: central channel = 0 is invalid
[   20.400845] reload DPD from flash , 0x9F = [c6ff] doReload bit7[1]
[   20.413158] CmdLoadDPDDataFromFlash: Channel = 0 (1~14), INSANE , return
[   20.426450] MtAsicSetTxPreamble(3027): Not support for HIF_MT yet!
[   20.438796] MtAsicSetPreTbtt(): bss_idx=0, PreTBTT timeout = 0xf0
[   20.450883] ap_ftkd> Initialize FT KDP Module...
[   20.460039] Main bssid = 00:00:00:00:00:50
[   20.468234] <==== rt28xx_init, Status=0
[   20.493820] @@@ ed_monitor_exit : ===>
[   20.501347] @@@ ed_monitor_exit : <===
[   20.508829] mt7628_set_ed_cca: TURN OFF EDCCA  mac 0x10618 = 0xd7083f0f, EDCCA_Status=0
[   20.524706] WiFi Startup Cost (ra0): 1.450s
[   21.237856] This channel is out of channel list, set as the first channel(0) 
[   21.237856]  reload DPD from flash , 0x9F = [c6ff] doReload bit7[1]
[   21.291971] CmdLoadDPDDataFromFlash: Channel = 6, DoReload = 1
[   21.327609] MtAsicSetPreTbtt(): bss_idx=0, PreTBTT timeout = 0x0
[   21.339597] MtAsicSetPiggyBack(796): Not support for HIF_MT yet!
[   21.351507] ap_ftkd> Release FT KDP Module...
[   21.360168] [PMF]ap_pmf_init:: apidx=0, MFPC=0, MFPR=0, SHA256=0
[   21.372105] MtAsicSetRalinkBurstMode(3048): Not support for HIF_MT yet!
[   21.385209] MtAsicSetPiggyBack(796): Not support for HIF_MT yet!
[   21.421879] reload DPD from flash , 0x9F = [c6ff] doReload bit7[1]
[   21.434202] CmdLoadDPDDataFromFlash: Channel = 6, DoReload = 1
[   21.445998] MtAsicSetTxPreamble(3027): Not support for HIF_MT yet!
[   21.458346] AddTxSType: already registered TxSType (PID = 32, Format = 0
[   21.471688] AddTxSType: already registered TxSType (PID = 6, Format = 0
[   21.484798] AddTxSType: already registered TxSType (PID = 8, Format = 0
[   21.497911] MtAsicSetPreTbtt(): bss_idx=0, PreTBTT timeout = 0xf0
[   21.509993] ap_ftkd> Initialize FT KDP Module...
[   21.519143] Main bssid = 00:00:00:00:00:50
[   21.549042] MtAsicSetPreTbtt(): bss_idx=0, PreTBTT timeout = 0x0
[   21.561030] MtAsicSetPiggyBack(796): Not support for HIF_MT yet!
[   21.572941] ap_ftkd> Release FT KDP Module...
[   21.581599] [PMF]ap_pmf_init:: apidx=0, MFPC=0, MFPR=0, SHA256=0
[   21.593545] MtAsicSetRalinkBurstMode(3048): Not support for HIF_MT yet!
[   21.606648] MtAsicSetPiggyBack(796): Not support for HIF_MT yet!
[   21.643276] reload DPD from flash , 0x9F = [c6ff] doReload bit7[1]
[   21.655583] CmdLoadDPDDataFromFlash: Channel = 6, DoReload = 1
[   21.667553] MtAsicSetTxPreamble(3027): Not support for HIF_MT yet!
[   21.679912] AddTxSType: already registered TxSType (PID = 32, Format = 0
[   21.693258] AddTxSType: already registered TxSType (PID = 6, Format = 0
[   21.706366] AddTxSType: already registered TxSType (PID = 8, Format = 0
[   21.719476] MtAsicSetPreTbtt(): bss_idx=0, PreTBTT timeout = 0xf0
[   21.731555] ap_ftkd> Initialize FT KDP Module...
[   21.740706] Main bssid = 00:00:00:00:00:50
[   21.766276] The Access Control List is empty!
[   21.775011] No need to clear the Access Control List!
[   39.302368] random: nonblocking pool is initialized
[  121.339274] reload DPD from flash , 0x9F = [c6ff] doReload bit7
```

kysonlok, 2018-01-22:
Closed wifi driver not based mac80211/cfg80211 framework, so iw/iwinfo cli can't work correctly. Here is a patch[1] for iwinfo scan.

[1] https://github.com/gl-inet/lede-17.01/blob/lede-17.01/package/network/utils/iwinfo/patches/000-add_rtwifi_support.patch

nicupavel, 2018-01-22:
Yes of course. Looks like it's easier to find 4.1.0.0 driver on google than mediatek online partener website so after getting 4.1.0.0 archive and some modifications I managed to get a working configuration. So probably only flash partition name and maybe wmac interrupt needs to be customised.

closed 2018-01-22 by nicupavel