Nossiac/mtk-openwrt-feeds #21: build ok ,but can not up wifi

hzjnet, 2018-01-25:
[  466.130000] mt7615: module license 'Proprietary' taints kernel.
[  466.150000] Disabling lock debugging due to kernel taint
[  466.230000] register mt_drv
[  466.260000] 
[  466.260000] 
[  466.260000] === pAd = c0c01000, size = 3629168 ===
[  466.260000] 
[  466.280000] <-- RTMPAllocAdapterBlock, Status=0
[  466.290000] get_wdev_by_idx: invalid idx(0)
[  466.300000] pAd->PciHif.CSRBaseAddress =0xc0b00000, csr_addr=0xc0b00000!
[  466.310000] get_wdev_by_idx: invalid idx(0)
[  466.320000] RTMPInitPCIeDevice():device_id=0x7615
[  466.330000] DriverOwn()::Try to Clear FW Own...
[  466.640000] DriverOwn()::Success to clear FW Own
[  466.650000] mt_pci_chip_cfg(): HWVer=0x8a10, FWVer=0x8a10, pAd->ChipID=0x7615
[  466.670000] mt_pci_chip_cfg(): HIF_SYS_REV=0x76150001
[  466.680000] RtmpChipOpsHook(493): Not support for HIF_MT yet! MACVersion=0x0
[  466.700000] mt7615_init()-->
[  466.700000] Use 1st ePAeLNA default bin.
[  466.710000] Use 0st /etc/wlan/MT7615E_EEPROM1.bin default bin.
[  466.730000] <--mt7615_init()
[  466.730000] ChipOpsMCUHook
[  466.750000] cut_through_token_list_init(): TokenList inited done!id_head/tail=0/4096
[  466.760000] cut_through_token_list_init(): 870d8588,870d8588
[  466.780000] cut_through_token_list_init(): TokenList inited done!id_head/tail=0/4096
[  466.790000] cut_through_token_list_init(): 870d8598,870d8598
[  466.810000] <-- RTMPAllocTxRxRingMemory, Status=0
[  466.820000] get_wdev_by_idx: invalid idx(0)
[  466.830000] get_wdev_by_idx: invalid idx(0)

MrFishchev, 2018-01-29:
Oh, I've the same error.. Also I tried build it from sources for 3.10.14 kernel, but I got the same.
@Nossiac Do you have precompiled .ko MT7615 for 4.9.77 platform MT7621, thanks!

MrFishchev, 2018-01-29:
I just try install openwrt lede 17.01 with kernel 4.4.50, and your driver (mtk properties -> mt7615) was install in /lib/modules but it hasn't autostart. If I use insmod, I get this output:

root@LEDE:/# insmod /lib/modules/4.4.50/mt7615.ko
[  112.980000] mt7615: module license 'Proprietary' taints kernel.
[  112.990000] Disabling lock debugging due to kernel taint
[  113.120000] register mt_drv
[  113.140000]
[  113.140000]
[  113.140000] === pAd = c1381000, size = 3629168 ===
[  113.140000]
[  113.160000] <-- RTMPAllocAdapterBlock, Status=0
[  113.170000] get_wdev_by_idx: invalid idx(0)
[  113.180000] pAd->PciHif.CSRBaseAddress =0xc1280000, csr_addr=0xc1280000!
[  113.200000] get_wdev_by_idx: invalid idx(0)
[  113.200000] RTMPInitPCIeDevice():device_id=0x7615
[  113.210000] DriverOwn()::Try to Clear FW Own...
[  113.450000] DriverOwn()::Success to clear FW Own
[  113.460000] mt_pci_chip_cfg(): HWVer=0x8a10, FWVer=0x8a10, pAd->ChipID=0x7615
[  113.470000] mt_pci_chip_cfg(): HIF_SYS_REV=0x76150001
[  113.480000] RtmpChipOpsHook(493): Not support for HIF_MT yet! MACVersion=0x0
[  113.500000] mt7615_init()-->
[  113.500000] Use 1st ePAeLNA default bin.
[  113.510000] Use 0st /etc/wlan/MT7615E_EEPROM1.bin default bin.
[  113.520000] <--mt7615_init()
[  113.530000] ChipOpsMCUHook
[  113.540000] cut_through_token_list_init(): TokenList inited done!id_head/tail                                                        =0/4096
[  113.560000] cut_through_token_list_init(): 8e244288,8e244288
[  113.570000] cut_through_token_list_init(): TokenList inited done!id_head/tail                                                        =0/4096
[  113.580000] cut_through_token_list_init(): 8e244298,8e244298
[  113.590000] <-- RTMPAllocTxRxRingMemory, Status=0
[  113.600000] get_wdev_by_idx: invalid idx(0)
[  113.610000] get_wdev_by_idx: invalid idx(0)
[  113.620000] get_wdev_by_idx: invalid idx(0)
[  113.630000] get_wdev_by_idx: invalid idx(0)
[  113.640000] get_wdev_by_idx: invalid idx(0)
[  113.650000] get_wdev_by_idx: invalid idx(0)
[  113.660000] get_wdev_by_idx: invalid idx(0)
[  113.660000] pci probe count=1
[  113.690000]
[  113.690000]
[  113.690000] === pAd = c1801000, size = 3629168 ===
[  113.690000]
[  113.720000] <-- RTMPAllocAdapterBlock, Status=0
[  113.730000] get_wdev_by_idx: invalid idx(0)
[  113.730000] pAd->PciHif.CSRBaseAddress =0xc1700000, csr_addr=0xc1700000!
[  113.750000] get_wdev_by_idx: invalid idx(0)
[  113.760000] RTMPInitPCIeDevice():device_id=0x7615
[  113.770000] DriverOwn()::Try to Clear FW Own...
[  114.010000] DriverOwn()::Success to clear FW Own
[  114.010000] mt_pci_chip_cfg(): HWVer=0x8a10, FWVer=0x8a10, pAd->ChipID=0x7615
[  114.030000] mt_pci_chip_cfg(): HIF_SYS_REV=0x76150001
[  114.040000] RtmpChipOpsHook(493): Not support for HIF_MT yet! MACVersion=0x0
[  114.050000] mt7615_init()-->
[  114.060000] Use 2nd ePAeLNA default bin.
[  114.070000] Use 1st /etc/wlan/MT7615E_EEPROM2.bin default bin.
[  114.080000] <--mt7615_init()
[  114.080000] ChipOpsMCUHook
[  114.100000] cut_through_token_list_init(): TokenList inited done!id_head/tail                                                        =0/4096
[  114.110000] cut_through_token_list_init(): 8da7c808,8da7c808
[  114.120000] cut_through_token_list_init(): TokenList inited done!id_head/tail                                                        =0/4096
[  114.140000] cut_through_token_list_init(): 8da7c818,8da7c818
[  114.150000] <-- RTMPAllocTxRxRingMemory, Status=0
[  114.160000] get_wdev_by_idx: invalid idx(0)
[  114.170000] get_wdev_by_idx: invalid idx(0)
[  114.180000] get_wdev_by_idx: invalid idx(0)
[  114.190000] get_wdev_by_idx: invalid idx(0)
[  114.190000] get_wdev_by_idx: invalid idx(0)
[  114.200000] get_wdev_by_idx: invalid idx(0)
[  114.210000] get_wdev_by_idx: invalid idx(0)
[  114.220000] pci probe count=2
root@LEDE:/# uname -r
4.4.50
root@LEDE:/# uname -a
Linux LEDE 4.4.50 #0 SMP Mon Feb 20 17:13:44 2017 mips GNU/Linux
root@LEDE:/#

MrFishchev, 2018-01-31:
I solved the problem, you need change in /etc/wireless/mt7615/*.dat files E2pAccessMode=2 and  add mt7615.eeprom.bin to /lib/firmware (I got this from different git with source of driver), then mtkwifi reload.

Nossiac, 2018-02-09:
@MrFishchev Thanks for your sharing.

I think a breaf intro on how to bring up the driver is necessary. I'm gonna post a draft later.