Nossiac/mtk-openwrt-feeds #24: Kernel 3.10.14

MrFishchev, 2018-02-27:
Hi @Nossiac, could you build the driver mt7615 on mt7621 platform for kernel 3.10.14 for me? I have not been able to build it for a very long time. Maybe I have a wrong sources but I have not other.
Thank you, nossiac.

Nossiac, 2018-02-27:
NP. I'll make it tomorrow.

Nossiac, 2018-02-28:
[http://nossiac.com/download/mtk-wifi-ko/mt7615-for-mt7621-linux-3.10.14.ko](http://nossiac.com/download/mtk-wifi-ko/mt7615-for-mt7621-linux-3.10.14.ko)

MrFishchev, 2018-02-28:
@Nossiac I tried to install module but I got error message: 

root@Witi:/# insmod /lib/modules/3.10.14-p112871/mt7615.ko
[  137.268000] mt7615e: module license 'Proprietary' taints kernel.
[  137.280000] Disabling lock debugging due to kernel taint
[  137.352000] mt7615e: Unknown symbol packet_source_delete_entry_hook (err 0)
[  137.364000] mt7615e: Unknown symbol wf_fwd_bpdu_halt_hook (err 0)
[  137.376000] mt7615e: Unknown symbol wf_fwd_probe_adapter (err 0)
[  137.388000] mt7615e: Unknown symbol wf_fwd_access_schedule_halt_hook (err 0)
[  137.404000] mt7615e: Unknown symbol wf_fwd_rx_hook (err 0)
[  137.416000] mt7615e: Unknown symbol wf_fwd_set_cb_num (err 0)
[  137.424000] mt7615e: Unknown symbol wf_fwd_pro_halt_hook (err 0)
[  137.436000] mt7615e: Unknown symbol wf_fwd_hijack_halt_hook (err 0)
[  137.452000] mt7615e: Unknown symbol wf_fwd_needed_hook (err 0)
[  137.464000] mt7615e: Unknown symbol wf_fwd_entry_insert_hook (err 0)
[  137.476000] mt7615e: Unknown symbol wf_fwd_delete_entry_inform_hook (err 0)
[  137.488000] mt7615e: Unknown symbol wf_fwd_get_rep_hook (err 0)
[  137.500000] mt7615e: Unknown symbol wf_fwd_check_device_hook (err 0)
[  137.512000] mt7615e: Unknown symbol wf_fwd_insert_repeater_mapping_hook (err 0)
[  137.528000] mt7615e: Unknown symbol wf_fwd_add_entry_inform_hook (err 0)
[  137.540000] mt7615e: Unknown symbol ppe_dev_register_hook (err 0)
[  137.556000] mt7615e: Unknown symbol wf_fwd_hijack_active_hook (err 0)
[  137.568000] mt7615e: Unknown symbol wf_fwd_show_entry_hook (err 0)
[  137.580000] mt7615e: Unknown symbol packet_source_show_entry_hook (err 0)
[  137.592000] mt7615e: Unknown symbol wf_fwd_feedback_map_table (err 0)
[  137.604000] mt7615e: Unknown symbol wf_fwd_pro_disabled_hook (err 0)
[  137.620000] mt7615e: Unknown symbol wf_fwd_entry_delete_hook (err 0)
[  137.632000] mt7615e: Unknown symbol wf_fwd_pro_active_hook (err 0)
[  137.644000] mt7615e: Unknown symbol wf_fwd_tx_hook (err 0)
[  137.656000] mt7615e: Unknown symbol wf_fwd_delete_entry_hook (err 0)
[  137.668000] mt7615e: Unknown symbol wf_fwd_bpdu_active_hook (err 0)
[  137.680000] mt7615e: Unknown symbol wf_fwd_access_schedule_active_hook (err 0)
[  137.696000] mt7615e: Unknown symbol wf_fwd_check_active_hook (err 0)
[  137.708000] mt7615e: Unknown symbol ppe_dev_unregister_hook (err 0)
kmod: failed to insert  /lib/modules/3.10.14-p112871/mt7615.ko

Nossiac, 2018-02-28:
seems you are not using an up to date sdk. I'll try to remove the wf_fwd code tmr...

Nossiac, 2018-03-02:
@MrFishchev  module updated, same link, md5sum = 6b959f33bf4a40c385c179a02b430203

MrFishchev, 2018-03-02:
@Nossiac Thanks, but it's not work for me, I got the same error. You said that I use not updated sdk, could you tell me, how can update it, please? And I'll try it.

LOG:
[  187.025975] mt7615e: module license 'Proprietary' taints kernel.
[  187.038067] Disabling lock debugging due to kernel taint
[  187.113807] mt7615e: Unknown symbol dev_kfree_skb_any (err 0)
[  187.125399] mt7615e: Unknown symbol __mips_clear_bit (err 0)
[  187.136769] mt7615e: Unknown symbol __mips_test_and_set_bit (err 0)
[  187.149483] mt7615e: Unknown symbol dev_kfree_skb_irq (err 0)
[  187.161026] mt7615e: Unknown symbol ra_sw_nat_hook_rx (err 0)
[  187.172531] mt7615e: Unknown symbol dev_get_drvdata (err 0)
[  187.183721] mt7615e: Unknown symbol __mips_set_bit (err 0)
[  187.194721] mt7615e: Unknown symbol ra_mtd_write_nm (err 0)
[  187.205888] mt7615e: Unknown symbol __mips_test_and_clear_bit (err 0)
[  187.218783] mt7615e: Unknown symbol mips_dma_map_ops (err 0)
[  187.230156] mt7615e: Unknown symbol ra_mtd_read_nm (err 0)
[  187.241214] mt7615e: Unknown symbol dev_set_drvdata (err 0)
[  187.252426] mt7615e: Unknown symbol ra_sw_nat_hook_tx (err 0)
[  187.264000] mt7615e: Unknown symbol procRegDir (err 0)
[  187.274310] mt7615e: Unknown symbol msecs_to_jiffies (err 0)
failed to insert mt7615-for-mt7621-linux-3.10.14.ko

Nossiac, 2018-03-02:
Similar errors were seen in earlier issues. those "mips" related symbols are from kernel, so I guess you are configuring the kernel improperly. Tell me your SDK version, and send me your kernel config, maybe I can tell what was wrong.

As for the latest SDK, it is protected by MediaTek's NDA, so I cannot help you here. You should consult MediaTek or its vendors.

MrFishchev, 2018-03-05:
I use kernel from https://github.com/mqmaker/witi-openwrt toolchain is  "toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2" 

Kernel Config:
[config.txt](https://github.com/Nossiac/mtk-openwrt-feeds/files/1779763/config.txt)

OpenWRT Config:
[config_openwrt.txt](https://github.com/Nossiac/mtk-openwrt-feeds/files/1779770/config_openwrt.txt)

Nossiac, 2018-03-05:
OK, that's a good point. I'll try it sometime.

MrFishchev, 2018-03-16:
@Nossiac Can you give me name or number of SDK, which you use for the issue?

closed 2018-07-13 by MrFishchev