Nossiac/mtk-openwrt-feeds #25: Kernel 4.14

Lu-Fi, 2018-02-28:
Can you please provide Kernel Modules also for 4.14 ?

Nossiac, 2018-03-02:
I've done the necessary patch to build these drivers for 4.14.
The automation server needs a few hours to build it. Maybe tmr.

Nossiac, 2018-03-04:
[http://nossiac.com/download/mtk-wifi-ko/](http://nossiac.com/download/mtk-wifi-ko/)

module for 4.14 updated, not verified yet.

BGFWDev, 2018-03-04:
Any plans for 4.14 for MT7623?

Nossiac, 2018-03-04:
@BGFWDev 

[http://nossiac.com/download/mtk-wifi-ko/mt7615-for-mt7623-linux-4.14.23.ko](http://nossiac.com/download/mtk-wifi-ko/mt7615-for-mt7623-linux-4.14.23.ko)

NOT verified.

closed 2018-03-12 by Nossiac

BGFWDev, 2018-05-16:
is there a MT7623+MT7615 file update? We are having issues getting this one to work. Or an MT7623+MT7612 .ko file?

frank-w, 2018-08-03:
can you post the patch for making driver for 4.4 can be compiled in 4.14?