Nossiac/mtk-openwrt-feeds #26: MT7615 - incorrect 5ghz calibration?

dissent1, 2018-02-28:
We have here a Xiaomi Mi router 3 Pro device here. The device includes 2xMT7615 cards, 1 per band.
Compiled in: 
- k4.4.116 (17.01 branch)
- mt7615 driver
- mtk luci plugin
- uci2dat

What we did:
- extracted eeprom per band from factory partition into /etc/wlan/*EEPROM{1,2}.bin respectively.
- copied stock SingleSKU.dat -> /etc/wireless/*sku.dat; but *sku-bf.dat is missing.
- copied mt7615e.eeprom.bin -> /lib/firmware/mt7615.eeprom.bin

With that setup 2.4ghz card works reasonably well, but 5ghz card seem to be severely miscalibrated and behaves on the border of reception.

@Nossiac what else may be missing? What else does the driver await?

PeLLmEn, 2018-02-28:
That's how it reboots:
`root@LEDE:/# mtkwifi reload
[ 2704.586323] WifiSysClose(), wdev idx = 0
[ 2704.593086] WifiSysGetBssInfoState(): BssInfoIdx 0 not found!!!
[ 2704.599020] WifiSysUpdateBssInfoState(): BssInfoIdx 0 not found!!!
[ 2704.606821] br-lan: port 2(ra0) entered disabled state
[ 2704.662356] device ra0 left promiscuous mode
[ 2704.666828] br-lan: port 2(ra0) entered disabled state
[ 2704.686495] WifiSysClose(), wdev idx = 0
[ 2704.693485] WifiSysGetBssInfoState(): BssInfoIdx 0 not found!!!
[ 2704.699408] WifiSysUpdateBssInfoState(): BssInfoIdx 0 not found!!!
[ 2704.777963] WifiSysClose(), wdev idx = 1
[ 2704.784732] WifiSysGetBssInfoState(): BssInfoIdx 1 not found!!!
[ 2704.790659] WifiSysUpdateBssInfoState(): BssInfoIdx 1 not found!!!
[ 2704.798356] br-lan: port 3(rax0) entered disabled state
[ 2704.852900] device rax0 left promiscuous mode
[ 2704.857427] br-lan: port 3(rax0) entered disabled state
[ 2704.891176] (ApCliCtrlDisconnectReqAction) 2. Before do ApCliLinkDown. ifIndex = 0, bValid = 0
[ 2704.993121] WifiSysClose(), wdev idx = 2
[ 2704.998047] br-lan: port 4(apcli0) entered disabled state
[ 2705.052953] device apcli0 left promiscuous mode
[ 2705.057692] br-lan: port 4(apcli0) entered disabled state
[ 2705.077570] (ApCliCtrlDisconnectReqAction) 2. Before do ApCliLinkDown. ifIndex = 1, bValid = 0
[ 2705.177636] WifiSysClose(), wdev idx = 3
[ 2705.181842] MtAsicSetPiggyBack(802): Not support for HIF_MT yet!
[ 2705.187954] RTMPSetLEDStatus: before AndesLedEnhanceOP , status=0, LED_CMD=0!
[ 2705.195096] AndesLedEnhanceOP: Success!
[ 2705.198973] WifiSysClose(), wdev idx = 0
[ 2705.202986] MtAsicSetPiggyBack(802): Not support for HIF_MT yet!
[ 2705.209001] RTMPSetLEDStatus: before AndesLedEnhanceOP , status=0, LED_CMD=0!
[ 2705.216158] AndesLedEnhanceOP: Success!
[ 2705.219980] WifiSysClose(), wdev idx = 1
[ 2705.225528] kill LoopBackTxTask task failed!
[ 2705.229921] RTMPSetLEDStatus: before AndesLedEnhanceOP , status=4, LED_CMD=1!
[ 2705.237110] AndesLedEnhanceOP: Success!
[ 2705.246009] AndesRestartCheck: Current TOP_MISC2(0x7)
[ 2705.292485] CmdReStartDLRsp: Status Success!, Status(0)
[ 2705.297735] EventExtEventHandler: Unknown Ext Event(6f)
[ 2705.297745] AndesRestartCheck:  TOP_MISC2(1)
[ 2705.297755] RTMPSetLEDStatus: before AndesLedEnhanceOP , status=2, LED_CMD=1!
[ 2705.297784] AndesLedEnhanceOP: Success!
[ 2705.318422] RT28xxPciAsicRadioOff(): Not support for HIF_MT yet!
[ 2705.324523] tx_kickout_fail_count = 0
[ 2705.328216] tx_timeout_fail_count = 0
[ 2705.331859] rx_receive_fail_count = 0
[ 2705.335526] alloc_cmd_msg = 2197
[ 2705.338742] free_cmd_msg = 2197
[ 2705.343793] cut_through_token_list_destroy(): 8dfe1688,8dfe1688
[ 2705.350666] cut_through_token_list_destroy(): 8dfe1698,8dfe1698
[ 2705.362125] FwOwn()::Set Fw Own
[ 2705.369314] RTMP_AllTimerListRelease: Size=0
[ 2705.373884] FwOwn()::Return since already in Fw Own...
[ 2705.379076] <---HwCtrlThread
[ 2705.382941] br-lan: port 5(apclix0) entered disabled state
[ 2705.438419] device apclix0 left promiscuous mode
[ 2705.443187] br-lan: port 5(apclix0) entered disabled state
[ 2705.462928] WifiSysClose(), wdev idx = 1
[ 2705.467615] MtAsicSetPiggyBack(802): Not support for HIF_MT yet!
[ 2705.473635] RTMPSetLEDStatus: before AndesLedEnhanceOP , status=0, LED_CMD=0!
[ 2705.480861] AndesLedEnhanceOP: Success!
[ 2705.493313] MtAsicSetPiggyBack(802): Not support for HIF_MT yet!
[ 2705.499353] RTMPSetLEDStatus: before AndesLedEnhanceOP , status=0, LED_CMD=0!
[ 2705.506510] AndesLedEnhanceOP: Success!
[ 2705.510346] WifiSysClose(), wdev idx = 0
[ 2705.515860] kill LoopBackTxTask task failed!
[ 2705.520237] RTMPSetLEDStatus: before AndesLedEnhanceOP , status=4, LED_CMD=1!
[ 2705.527389] AndesLedEnhanceOP: Success!
[ 2705.536244] AndesRestartCheck: Current TOP_MISC2(0x7)
[ 2705.582936] CmdReStartDLRsp: Status Success!, Status(0)
[ 2705.588195] AndesRestartCheck:  TOP_MISC2(1)
[... 522 lines cut ...]
[ 2711.795892] HcUpdatePhyMode(): Update PhyMode for all wdev for this band PhyMode:48,Channel=36
[ 2711.804594] CountryCode(2.4G/5G)=5/7, RFIC=25, PHY mode(2.4G/5G)=14/48, support 38 channels
[ 2711.865552] device apcli0 entered promiscuous mode
[ 2711.870478] br-lan: port 4(apcli0) entered forwarding state
[ 2711.876133] br-lan: port 4(apcli0) entered forwarding state
[ 2711.897468] WifiSysOpen(), wdev idx = 3
[ 2711.901601] MtCmdSetDbdcCtrl:(ret = 0)
[ 2711.905996] HcUpdatePhyMode(): Update PhyMode for all wdev for this band PhyMode:14,Channel=6
[ 2711.914482] HcUpdatePhyMode(): Update PhyMode for all wdev for this band PhyMode:14,Channel=6
[ 2711.923198] CountryCode(2.4G/5G)=5/7, RFIC=25, PHY mode(2.4G/5G)=14/48, support 38 channels
[ 2711.983424] device apclix0 entered promiscuous mode
[ 2711.988493] br-lan: port 5(apclix0) entered forwarding state
[ 2711.994197] br-lan: port 5(apclix0) entered forwarding state
[ 2712.015681] WifiSysOpen(), wdev idx = 1
[ 2712.020124] MtCmdSetDbdcCtrl:(ret = 0)
[ 2712.024777] HcUpdatePhyMode(): Update PhyMode for all wdev for this band PhyMode:14,Channel=6
[ 2712.033955] HcUpdatePhyMode(): Update PhyMode for all wdev for this band PhyMode:14,Channel=6
[ 2712.042675] CountryCode(2.4G/5G)=5/7, RFIC=25, PHY mode(2.4G/5G)=14/1, support 38 channels
[ 2712.103542] device apclii0 entered promiscuous mode
[ 2712.108629] br-lan: port 6(apclii0) entered forwarding state
[ 2712.114320] br-lan: port 6(apclii0) entered forwarding state
root@LEDE:/# [ 2713.735083] br-lan: port 3(rax0) entered forwarding state
[ 2713.875091] br-lan: port 4(apcli0) entered forwarding state
[ 2713.985072] br-lan: port 5(apclix0) entered forwarding state
[ 2714.105074] br-lan: port 6(apclii0) entered forwarding state`

dissent1, 2018-02-28:
Must add that absence or presence of /lib/firmware/mt7615.eeprom.bin doesn't have any effect on behavior.

PeLLmEn, 2018-02-28:
Here's the log: 

```
root@LEDE:/# mtkwifi reload
[ 2704.586323] WifiSysClose(), wdev idx = 0
[ 2704.593086] WifiSysGetBssInfoState(): BssInfoIdx 0 not found!!!
[ 2704.599020] WifiSysUpdateBssInfoState(): BssInfoIdx 0 not found!!!
[ 2704.606821] br-lan: port 2(ra0) entered disabled state
[ 2704.662356] device ra0 left promiscuous mode
[ 2704.666828] br-lan: port 2(ra0) entered disabled state
[ 2704.686495] WifiSysClose(), wdev idx = 0
[ 2704.693485] WifiSysGetBssInfoState(): BssInfoIdx 0 not found!!!
[ 2704.699408] WifiSysUpdateBssInfoState(): BssInfoIdx 0 not found!!!
[ 2704.777963] WifiSysClose(), wdev idx = 1
[ 2704.784732] WifiSysGetBssInfoState(): BssInfoIdx 1 not found!!!
[ 2704.790659] WifiSysUpdateBssInfoState(): BssInfoIdx 1 not found!!!
[ 2704.798356] br-lan: port 3(rax0) entered disabled state
[ 2704.852900] device rax0 left promiscuous mode
[ 2704.857427] br-lan: port 3(rax0) entered disabled state
[ 2704.891176] (ApCliCtrlDisconnectReqAction) 2. Before do ApCliLinkDown. ifIndex = 0, bValid = 0
[ 2704.993121] WifiSysClose(), wdev idx = 2
[ 2704.998047] br-lan: port 4(apcli0) entered disabled state
[ 2705.052953] device apcli0 left promiscuous mode
[ 2705.057692] br-lan: port 4(apcli0) entered disabled state
[ 2705.077570] (ApCliCtrlDisconnectReqAction) 2. Before do ApCliLinkDown. ifIndex = 1, bValid = 0
[ 2705.177636] WifiSysClose(), wdev idx = 3
[ 2705.181842] MtAsicSetPiggyBack(802): Not support for HIF_MT yet!
[ 2705.187954] RTMPSetLEDStatus: before AndesLedEnhanceOP , status=0, LED_CMD=0!
[ 2705.195096] AndesLedEnhanceOP: Success!
[ 2705.198973] WifiSysClose(), wdev idx = 0
[ 2705.202986] MtAsicSetPiggyBack(802): Not support for HIF_MT yet!
[ 2705.209001] RTMPSetLEDStatus: before AndesLedEnhanceOP , status=0, LED_CMD=0!
[ 2705.216158] AndesLedEnhanceOP: Success!
[ 2705.219980] WifiSysClose(), wdev idx = 1
[ 2705.225528] kill LoopBackTxTask task failed!
[ 2705.229921] RTMPSetLEDStatus: before AndesLedEnhanceOP , status=4, LED_CMD=1!
[ 2705.237110] AndesLedEnhanceOP: Success!
[ 2705.246009] AndesRestartCheck: Current TOP_MISC2(0x7)
[ 2705.292485] CmdReStartDLRsp: Status Success!, Status(0)
[ 2705.297735] EventExtEventHandler: Unknown Ext Event(6f)
[ 2705.297745] AndesRestartCheck:  TOP_MISC2(1)
[ 2705.297755] RTMPSetLEDStatus: before AndesLedEnhanceOP , status=2, LED_CMD=1!
[ 2705.297784] AndesLedEnhanceOP: Success!
[ 2705.318422] RT28xxPciAsicRadioOff(): Not support for HIF_MT yet!
[ 2705.324523] tx_kickout_fail_count = 0
[ 2705.328216] tx_timeout_fail_count = 0
[ 2705.331859] rx_receive_fail_count = 0
[ 2705.335526] alloc_cmd_msg = 2197
[ 2705.338742] free_cmd_msg = 2197
[ 2705.343793] cut_through_token_list_destroy(): 8dfe1688,8dfe1688
[ 2705.350666] cut_through_token_list_destroy(): 8dfe1698,8dfe1698
[ 2705.362125] FwOwn()::Set Fw Own
[ 2705.369314] RTMP_AllTimerListRelease: Size=0
[ 2705.373884] FwOwn()::Return since already in Fw Own...
[ 2705.379076] <---HwCtrlThread
[ 2705.382941] br-lan: port 5(apclix0) entered disabled state
[ 2705.438419] device apclix0 left promiscuous mode
[ 2705.443187] br-lan: port 5(apclix0) entered disabled state
[ 2705.462928] WifiSysClose(), wdev idx = 1
[ 2705.467615] MtAsicSetPiggyBack(802): Not support for HIF_MT yet!
[ 2705.473635] RTMPSetLEDStatus: before AndesLedEnhanceOP , status=0, LED_CMD=0!
[ 2705.480861] AndesLedEnhanceOP: Success!
[ 2705.493313] MtAsicSetPiggyBack(802): Not support for HIF_MT yet!
[ 2705.499353] RTMPSetLEDStatus: before AndesLedEnhanceOP , status=0, LED_CMD=0!
[ 2705.506510] AndesLedEnhanceOP: Success!
[ 2705.510346] WifiSysClose(), wdev idx = 0
[ 2705.515860] kill LoopBackTxTask task failed!
[ 2705.520237] RTMPSetLEDStatus: before AndesLedEnhanceOP , status=4, LED_CMD=1!
[ 2705.527389] AndesLedEnhanceOP: Success!
[ 2705.536244] AndesRestartCheck: Current TOP_MISC2(0x7)
[... 525 lines cut ...]
[ 2711.804594] CountryCode(2.4G/5G)=5/7, RFIC=25, PHY mode(2.4G/5G)=14/48, support 38 channels
[ 2711.865552] device apcli0 entered promiscuous mode
[ 2711.870478] br-lan: port 4(apcli0) entered forwarding state
[ 2711.876133] br-lan: port 4(apcli0) entered forwarding state
[ 2711.897468] WifiSysOpen(), wdev idx = 3
[ 2711.901601] MtCmdSetDbdcCtrl:(ret = 0)
[ 2711.905996] HcUpdatePhyMode(): Update PhyMode for all wdev for this band PhyMode:14,Channel=6
[ 2711.914482] HcUpdatePhyMode(): Update PhyMode for all wdev for this band PhyMode:14,Channel=6
[ 2711.923198] CountryCode(2.4G/5G)=5/7, RFIC=25, PHY mode(2.4G/5G)=14/48, support 38 channels
[ 2711.983424] device apclix0 entered promiscuous mode
[ 2711.988493] br-lan: port 5(apclix0) entered forwarding state
[ 2711.994197] br-lan: port 5(apclix0) entered forwarding state
[ 2712.015681] WifiSysOpen(), wdev idx = 1
[ 2712.020124] MtCmdSetDbdcCtrl:(ret = 0)
[ 2712.024777] HcUpdatePhyMode(): Update PhyMode for all wdev for this band PhyMode:14,Channel=6
[ 2712.033955] HcUpdatePhyMode(): Update PhyMode for all wdev for this band PhyMode:14,Channel=6
[ 2712.042675] CountryCode(2.4G/5G)=5/7, RFIC=25, PHY mode(2.4G/5G)=14/1, support 38 channels
[ 2712.103542] device apclii0 entered promiscuous mode
[ 2712.108629] br-lan: port 6(apclii0) entered forwarding state
[ 2712.114320] br-lan: port 6(apclii0) entered forwarding state
root@LEDE:/# [ 2713.735083] br-lan: port 3(rax0) entered forwarding state
[ 2713.875091] br-lan: port 4(apcli0) entered forwarding state
[ 2713.985072] br-lan: port 5(apclix0) entered forwarding state
[ 2714.105074] br-lan: port 6(apclii0) entered forwarding state
```

PeLLmEn, 2018-03-17:
Any progress???

ZiqiangYu, 2018-03-18:
the issue seems is that 2G and 5G calibration are different,but the 7615 wifi driver just get  /lib/firmware/mt7615.eeprom.bin, i will fix it so, and i will tell you the correct calibration data file names.

PeLLmEn, 2018-03-19:
@ZiqiangYu thanks, we're waiting for your solution.

ZiqiangYu, 2018-03-24:
e2p files in /lib/firmware/ like "mt7615.1.eeprom.bin  mt7615.2.eeprom.bin  mt7615.3.eeprom.bin".
mt7615.1.eeprom.bin for ra0,the first card
mt7615.2.eeprom.bin for rai0,the second card,
mt7615.3.eeprom.bin for the third card

normally,the first card is 2G,the second card is 5G

BGFWDev, 2018-04-18:
> it's a little complicated for dual-band chips like mt7615.
>
> as @ZiqiangYu <https://github.com/ZiqiangYu> stated, the 7615 driver can be working on either 2g or 5g band, depending on its profile (.dat) and eeprom (.bin). so if you got more than 1 mt7615 chips on your set, you will need 1 profile and 1 eeprom binary for every device/band.
>
> for example, you got 2 mt7615, one works on 2G band, the other works on 5G band, then you will need:
>
> /etc/wireless/mt7615/mt7615.1.dat
> /etc/wireless/mt7615/mt7615.2.dat
> /lib/firmware/mt7615.1.eeprom.bin
> /lib/firmware/mt7615.2.eeprom.bin
>
> when the 1st 7615 chip init, the driver will looking for /etc/wireless/mt7615/mt7615.1.dat and /lib/firmware/mt7615.1.eeprom.bin. it is the same for the 2nd 7615.
>
> if you missed these files (or wrong path), there will be errors in the console/dmesg when bringing up wifi interfaces, and wifi may work improperly.

What about  the MT7615D chipset support, with independent dual band?