Nossiac/mtk-openwrt-feeds #28: mt7612 crashes System.

Lu-Fi, 2018-02-28:
System is build from GIT using Kernel 4.4.
System freezes while loading "mt7612" Kernel Module.
Occurs only if "rai0" is bound to an Interface.
Sry. No log. System is crashing immediately.

Xiaomi Mi 3G

Nossiac, 2018-03-02:
I just found a Mi 3G, I'm gonna reproduce the bug with it next week.

Nossiac, 2018-03-16:
May this help?

https://github.com/Nossiac/mtk-openwrt-feeds/wiki 

> 
> So, disable/remove the following modules/packages first.
> 
> As stated above, you'd better disable/remove the following modules/packages before using MediaTek drivers, otherwise your system may run into various bugs, even kernel crashes.
> 
>     rt2x00
>     mt76
>     cfg80211
>     mac80211
>     wpad
>     supplicant