Nossiac/mtk-openwrt-feeds #31: Mt7628: Kernel Panic when set up it.

kysonlok, 2018-03-08:
root@LEDE:/# [  209.466592] Unhandled kernel unaligned access[#1]:
[  209.471477] CPU: 0 PID: 2092 Comm: nmbd Tainted: P                4.4.92 #0
[  209.478540] task: 87cce338 ti: 86d42000 task.ti: 86d42000
[  209.484012] $ 0   : 00000000 7fbc7888 00000001 00000010
[  209.489332] $ 4   : fffffdc9 80360000 00000010 0000082c
[  209.494641] $ 8   : 00000000 00000000 00000000 00000000
[  209.499951] $12   : 00000000 ffffff80 00000000 00000000
[  209.505260] $16   : fffffdc9 fffffdc9 0000082c 803a4820
[  209.510570] $20   : 00000010 80390000 00000000 00000000
[  209.515883] $24   : 00000000 00000000                  
[  209.521191] $28   : 86d42000 86d43be8 00000000 801f7324
[  209.526500] Hi    : 00000000
[  209.529417] Lo    : 000033a6
[  209.532371] epc   : 801f7348 netdev_master_upper_dev_get+0x38/0x70
[  209.538643] ra    : 801f7324 netdev_master_upper_dev_get+0x14/0x70
[  209.544904] Status: 1100e403	KERNEL EXL IE 
[  209.549158] Cause : 00800010 (ExcCode 04)
[  209.553220] BadVA : fffffe2d
[  209.556139] PrId  : 00019655 (MIPS 24KEc)
[  209.560200] Modules linked in: mt7628(P) pppoe ppp_async option l2tp_ppp iptable_nat uvcvideo usb_wwan sierra_net sierra rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_nat_ipv4 nf_nat_amanda nf_conntrack_pptp nf_conntrack_ipv6 nf_conntrack_ipv4 nf_conntrack_amanda ipt_REJECT ipt_MASQUERADE huawei_cdc_ncm gspca_zc3xx gspca_sonixj gspca_sonixb gspca_main cp210x cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY videobuf2_v4l2 usbserial usbnet ts_kmp ts_fsm ts_bm slhc nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_nat_h323 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast iptable_mangle iptable_filter ipt_ECN ipheth ip_tables input_core crc_ccitt cdc_wdm cdc_acm fuse act_connmark nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress videobuf2_vmalloc videobuf2_memops videobuf2_core v4l2_common videodev i2c_dev i2c_core ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip_gre gre ifb l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel ip_tunnel tun vfat fat ntfs nls_utf8 nls_iso8859_1 nls_cp437 dma_shared_buffer sha1_generic ecb usb_storage leds_gpio uhci_hcd ohci_platform ohci_hcd ehci_platform ehci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache exfat usbcore nls_base usb_common crc16 mii cryptomgr aead crypto_null crc32c_generic crypto_hash
[  209.755188] Process nmbd (pid: 2092, threadinfo=86d42000, task=87cce338, tls=77a1bd48)
[  209.763207] Stack : 00000000 00000000 00000000 86d66900 86d66900 8020bde4 801ebe88 000020c0
	  879d4640 00000002 00000010 00000002 80390000 803a54a8 00000000 00010002
	  000003e8 000005dc 00000000 00000000 00000001 00000002 00000007 00000000
	  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
	  803360d8 80330000 86d66900 00000002 020152c0 87c02780 801ebe88 000020c0
	  ...
[  209.799321] Call Trace:
[  209.801814] [<801f7348>] netdev_master_upper_dev_get+0x38/0x70
[  209.807742] [<8020bde4>] rtnl_fill_ifinfo+0x58/0x9b0
[  209.812780] [<8020c864>] rtnl_dump_ifinfo+0x128/0x1d8
[  209.817910] [<802250a0>] netlink_dump+0x108/0x294
[  209.822683] [<8022540c>] netlink_recvmsg+0x1e0/0x36c
[  209.827732] [<801e42e8>] SyS_recvfrom+0xac/0x120
[  209.832426] [<8000686c>] syscall_common+0x30/0x54
[  209.837191] 
[  209.838698] 
Code: 2484f83c  0c0528f2  00000000 <8e020064> 26100064  10500006  00000000  9043fff8  10600003 
[  209.848907] ---[ end trace 1ea2de28224f5cd2 ]---
[  209.857655] Fatal exception: panic in 5 seconds 
root@LEDE:/# 
[  214.865794] Kernel panic - not syncing: Fatal exception
[  214.873780] Rebooting in 3 seconds..

Nossiac, 2018-03-08:
similar issues was reported before. but I was not able to reproduce it. can u please post the build config here?

kysonlok, 2018-03-08:
@Nossiac Thanks, I try to clean the buildroot and rebuild, it works now.

kysonlok, 2018-03-08:
This build config will crash.
[.config.zip](https://github.com/Nossiac/mtk-openwrt-feeds/files/1792493/default.config.zip)

Nossiac, 2018-03-26:
I cannot find a "gl-mt300n-v2" router to verify this. But I tested with 3 other models of mt7628, still cannot reproduce this issue.

The call stack was from netlink, which seems irrelevant with mt7628.ko. I guess there must be some applications/serviers that tries to issue a netlink command, which may confuse or crash the module.

You can follow the wiki here, remove suspicious packages first, then have another try. [https://github.com/Nossiac/mtk-openwrt-feeds/wiki](https://github.com/Nossiac/mtk-openwrt-feeds/wiki)

kysonlok, 2018-03-31:
The "gl-mt300n-v2" profile is available on master branch. Yes, if I use default config and only select mt7628 from this feed, it works. But using the config I mention above and it will crash. It is strange that when I use the mt7628 driver's source code to build, even if selecting cfg80211/mac80211/wpad/supplicant/iw package, it won't crash.

josephmarlin, 2018-04-09:
`Tue Oct 17 13:48:41 2017 kern.warn kernel: [   78.965749] mt7628: module license 'unspecified' taints kernel.
Tue Oct 17 13:48:41 2017 kern.warn kernel: [   78.971769] Disabling lock debugging due to kernel taint
Tue Oct 17 13:48:41 2017 kern.warn kernel: [   79.037654] 
Tue Oct 17 13:48:41 2017 kern.warn kernel: [   79.037654] 
Tue Oct 17 13:48:41 2017 kern.warn kernel: [   79.037654] === pAd = c09cd000, size = 1292832 ===
Tue Oct 17 13:48:41 2017 kern.warn kernel: [   79.037654] 
Tue Oct 17 13:48:41 2017 kern.warn kernel: [   79.047213] <-- RTMPAllocTxRxRingMemory, Status=0, ErrorValue=0x
Tue Oct 17 13:48:41 2017 kern.warn kernel: [   79.054538] <-- RTMPAllocAdapterBlock, Status=0
Tue Oct 17 13:48:41 2017 kern.warn kernel: [   79.059181] RtmpChipOpsHook(492): Not support for HIF_MT yet!
Tue Oct 17 13:48:41 2017 kern.warn kernel: [   79.065005] mt7628_init()-->
Tue Oct 17 13:48:41 2017 kern.warn kernel: [   79.067951] mt7628_init(FW(8a00), HW(8a01), CHIPID(7628))
Tue Oct 17 13:48:41 2017 kern.warn kernel: [   79.073426] e2.bin mt7628_init(1135)::(2), pChipCap->fw_len(63888)
Tue Oct 17 13:48:41 2017 kern.warn kernel: [   79.079705] mt_bcn_buf_init(218): Not support for HIF_MT yet!
Tue Oct 17 13:48:41 2017 kern.warn kernel: [   79.085529] <--mt7628_init()
Tue Oct 17 13:48:42 2017 kern.warn kernel: [   79.766116] Unhandled kernel unaligned access[#1]:
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   79.771004] CPU: 0 PID: 1821 Comm: luci Tainted: P                4.4.92 #0
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   79.778068] task: 87da5e30 ti: 8618e000 task.ti: 8618e000
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   79.783539] $ 0   : 00000000 7ffaf380 00000001 00000009
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   79.788857] $ 4   : fffffdc9 80360000 00000010 0000071d
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   79.794168] $ 8   : 00000000 00000000 00000000 00000000
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   79.799476] $12   : 00000000 00000000 00000000 00000000
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   79.804787] $16   : fffffdc9 fffffdc9 0000071d 803a4820
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   79.810096] $20   : 00000010 80390000 00000000 00000000
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   79.815408] $24   : 00000000 00000000                  
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   79.820718] $28   : 8618e000 8618fbe8 00000000 801f6d04
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   79.826025] Hi    : 00000000
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   79.828942] Lo    : 00001ee3
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   79.831896] epc   : 801f6d28 netdev_master_upper_dev_get+0x38/0x70
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   79.838167] ra    : 801f6d04 netdev_master_upper_dev_get+0x14/0x70
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   79.844427] Status: 1100a403	KERNEL EXL IE 
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   79.848683] Cause : 00800010 (ExcCode 04)
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   79.852744] BadVA : fffffe2d
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   79.855664] PrId  : 00019655 (MIPS 24KEc)
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   79.859723] Modules linked in: mt7628(P) pppoe ppp_async option iptable_nat usb_wwan sierra_net sierra rndis_host qmi_wwan pppox ppp_mppe ppp_generic nf_nat_pptp nf_nat_ipv4 nf_nat_amanda nf_conntrack_pptp nf_conntrack_ipv6 nf_conntrack_ipv4 nf_conntrack_amanda ipt_REJECT ipt_MASQUERADE huawei_cdc_ncm cp210x cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn Tue Oct 17 13:48:43 2017 kern.warn kernel: [   80.024405] Process luci (pid: 1821, threadinfo=8618e000, task=87da5e30, tls=77fe7d48)
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   80.032424] Stack : 00000000 00000000 00000000 87c1da80 87c1da80 8020b7c4 801eb868 000020c0
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   80.032424] 	  87922a00 00000002 00000010 00000002 80390000 803a54a8 00000000 00010002
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   80.032424] 	  000003e8 000005dc 00000000 00000000 00000001 00000002 00000005 00000000
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   80.032424] 	  00000000 00000000 00000000 00000000 b0300000 00000000 00000006 00000000
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   80.032424] 	  80335058 80330000 87c1da80 00000002 020152c0 87c02780 801eb868 000020c0
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   80.032424] 	  ...
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   80.068534] Call Trace:
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   80.071024] [<801f6d28>] netdev_master_upper_dev_get+0x38/0x70
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   80.076952] [<8020b7c4>] rtnl_fill_ifinfo+0x58/0x9b0
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   80.081990] [<8020c244>] rtnl_dump_ifinfo+0x128/0x1d8
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   80.087121] [<80224a80>] netlink_dump+0x108/0x294
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   80.091892] [<80224dec>] netlink_recvmsg+0x1e0/0x36c
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   80.096942] [<801e3cc8>] SyS_recvfrom+0xac/0x120
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   80.101639] [<8000686c>] syscall_common+0x30/0x54
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   80.106404] 
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   80.107911] 
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   80.107911] Code: 2484e780  0c0527cc  00000000 <8e020064> 26100064  10500006  00000000  9043fff8  10600003 
Tue Oct 17 13:48:43 2017 kern.warn kernel: [   80.118258] ---[ end trace 22c4f56c84c0c06f ]---
`

Same problem, same device. (GL-iNet's GL-MT300N-V2)

@kysonlok, were you ever able to resolve this panic?

josephmarlin, 2018-04-09:
Interestingly, this panic happens mere moments after inserting the mt7628 with insmod.

josephmarlin, 2018-04-09:
Sorry for the spam - looks like something on the LuCI overview page is calling something that causes the crash. The system is stable-ish until I load the webpage.

kysonlok, 2018-04-10:
@josephmarlin I don't know how to resolve this, it is strange for me. I build it via source code. You can refer to https://github.com/dongqifan/lede-mt7628/tree/dcom/package/mtk/mt7628. Note that you should know MTK's proprietary license before you use it.

jinglei207, 2020-06-22:
does anyone have solved this problem? thanks.