Nossiac/mtk-openwrt-feeds #34: Krack attacks

pparent76, 2018-03-12:
Am I right to think that this driver is vulnerable to Krack attacks and will always be?

maurerr, 2018-03-15:
only in repeater/client mode
AP should be safe

pparent76, 2018-03-15:
Thank's a lot!

Can you explain how can it be safe in AP mode, when it was written before krack attacks were revealed?

Nossiac, 2018-03-16:
Mediatek has released patches to its vendors to protect against krack attacks.
Not merged here yet... busy days...

maurerr, 2018-03-16:
read the KRACK documentation - it's not AP related but client.
It's useless to patch your AP if you still use client OS (Android/IOS) unpatched

Nossiac, 2018-03-16:
Hmm, as I learned, AP still got the chance to detect suspicious attacks and drop the athentication session immediately.

pparent76, 2018-03-16:
Yes as far I as understand AP and client can and shoud be patched.
for exemple hostapd has been patched.