Nossiac/mtk-openwrt-feeds #35: mt7628.ko build but with zero bytes size

docbee63, 2018-03-12:
When I do the steps in adding the feed, selecting mt7628 as additional driver via menuconfig and then run a "make" just a zero byte "mt7628.ko" is built. Make runs without any errors. I work with kernel 4.9.77. Any idea why the .ko file is not built or how to debug this?

Werner1992, 2018-03-13:
The problem is.
The subtarget in lede is named mt76x8.
The subtarget in the makefile is named mt7628.
The file was not found.

Change the makefile:
/feeds/mtk/drivers/mt7628/Makefile.
And search PKG_KO:
And change it to this.
PKG_KO:=mt7628-for-mt7628-linux-4.9.*.ko

You must disabled all wifi driver, because you have a kernel panic by load mt7628.ko

docbee63, 2018-03-13:
Thanks, that did the trick for compilation.
I can "insmod" the new mt7628.ko fine but when I try to activate the wifi by "ifconfig ra0 up" I get this error message:
`ERROR!!! 
RTMPReadParametersHook failed, Status[=0x00000001]`

Any idea what I do still miss?

closed 2018-03-13 by docbee63

docbee63, 2018-03-13:
one step further... adding "/etc/wireless/mt7628/mt7628.dat" did remove the error.
Now "ifconfig ra0 up" returns
`MCUSysInit: NICLoadFirmware failed, Status[=0x00000001]`

Do I have to put a NIC firmware file somewhere as well?

sumitmamoria, 2023-01-25:
> one step further... adding "/etc/wireless/mt7628/mt7628.dat" did remove the error. Now "ifconfig ra0 up" returns `MCUSysInit: NICLoadFirmware failed, Status[=0x00000001]`
> 
> Do I have to put a NIC firmware file somewhere as well?

Did you mange to solve this? I am getting the same error.