Nossiac/mtk-openwrt-feeds #37: Can't get "mt7628.ko" working in STA mode

docbee63, 2018-03-14:
So far I had no success to have the mt7628 operating as AP client. Is the mt7628.ko compiled in a way that supports having it operation as a client to an exiting WLAN only? If so, is there an example for a working "mt7628.dat" and "/etc/config/network" and "/etc/config/wireless"?

I tried as how I do with lede on other platforms (where lede is natively supporting WLAN), but no success so far.

Nossiac, 2018-03-16:
It has nothing to do with "/etc/config/network" and "/etc/config/wireless". The best practice of configuring apcli0 interface is to use "iwpriv" command, and using mt7628.dat as a database only.

Here's an example:

```
iwpriv apcli0 set ApCliEnable=0          // disable client mode first
iwpriv apcli0 set ApCliAuthMode=WPA2PSK  // root ap authentication mode
iwpriv apcli0 set ApCliEncrypType=AES    // root ap encryption type
iwpriv apcli0 set ApCliSsid=HelloHi      // root ap ssid
iwpriv apcli0 set ApCliWPAPSK=12345678   // root ap password
iwpriv apcli0 set ApCliSsid=HelloHi      // root ap ssid, again...
iwpriv apcli0 set ApCliEnable=1          // enable client mode
```

docbee63, 2018-03-21:
thanks that solved the issue!

closed 2018-03-21 by docbee63