Nossiac/mtk-openwrt-feeds #38: mt7612 kernel panic

maurerr, 2018-03-15:
i've built latest patched lede-17.01 with kernel 4.4.116 for my WITI device  - mt7602+mt7612
when doing modprobe mt7612 i get no interface - just 3 lines in dmesg:
[  652.201213] mt7612: module license 'MediaTek' taints kernel.
[  652.212528] Disabling lock debugging due to kernel taint
[  652.257789] register rt2860

if i try to unload current mt76 driver 
root@LEDE:~# rmmod mt76x2e # then
root@LEDE:~# modprobe mt7612

i get some dmesg before a system reboot:
https://pastebin.com/wkJ5aWvB

please help !
and a big thanks for your work

ZiqiangYu, 2018-03-15:
did you built pcie driver ?

maurerr, 2018-03-15:
don't think so
anyway here's the config https://pastebin.com/70NL3er1

ZiqiangYu, 2018-03-16:
post all your kernel start log,if kernel find pcie card ,it will show.

Nossiac, 2018-03-16:
May this help?

https://github.com/Nossiac/mtk-openwrt-feeds/wiki 

> 
> So, disable/remove the following modules/packages first.
> 
> As stated above, you'd better disable/remove the following modules/packages before using MediaTek drivers, otherwise your system may run into various bugs, even kernel crashes.
> 
>     rt2x00
>     mt76
>     cfg80211
>     mac80211
>     wpad
>     supplicant

maurerr, 2018-03-16:
thanks for the reply.
I've removed the offending packages but still get the crash
https://pastebin.com/zShqDGei