Nossiac/mtk-openwrt-feeds #41: MAC address nulled

docbee63, 2018-03-21:
With the friendly help from this feed I got WiFi basically working on a MR3020V3. 
Only thing the precompiled mt7628.ko does not read out correctly is the MAC.

`apcli0    Link encap:Ethernet  HWaddr 02:00:00:00:00:00  
          inet addr:192.168.123.250  Bcast:192.168.123.255  Mask:255.255.255.0
          inet6 addr: fe80::ff:fe00:0/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ra0       Link encap:Ethernet  HWaddr 00:00:00:00:00:00  
          inet6 addr: fe80::200:ff:fe00:0/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:6`

I tried setting MAC manually with "ifconfig apcli0 hw ether 12:23:34:45:56:6"7 but only get a "SIOCSIFHWADDR: Not supported" error. Adding MAC as an option to /etc/config/network" gets also ignored. 

Any idea how I can change the MAC to the one "eth0" is showing?

docbee63, 2018-03-21:
I also tried with "macchanger". As expected also no avail :-(
"[ERROR] Could not change MAC: interface up or insufficient permissions: Not supported"

Can the mt7628.ko somehow be redirected where on the mtd devices to read the MAC from? Or can I give the mac as a parameter when installing the kernel module? As I don't have the mt7628 sources, I can't find out myself :-(

ZiqiangYu, 2018-03-24:
did you have /lib/firmware/mt7628.eeprom.bin in your board? it is the hw parameter. and mac is in it..

docbee63, 2018-03-25:
Thanks for the suggestion. The vocore system if have does not have a "/lib/firmware/*.bin" file and there the mt7628 assigns the correct MAC to ra0. When I use the same setup on my TL-MR3020V3 I get a nulled MAC. 

As far as I can see the MAC to be used is stored in different flash areas, as factory is located in /dev/mtd6 on MR3020V3 while it is located in /dev/mtd2 on vocore. This again brings up the question, where the pre-compiled mt7628.ko is looking up the MAC from and how I might be able to tell it the MAC to use.

docbee63, 2018-03-25:
I also included a "/lib/firmware/mt7628.eeprom.bin" and as an alternative a "mt7628.eeprom" file but as expected, no change. I would really love to get a hint from someone knowing the source of the mt7628.ko how to tell this driver the mac for "ra0" interface. Any idea?

ZiqiangYu, 2018-03-25:
where did you get the  /lib/firmware/mt7628.eeprom.bin? it should be dumped form  /dev/mtd6 on MR3020V3 use hexdump or other tools.it is the factory rf parameter. the mac is  is in the following feild.
root@OpenWrt:/lib/firmware# hexdump mt7615.1.eeprom.bin
0000000 7615 0090 **0c00 2643 0060** 7615 14c3 8000
0000010 0002 14c3 7615 2203 ffff 0423 020d 028f

closed 2018-03-25 by ZiqiangYu

docbee63, 2018-03-25:
Thanks. Hexdump is not suited to extract a file, but to inspect it. I guess you are talking about dd and I am pretty familiar with this ;-)

dmesg tells me NOR flash partions as follows:
```
[    0.323486] Creating 4 MTD partitions on "spi32766.0":
[    0.328701] 0x000000000000-0x000000020000 : "boot"
[    0.335192] 0x000000020000-0x0000007c0000 : "firmware"
[    0.391267] 2 tplink-fw partitions found on MTD device firmware
[    0.397294] 0x000000020000-0x00000016b1b4 : "kernel"
[    0.403827] 0x00000016b1b4-0x0000007c0000 : "rootfs"
[    0.410603] mtd: device 3 (rootfs) set to be root filesystem
[    0.416871] 1 squashfs-split partitions found on MTD device rootfs
[    0.423214] 0x000000540000-0x0000007c0000 : "rootfs_data"
[    0.430248] 0x0000007c0000-0x0000007d0000 : "config"
[    0.437000] 0x0000007d0000-0x000000800000 : "factory"
```

factory partition is "/dev/mtd6". Looking into that we find:
```
00000000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
0000f100  7c 8b ca 6c 9f 5e ff ff  ff ff ff ff ff ff ff ff  ||..l.^..........|
0000f110  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
0000f200  09 a8 25 01 00 00 00 00  ff ff ff ff ff ff ff ff  |..%.............|
0000f210  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
```

Which is fine as the default dts file for kernel compilation says to expect MAC at 0xf100 in "factory" partition:
```
&wmac {
        status = "okay";
        mtd-mac-address = <&factory 0xf100>;
        mediatek,mtd-eeprom = <&factory 0x20000>;
};
```

docbee63, 2018-03-25:
thanks again for your patience. I downloaded your file and included the MAC:

```
# hexdump mt7628.eeprom.bin 
0000000 7628 0001 8b7c 6cca 5d9f 0000 0000 0000
0000010 ffff ffff ffff ffff ffff ffff ffff ffff
0000020 0000 0000 0020 0000 0000 0000 0000 0000
0000030 0000 0000 3422 2000 ffff 0100 0000 0000
0000040 0000 0022 0000 0000 0030 0000 0000 0000
0000050 0081 9400 b040 c640 0027 0000 c5c0 0027
0000060 ffff ffff ffff ffff ffff ffff ffff ffff
*
00000a0 c6c6 c4c4 c0c4 c4c0 c4c4 c0c4 c0c0 0000
00000b0 ffff ffff ffff ffff ffff ffff ffff ffff
*
00000f0 0000 0000 009b 8800 0000 0000 0000 0000
0000100 ffff ffff ffff ffff ffff ffff ffff ffff
*
0000120 0000 0000 0000 0000 0000 0000 0000 0077
0000130 1d11 1d11 7f15 7f15 7f17 7f17 3b10 3b10
0000140 ffff ffff ffff ffff ffff ffff ffff ffff
*
0000200
```
Unfortunately, the MAC inside "/lib/firmware/mt7628.eeprom.bin" is not used at all. After a rebbot "ifconfig" tells me:

```
ra0       Link encap:Ethernet  HWaddr 00:00:00:00:00:00  
          inet6 addr: fe80::200:ff:fe00:0/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:6 
```

ZiqiangYu, 2018-03-25:
add a line in wireless/mt7628/mt7628.dat as follow and try again:
E2pAccessMode=2

i'm not should where is the problem ,you can post the start up log, it may be help

docbee63, 2018-03-25:
I added all three possible MACs into the file:

```
# hexdump /lib/firmware/mt7628.eeprom.bin
0000000 7628 0001 8b7c 6cca 5d9f 0000 0000 0000
0000010 ffff ffff ffff ffff ffff ffff ffff ffff
0000020 0000 0000 0020 0000 8b7c 6cca 5d9f 8b7c
0000030 6cca 5d9f 3422 2000 ffff 0100 0000 0000
0000040 0000 0022 0000 0000 0030 0000 0000 0000
0000050 0081 9400 b040 c640 0027 0000 c5c0 0027
0000060 ffff ffff ffff ffff ffff ffff ffff ffff
*
00000a0 c6c6 c4c4 c0c4 c4c0 c4c4 c0c4 c0c0 0000
00000b0 ffff ffff ffff ffff ffff ffff ffff ffff
*
00000f0 0000 0000 009b 8800 0000 0000 0000 0000
0000100 ffff ffff ffff ffff ffff ffff ffff ffff
*
0000120 0000 0000 0000 0000 0000 0000 0000 0077
0000130 1d11 1d11 7f15 7f15 7f17 7f17 3b10 3b10
0000140 ffff ffff ffff ffff ffff ffff ffff ffff
*
0000200
```

...and also copied that file into "/etc/wireless/mt7628" directory, and added the "E2pAccessMode=2" into my "/etc/wireless/mt7628.dat".

But after reboot wireless adapters still show up with a nulled MAC.

Kernel boot log is as follows:
```
[    0.000000] Linux version 4.9.73 (boris@develop) (gcc version 5.5.0 (OpenWrt GCC 5.5.0 r5669-a367645) ) #0 Tue Jan 2 16:22:19 2018
[    0.000000] Board has DDR2
[    0.000000] Analog PMU set to hw control
[    0.000000] Digital PMU set to hw control
[    0.000000] SoC Type: MediaTek MT7628AN ver:1 eco:2
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019655 (MIPS 24KEc)
[    0.000000] MIPS: machine is TP-Link TL-MR3020v3
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 04000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] On node 0 totalpages: 16384
[    0.000000] free_area_init_node: node 0, pgdat 803f7174, node_mem_map 81000020
[    0.000000]   Normal zone: 128 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 16384 pages, LIFO batch:3
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Writing ErrCtl register=00052180
[    0.000000] Readback ErrCtl register=00052180
[    0.000000] Memory: 60440K/65536K available (3149K kernel code, 159K rwdata, 756K rodata, 152K init, 209K bss, 5096K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:256
[    0.000000] intc: using register map from devicetree
[    0.000000] CPU Clock: 580MHz
[    0.000000] clocksource_probe: no matching clocksources found
[... 117 lines cut ...]
[   12.949936] usbserial: USB Serial support registered for generic
[   12.974729] xt_time: kernel timezone is -0000
[   12.981112] usbcore: registered new interface driver ch341
[   12.986777] usbserial: USB Serial support registered for ch341-uart
[   12.995330] usbcore: registered new interface driver cp210x
[   13.001149] usbserial: USB Serial support registered for cp210x
[   13.011091] usbcore: registered new interface driver ftdi_sio
[   13.017027] usbserial: USB Serial support registered for FTDI USB Serial Device
[   13.033933] usbcore: registered new interface driver pl2303
[   13.039780] usbserial: USB Serial support registered for pl2303
[   13.049950] PPP generic driver version 2.4.2
[   13.056969] NET: Registered protocol family 24
[   13.066755] usbcore: registered new interface driver option
[   13.072619] usbserial: USB Serial support registered for GSM modem (1-port)
[   13.084848] kmodloader: done loading kernel modules from /etc/modules.d/*
[   14.764949] rt3050-esw 10110000.esw: link changed 0x00
[   16.843267] rt3050-esw 10110000.esw: link changed 0x01
[   18.764687] rt3050-esw 10110000.esw: link changed 0x00
[   20.873728] rt3050-esw 10110000.esw: link changed 0x01
[   22.764417] rt3050-esw 10110000.esw: link changed 0x00
[   24.002841] rt28xx_init, status=0
[   25.870853] rt3050-esw 10110000.esw: link changed 0x01
[   64.708855] random: crng init done

```

docbee63, 2018-03-25:
Does your WiFi MAC change when you change the MAC in the "mt7628.eeprom.bin" file?

ZiqiangYu, 2018-03-26:
if the E2pAccessMode=2 , of course it will change. you should post the wifi start log,then i'll see where is the problem.

Nossiac, 2018-03-26:
@ZiqiangYu  @docbee63  
AFAIK, the eeprom only holds the mac for ap interface (ra0, rai0, etc). "apcli" does not have an independent mac, its mac was calculated and set based on the ap interface.

And unfortunately, most mtk wifi drivers do not support customizing mac via `ifconfig`. Some necessary ioctl APIs are not implemented yet.

docbee63, 2018-03-26:
I reflashed the factory partition so that the typical 512 byte eeprom section starting with "7628 ...." which is located at offset 0x20000 is copied to offset 0x0 and I patched in desired MACs at offsets 4, 40 and 46. Having done so, the MAC is used for "ra0" and a modified mac (added 0x02 on the MSB) is used by the mt7628.ko. So for the moment I can live with this solution, but it would be much nicer if I could supply the 512 bytes eeprom data as a file without the need to remove the "read-only" tag in the kernel dtsi files to have factory partition flashable via "mtd write".

Is there a way to have the +2 increment on the MAC of the "apcli0" interface applied to the LSB instead of the MSB, where the vendor ID is coded inside the MAC?

Again, thanks for your help...

"factory" partition that works for MR3020v3:
```
0000000 7628 0200 8b7c 6cca 498c 0000 0000 0000
0000010 ffff ffff ffff ffff ffff ffff ffff ffff
0000020 0000 0000 0020 0000 8b7c 6cca 498c 8b7c
0000030 6cca 498c 3422 2000 ffff 0100 0000 0000
0000040 0000 0022 0000 0000 0030 0000 0000 0000
0000050 0000 9400 bc40 cac0 821e 8182 ca40 811e
0000060 8181 0000 0000 0000 0000 0000 0000 0000
0000070 0000 0000 0000 0000 0000 0000 0000 0000
*
00000a0 0000 0000 0000 0000 0000 0000 8181 0000
00000b0 0000 0000 0000 0000 0000 0000 0000 0000
00000c0 0000 0000 0000 0000 1a00 2a22 3531 3501
00000d0 4039 4d46 7f7f 007f 0000 0000 0000 0000
00000e0 1d11 1d11 351c 351c 351e 351e 1917 1917
00000f0 0002 0000 80b7 88ca 000a 0000 0000 0000
0000100 ffff ffff ffff ffff ffff ffff ffff ffff
*
0000120 0000 0000 0000 0000 0000 0000 0000 0077
0000130 1d11 1d11 7f15 7f15 7f17 7f17 3b10 3b10
0000140 ffff ffff ffff ffff ffff ffff ffff ffff
*
000f100 8b7c 6cca 4a8c ffff ffff ffff ffff ffff
000f110 ffff ffff ffff ffff ffff ffff ffff ffff
*
000f200 fc3c 042a 0000 0000 ffff ffff ffff ffff
000f210 ffff ffff ffff ffff ffff ffff ffff ffff
*
0020000 7628 0200 0c00 e143 5076 0000 0000 0000
0020010 ffff ffff ffff ffff ffff ffff ffff ffff
0020020 0000 0000 0020 0000 0c00 e143 2976 0c00
0020030 e143 2a76 3422 2000 ffff 0100 0000 0000
0020040 0000 0022 0000 0000 0030 0000 0000 0000
0020050 0000 9400 bc40 cac0 821e 8182 ca40 811e
0020060 8181 0000 0000 0000 0000 0000 0000 0000
0020070 0000 0000 0000 0000 0000 0000 0000 0000
*
00200a0 0000 0000 0000 0000 0000 0000 8181 0000
00200b0 0000 0000 0000 0000 0000 0000 0000 0000
00200c0 0000 0000 0000 0000 1a00 2a22 3531 3501
00200d0 4039 4d46 7f7f 007f 0000 0000 0000 0000
00200e0 1d11 1d11 351c 351c 351e 351e 1917 1917
00200f0 0002 0000 80b7 88ca 000a 0000 0000 0000
0020100 ffff ffff ffff ffff ffff ffff ffff ffff
*
0020120 0000 0000 0000 0000 0000 0000 0000 0077
0020130 1d11 1d11 7f15 7f15 7f17 7f17 3b10 3b10
0020140 ffff ffff ffff ffff ffff ffff ffff ffff
*
0020200 0000 0000 0000 0000 0000 0000 0000 0000
*
0020400 ffff ffff ffff ffff ffff ffff ffff ffff
*
002fe00 0000 0000 0000 0000 0000 0000 0000 0000
*
0030000
```

ZiqiangYu, 2018-03-26:
Is there a way to have the +2 increment on the MAC of the "apcli0" interface applied to the LSB instead of the MSB, where the vendor ID is coded inside the MAC?

**no, hw decided**