Nossiac/mtk-openwrt-feeds #43: mt7628 for mt7688 kernel panic when scan ap

CEC0D5F1B6AB, 2018-04-06:
[   37.220854] Main bssid = ec:0c:45:82:00:01
[   37.229041] <==== rt28xx_init, Status=0
[   37.254262] @@@ ed_monitor_exit : ===>
[   37.261707] @@@ ed_monitor_exit : <===
[   37.269222] mt7628_set_ed_cca: TURN OFF EDCCA  mac 0x10618 = 0xd7083f0f, EDCCA_Status=0
[   37.285102] WiFi Startup Cost (ra0): 3.760s
root@LEDE:~#
root@LEDE:~# ls
scan.sh
root@LEDE:~# sh scan.sh
scan.sh: line 1: wpriv: not found
[   43.000410] AddTxSType: already registered TxSType (PID = 32, Format = 0
[   43.013787] The 2-BSSID mode is enabled, the BSSID byte5 MUST be the multiple of 2
[   44.994337] reload DPD from flash , 0x9F = [c6ff] doReload bit7[1]
[   45.006622] CmdLoadDPDDataFromFlash: Channel = 6, DoReload = 1
[   45.054418] Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: 8705de64
[   45.054418]
[   45.077542] Rebooting in 3 seconds..

CEC0D5F1B6AB, 2018-04-07:
[  567.597446] /home/shello/workspace/lede-1701/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7628/tm7628/src/mt_wifi_ap/../mt_wifi/embedded/common/scan.c:118 assert (ch != 0)failed
[  567.630756] CmdChannelSwitch: central channel = 0 is invalid
root@LEDE:/# [  567.643132] reload DPD from flash , 0x9F = [c6ff] doReload bit7[1]
[  567.657514] CmdLoadDPDDataFromFlash: Channel = 0 (1~14), INSANE , return

Nossiac, 2018-04-20:
what did you do in scan.sh? can you post it here?
have you tried the latest luci-app-mtk plugin to scan ap around?