Nossiac/mtk-openwrt-feeds #45: LED indicators

venustrg, 2018-04-11:
tried MT7603 driver on mt7621s 64Mb RAM device.
i also have 7612 but not tried driver yet.
openwrt trunk/4.14.32 - i'm always getting 'out of memory' errors and reboot after a couple of minutes.
openwrt lede/4.4.92 - driver works, the only cons is non-working wifi LED.

2.4GHz/5Ghz LEDs work fine when running firmware compiled with rather old Raling SDK:
MTK_Ralink_ApSoC_SDK_4210_20140326 + MT7603E_DPA_LinuxAP_4.0.0.7_20140702 + MT7612E_MT7602E_LinuxAP_3.0.3.2_DPA_20140709

is there a way to enable/control LEDs?

venustrg, 2018-04-11:
also tested mt7612, it works.

found a difference between 7603 and 7612 - while there are no eeprom files supplied in /etc/wireless, mt7612 always sets zero mac/bssid, while mt7603 sets some reasonable value.

i got eeprom files for both chips from ralink sdk, edited them for correct MACs (they had zero values), and now both chips have correct bssid.

mtk luci plugin shows me interface 'ra0' for both 7603 and 7612, however configures them correctly as 'ra0' and 'rai0'.

Nossiac, 2018-04-12:
@venustrg there's an option for LED in wifi driver's Kconfig, i guess i forgot to enable it ....

Nossiac, 2018-04-17:
for luci plugin bug, you can start a new issue.
I already noticed some issues in the plugin, will push an update, someday...

closed 2018-04-17 by Nossiac