Nossiac/mtk-openwrt-feeds #47: mt7612 kernel 4.14.32 reboot immediately when clicking reload button

updatede, 2018-04-14:
Xiaomi router 3g 
latest openwrt snapshot with kernel 4.14.32
not install mt76 open driver
"modprobe mt7612" looks like nothing wrong
"ifconfig rai0 up" system hang, must pull plug 
install mt76 open driver firstly, then mannually uninstall them
"modprobe mt7612" and "ifconfig rai0 up" looks like nothing wrong
click reload button in mtk-uci-plugin, system reboot
lead to driver unusable

updatede, 2018-04-16:
I use openwrt offical precompiled kernel, maybe not compatible with your drivers, have you ever tried openwrt precompiled kernel ?
You can download it at:
https://downloads.openwrt.org/snapshots/targets/ramips/mt7621/packages/kernel_4.14.34-1-0643775945eeef3b6bc3f0c7a01e8f0d_mipsel_24kc.ipk

Nossiac, 2018-04-17:
sorry bro, I haven't got the chance to test it with xiaomi 3g yet ... 
if you run into any crash, plz post the console/dmesg log here.

updatede, 2018-04-18:
There isn't any console/dmesg log, because it reboot and I have not soldered a serial console.

ingamedeo, 2018-06-01:
Confirmed. Xiaomi 3G, latest OpenWrt trunk + mtk drivers (compiled from source)
Serial log: [https://gist.github.com/ingamedeo/0c63e56cfb7782fbbf29bcd20f07070b](https://gist.github.com/ingamedeo/0c63e56cfb7782fbbf29bcd20f07070b)