Nossiac/mtk-openwrt-feeds #48: how to change mac/bssid

venustrg, 2018-04-17:
how can i change mac/bssid without binary eeprom editing?
wireless interfaces not listed in openwrt config, so uci can't be used.
mtk luci plugin also gives me no such option.

Nossiac, 2018-04-18:
Tt seems mtk wifi drivers did not impl the hwaddr api, so, for the time being, nope, you cannot.
This requirement seems reasonable, I'd like to have a try.

I do hope mtk open its driver, there's a lot of improvements can be done.

Nossiac, 2018-04-25:
the term "mac" actually means "media access control", so those params access the registers for "media access control".
as i said, mtk's driver does not impl the hook to set mac, so technically you cannot change the mac via cmdline.

closed 2018-05-07 by Nossiac