Nossiac/mtk-openwrt-feeds #51: Can't load MT7603 & MT7615 driver simultaneously

mark0201tw, 2018-04-19:
How can I load mt7603.ko and mt7615.ko simultaneously? I got the error message while loading mt7615 after mt7603 as below (Load either mt7603 or mt7615 is OK that could be probed)
[483.786344] mt7615: exports duplicate symbol AsicSwtichChannel (owned by mt7)
1 module could not be probed
- mt7615
root@openwrt:/lib/modules/4.14.32#

Nossiac, 2018-04-20:
These 2 drivers have serveral conflicts actually: they try to export the same symbols, they try to register the same driver signature... MTK gives some workaround tricks to build them together, but in my case, these tricks are aweful.  I have to strip some code to fix this.

You may try the next autobuild mt7603.ko

mark0201tw, 2018-04-24:
I tried today's build. It seems the issue still there. Look forward your fix !