Nossiac/mtk-openwrt-feeds #55: 请教K2P如何配置？

hzjnet, 2018-04-23:
现在的情况是:
1 modprobe mt7621无特别的出错，ifconfig -a 能看见无线网卡ra0 ,dmesg信息如下：
  [  578.786120] mt7615: module license 'Proprietary' taints kernel.
[  578.799531] Disabling lock debugging due to kernel taint
[  578.902239] register mt_drv
[  578.908692] bus=0x1, slot = 0x0, irq=0x0
[  578.931906] 
[  578.931906] 
[  578.931906] === pAd = c0e01000, size = 3620696 ===
[  578.931906] 
[  578.954529] <-- RTMPAllocAdapterBlock, Status=0
[  578.964806] get_wdev_by_idx: invalid idx(0)
[  578.974259] pAd->PciHif.CSRBaseAddress =0xc0d00000, csr_addr=0xc0d00000!
[  578.989419] get_wdev_by_idx: invalid idx(0)
[  578.998881] RTMPInitPCIeDevice():device_id=0x7615
[  579.009509] DriverOwn()::Try to Clear FW Own...
[  579.320314] DriverOwn()::Success to clear FW Own
[  579.330775] mt_pci_chip_cfg(): HWVer=0x8a10, FWVer=0x8a10, pAd->ChipID=0x7615
[  579.346913] mt_pci_chip_cfg(): HIF_SYS_REV=0x76150001
[  579.358333] RtmpChipOpsHook(493): Not support for HIF_MT yet! MACVersion=0x0
[  579.374263] mt7615_init()-->
[  579.380771] Use 1st ePAeLNA default bin.
[  579.389641] Use 0st /etc/wlan/MT7615E_EEPROM1.bin default bin.
[  579.402929] <--mt7615_init()
[  579.409516] ChipOpsMCUHook
[  579.421997] cut_through_token_list_init(): TokenList inited done!id_head/tail=0/4096
[  579.439536] cut_through_token_list_init(): 871a8b08,871a8b08
[  579.453053] cut_through_token_list_init(): TokenList inited done!id_head/tail=0/4096
[  579.470582] cut_through_token_list_init(): 871a8b18,871a8b18
[  579.483648] <-- RTMPAllocTxRxRingMemory, Status=0
[  579.494288] get_wdev_by_idx: invalid idx(0)
[  579.503739] get_wdev_by_idx: invalid idx(0)
[  579.513488] get_wdev_by_idx: invalid idx(0)
[  579.522982] get_wdev_by_idx: invalid idx(0)
[  579.532435] get_wdev_by_idx: invalid idx(0)
[  579.541897] get_wdev_by_idx: invalid idx(0)
[  579.551349] get_wdev_by_idx: invalid idx(0)
[  579.562199] get_wdev_by_idx: invalid idx(0)
[  579.572079] pci probe count=1

2 ifconfig ra0 up 执行后直接网络断掉，mt7615驱动也掉落
lsmod|grep mt7615输出变成空

请教如何处理才能正常启动无线

Nossiac, 2018-04-25:
it seems the driver probes properly in step 1.  did you get any log for step 2?

hzjnet, 2018-04-25:
step 2 no log

helloray, 2018-04-27:
Same issue.  My router will reboot when I ifconfig ra0 up. I guess it causes kernel panic.

jigaowan, 2018-04-29:
相同问题，ra0 的 HWaddr为 00:00:00:00:00:00 

```
ra0       Link encap:Ethernet  HWaddr 00:00:00:00:00:00
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:22
```

haswelliris, 2018-05-14:
同样的问题，参照issue #61 的方法做了，并没有什么作用，不知道他的.eeprom.bin和sku.dat到底是怎么弄的

recyard, 2018-05-15:
Same issue here. R6220, mt7603/7612, both ra0 and rai0 HWaddr are 0.

Nossiac, 2018-05-15:
`HWaddr 00:00:00:00:00:00` MAC为0一般都是因为driver找不到eeprom data。可以在dmesg里看到“failed to open /lib/firmware/mt76xx.eeprom.bin”这样的error。不同的产品，eeprom data在flash上的位置可能不一样。一般的，可以通过 ·cat /proc/mtd· 看看是否存在一个factory分区，有的话里面就是eeprom data。将它的内容dd dump到/lib/firmware下，让driver看到，就解决了。

kernel crash 的话，就比较难搞了。这个项目上面收到很多个crash issue，我都没有办法复现和debug。

helloray, 2018-05-16:
@Nossiac , 谢谢提供信息，这个比较重要，请问针对mt7615,  eeprom在/lib/firmware/下对应的准确的名字是什么？

haswelliris, 2018-05-16:
1、查看factory分区
```
root@OpenWrt:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00030000 00010000 "u-boot"
mtd1: 00010000 00010000 "u-boot-env"
mtd2: 00010000 00010000 "factory"
```
2、把factory信息dd到eeprom.bin
```
root@OpenWrt:~# dd if=/dev/mtd2 of=/lib/firmware/mt7615.eeprom.bin
128+0 records in
128+0 records out
```
3、为了跟issue61那位的步骤一样，我复制了mt7615.eeprom.bin到/lib/firmware/mt7615.1.eeprom.bin
并且dmesg有提到`[  161.567072] Use 0st /etc/wlan/MT7615E_EEPROM1.bin default bin.
`，我还把eeprom复制到上述路径：/etc/wlan/MT7615E_EEPROM1.bin

4、执行`modprobe mt7615`，然而ra0的max地址还是00。
查看dmesg相比原来没有导出eeprom.bin的时候，只有最后少了一行` get_wdev_by_idx: invalid idx(0)`，除此之外并没有什么区别，也没有`failed to open /lib/firmware/mt76xx.eeprom.bin`这样的报错
```
[   51.476755] mt7615: module license 'Proprietary' taints kernel.
[   51.488577] Disabling lock debugging due to kernel taint
[   51.578242] register mt_drv
[   51.583924] bus=0x1, slot = 0x0, irq=0x0
[   51.604416]
[   51.604416]
[   51.604416] === pAd = c0c81000, size = 3620696 ===
[   51.604416]
[   51.624532] <-- RTMPAllocAdapterBlock, Status=0
[   51.633619] get_wdev_by_idx: invalid idx(0)
[   51.642075] pAd->PciHif.CSRBaseAddress =0xc0b80000, csr_addr=0xc0b80000!
[   51.655424] get_wdev_by_idx: invalid idx(0)
[   51.663764] RTMPInitPCIeDevice():device_id=0x7615
[   51.673167] DriverOwn()::Try to Clear FW Own...
[   51.982388] DriverOwn()::Success to clear FW Own
[   51.991590] mt_pci_chip_cfg(): HWVer=0x8a10, FWVer=0x8a10, pAd->ChipID=0x7615
[   52.005798] mt_pci_chip_cfg(): HIF_SYS_REV=0x76150001
[   52.015866] RtmpChipOpsHook(493): Not support for HIF_MT yet! MACVersion=0x0
[   52.029908] mt7615_init()-->
[   52.035626] Use 1st ePAeLNA default bin.
[   52.043440] Use 0st /etc/wlan/MT7615E_EEPROM1.bin default bin.
[   52.055067] <--mt7615_init()
[   52.060807] ChipOpsMCUHook
[   52.072231] cut_through_token_list_init(): TokenList inited done!id_head/tail     =0/4096
[   52.087666] cut_through_token_list_init(): 87183e08,87183e08
[   52.099506] cut_through_token_list_init(): TokenList inited done!id_head/tail     =0/4096
[   52.114962] cut_through_token_list_init(): 87183e18,87183e18
[   52.126318] <-- RTMPAllocTxRxRingMemory, Status=0
[   52.135694] get_wdev_by_idx: invalid idx(0)
[   52.144119] get_wdev_by_idx: invalid idx(0)
[   52.152642] get_wdev_by_idx: invalid idx(0)
[   52.160986] get_wdev_by_idx: invalid idx(0)
[   52.169293] get_wdev_by_idx: invalid idx(0)
[   52.177625] get_wdev_by_idx: invalid idx(0)
[   52.185953] get_wdev_by_idx: invalid idx(0)
[   52.195571] get_wdev_by_idx: invalid idx(0)
[   52.204247] pci probe count=1
```
```
ra0       Link encap:Ethernet  HWaddr 00:00:00:00:00:00
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:22
```

haoyixin, 2018-05-16:
panic我遇到过，具体trace忘记了，过程不提了，直接说结论，去掉kmod-bridge和kmod-llc (4.4内核)

helloray, 2018-05-17:
@haoyixin, bridge都不要的话，LAN这边咋弄

haoyixin, 2018-05-18:
@helloray 不影响

helloray, 2018-05-18:
@haoyixin , 除非将ra0添加成新的LAN2，并开启DHCP Server，但这样的话，其他有线以太网口就和ra0不是桥接在一起了，得通过路由转发通讯，效率应该没有桥接效率高

Nossiac, 2018-05-25:
IMO, there's no reason to remove kernel bridge. so far as I know, a lot of compatibility issues (like probe failure, crash at insmod, etc. ) were dts related (it's configuration causes conflicts against wifi drivers here, since mediatek wifi drivers do not read dts at all). Unfortunately I don't have all those models to test it.

If this project gets popular, I guess a nice shot would be setting up a table here, saying: 
```
the driver has been verified with the following models: including XXX, XXXX, XXXXX, ....,
the steps to use it with model XXX would be BLAHBLAHBLAH....
```

Iy204, 2018-06-01:
在lib/firmware放入了eeprom也在/etc/wireless放入了sku.dat。加载了驱动以后一样是mac地址为0。2.4G有个错误的Mac地址，2.4G能接入信号特别差。5G 无法链接。

Iy204, 2018-06-02:
在`/etc/wireless/mt7615`放入`mt7615-sku.dat` `mt7615-sku-bf.dat`两个文件提取自原厂固件
修改同目录下Wifi的配置文件`mt7615.1.2/5G.dat`里的`E2pAccessMode=2`
然后在`/lib/firmware`中放入eeprome，文件名 `mt7615.1.eeprom.bin`
`modprobe mt7615`
`mtkwifi reload`

helloray, 2018-06-02:
@Iy204 这样就OK了吗？

Iy204, 2018-06-02:
@helloray 至少mac地址已经可以正确显示了，WIFI信号也正常，至于其他的就得大家来试试了。我并非专业的。

haswelliris, 2018-06-03:
@Iy204  按照您说的方法做了，在`modprobe mt7615`之后用`ifconfig -a`查看mac还是00，接着运行`mtkwifi reload`直接系统崩溃重启。
我试过编译了两个内核版本(4.14.34和4.14.44)的openwrt，都不能正确显示mac地址，不知道您的系统和内核版本，我现在怀疑有可能是编译选项的问题。
如果可以的话，能提供您的编译配置`.config`文件或者固件吗？

另外就是在modprobe加载驱动后，也没有看到rax0接口，只有ra0。感觉还是eeprom没有正确被驱动读取，但是已经按照上面的路径和命名导出了，就很迷。

内核崩溃信息记录
modprobe mt7615
```
[   58.670939] mt7615: module license 'Proprietary' taints kernel.
[   58.682757] Disabling lock debugging due to kernel taint
[   58.779006] register mt_drv
[   58.784744] bus=0x1, slot = 0x0, irq=0x0
[   58.805816]
[   58.805816]
[   58.805816] === pAd = c1801000, size = 3620696 ===
[   58.805816]
[   58.825943] <-- RTMPAllocAdapterBlock, Status=0
[   58.835282] get_wdev_by_idx: invalid idx(0)
[   58.843718] pAd->PciHif.CSRBaseAddress =0xc1700000, csr_addr=0xc1700000!
[   58.857187] get_wdev_by_idx: invalid idx(0)
[   58.865563] RTMPInitPCIeDevice():device_id=0x7615
[   58.874999] DriverOwn()::Try to Clear FW Own...
[   59.203074] DriverOwn()::Success to clear FW Own
[   59.212320] mt_pci_chip_cfg(): HWVer=0x8a10, FWVer=0x8a10, pAd->ChipID=0x7615
[   59.226592] mt_pci_chip_cfg(): HIF_SYS_REV=0x76150001
[   59.236941] RtmpChipOpsHook(493): Not support for HIF_MT yet! MACVersion=0x0
[   59.251057] mt7615_init()-->
[   59.256874] Use 1st ePAeLNA default bin.
[   59.264752] Use 0st /etc/wlan/MT7615E_EEPROM1.bin default bin.
[   59.276506] <--mt7615_init()
[   59.282264] ChipOpsMCUHook
[   59.293544] cut_through_token_list_init(): TokenList inited done!id_head/tail                                                                                                             =0/4096
[   59.308986] cut_through_token_list_init(): 854a1608,854a1608
[   59.320830] cut_through_token_list_init(): TokenList inited done!id_head/tail                                                                                                             =0/4096
[   59.336309] cut_through_token_list_init(): 854a1618,854a1618
[   59.347690] <-- RTMPAllocTxRxRingMemory, Status=0
[   59.357088] get_wdev_by_idx: invalid idx(0)
[   59.365581] get_wdev_by_idx: invalid idx(0)
[   59.374358] get_wdev_by_idx: invalid idx(0)
[   59.382781] get_wdev_by_idx: invalid idx(0)
[   59.391124] get_wdev_by_idx: invalid idx(0)
[   59.399472] get_wdev_by_idx: invalid idx(0)
[   59.407784] get_wdev_by_idx: invalid idx(0)
[   59.417493] get_wdev_by_idx: invalid idx(0)
[   59.426089] pci probe count=1
```
mtkwifi reload
```
[   69.358644] get_wdev_by_idx: invalid idx(0)
[   79.052752] get_wdev_by_idx: invalid idx(0)
[   79.061136] get_wdev_by_idx: invalid idx(0)
[   79.069762] get_wdev_by_idx: invalid idx(0)
[   79.078072] get_wdev_by_idx: invalid idx(0)
[   79.086414] get_wdev_by_idx: invalid idx(0)
[   79.094764] get_wdev_by_idx: invalid idx(0)
[   79.103312] get_wdev_by_idx: invalid idx(0)
[   79.111660] DriverOwn()::Return since already in Driver Own...
[   79.123308] ===============================
[   79.131687] Current DevInfo Num: 0
[   79.138510] ===============================
[   79.146853] ===============================
[   79.155249] Current BssInfo Num: 0
[   79.162037] ===============================
[   79.170433] ===============================
[   79.178975] Current StaRec Num: 0
[   79.185583] ===============================
[   79.321635] multi-profile merge success, en:1,pf1_num:1,pf2_num:1,total:2
[   79.367135] RT_CfgSetMacAddress : invalid length (0)
[   79.377243] RT_CfgSetMacAddress : invalid length (0)
[... 97 lines cut ...]
[   80.122618] S
[   80.125737]
[   80.131853] hw/sw version =
[   80.137616] 8a
[   80.137623] 10
[   80.140939] 8a
[   80.144233] 10
[   80.147525]
[   80.153782] patch version =
[   80.159513] 00
[   80.159518] 00
[   80.162810] 00
[   80.166102] 10
[   80.169432]
[   80.175717] skbuff: skb_under_panic: text:8511a92c len:36 put:32 head:84d6eee                                                                                                             0 data:84d6eec0 tail:0x84d6eee4 end:0x84d6ef40 dev:<NULL>
[   80.199851] Kernel bug detected[#1]:
[   80.206973] CPU: 3 PID: 2150 Comm: ifconfig Tainted: P                4.14.34                                                                                                              #0
[   80.221681] task: 8704f6c0 task.stack: 862da000
[   80.230690] $ 0   : 00000000 00000001 0000007a 00000000
[   80.241091] $ 4   : 81139334 81139334 8113de78 00005688
[   80.251486] $ 8   : 00000000 00000172 00000007 00000000
[   80.261880] $12   : 00000000 80580000 0002ae65 00000000
[   80.272276] $16   : 8567c880 8567c880 c1b71000 c1801000
[   80.282670] $20   : 8634fb40 c1b70b28 c1b70b40 8634fb40
```

Iy204, 2018-06-03:
@haswelliris 我用的就是Master的代码。4.14.44的。
实际上使用中问题不少。虽然WIFI信号正常了，可是断流很明显。
而且无故的就重启了。日志里看不出东西。
https://www.upload.ee/files/8516176/openwrt-ramips-mt7621-k2p-squashfs-sysupgrade.zip.html

接口的话   `ifconfig -a` 看看

haswelliris, 2018-06-03:
@Iy204 非常感谢您提供的信息。
使用您提供的配置后就能正确挂载mt7615驱动，mac也是对的，wifi也能用。
我这里wan-rax0测速只跑到120Mbps左右。在测试lan-rax0的时候，只要出现大流量（比如速度超过300Mbps），就会导致wifi与br-lan的桥接完全断开的情况（即设备还能连上wifi，但是无法ping通路由器）。然而此时并未触发内核错误。
之后，差不多20分钟后，再查看dmesg，就出现`-ash: can't fork: Out of memory`。在这之后的约5分钟，最终路由器完全卡死宕机，外网中断，但依然能ping通路由器。

看上去是内存泄漏了。详细测试见后文。我的系统内核版本是4.14.44，用openwrt最新master代码（r7093-4fdc6ca）编译。使用对应内核版本的mt7615驱动。多次测试发现，该现象多发生在空闲时突然发起高速率（大于300Mbps）wifi数据传输的时候。

此外，我尝试复现该错误，用连接lan口的设备从连接wifi的设备上下载数据，峰值速度超过了300Mbps，持续了十多秒，之后路由器瞬间卡死，ssh断开，跟wifi设备的连接断开，出现上面描述的最终路由器卡死宕机的情况。

再次尝试复现错误，此次使用两台连接wifi的设备传输文件，全程速度在60Mbps左右，此时没有出现因内存占满导致宕机，但是中途出现了内存占满的报错。使用free -h不断查看，会发现在开始连接之后内存一直上涨（此时速度在120Mbps左右），直到第一次内存用光报错出现后，传输速度降到60Mbps以内，之后内存占用率维持在80%以上（100MB/128MB），时不时出现内存占满的报错，但因为传输速度掉下去了，所以并没有整个宕机。
紧接着，跟前两次一样用lan口设备下载wifi设备的文件，此次高速度维持了很久，监控内存使用量从空闲时的40MB增加到了70MB左右并一直稳定在80MB/128MB左右。之后，突然又出现上面的连接断开的情况，不过这次直接导致路由崩溃重启了。

之后继续复现，基本跟上述一样，小流量可以使用wifi，大流量直接导致wifi断开，路由器宕机。在200Mbps左右的情况下，占用内存达到90MB/128Mb，一旦速度超过300M会导致内存占满，速度立刻降到200以内甚至100Mbps，如果此时没有降速，那么大概率路由器宕机。

wan-lan测速是正常，速度在900Mbps左右，同时不会导致内存占用增加，只有使用wifi的时候会导致大量内存占用，最终因长时间爆内存导致整个系统崩溃，路由宕机。

但是，最后一次测试中，反复大流量wifi传输（保持500Mbps传完10GB，测试多次），没有出现内存占用增加的情况。但在这几次传输之前的测试中，出现过短暂内存占满的情况，但因传输速度降低，内存占用马上就降下来，没有导致系统崩溃。也正是在此之后，即便大流量传输也没有出现过内存占满的情况。不过等路由器空闲半小时之后，进行同样的传输测试，但在几秒后路由直接崩溃重启。

总的来说，应该是驱动导致内存泄漏了，并且这个现象（内存占满导致wifi断开，路由卡死宕机）只会发生在瞬时流量激增的情况下。如果长时间进行低速率200Mbps以内的传输，则内存泄漏现象会自动消除（根据我的观察从之前90MB/128MB占用降到60MB/128MB并稳定下来），这是即便再进行大流量传输也不会导致内存占用过多。
@Nossiac 出错和测试记录如上，抱歉描述有点长，不知道是驱动和系统不兼容还是什么情况导致的。毕竟很玄学的地方是如果进行一段时间低速率“预热”，则不会出现上述错误。
我的K2P系统内核版本是4.14.44，用openwrt最新master代码（r7093-4fdc6ca）编译。使用mt7621对应内核版本4.14.44的mt7615驱动。

相关内核信息如下：
第一次崩溃前的信息：
```
[  890.281945] Rcv Wcid(1) AddBAReq
[  890.288667] Start Seq = 00000015
[  944.863193] Rcv Wcid(1) AddBAReq
[  944.869669] Start Seq = 00000014
[ 1183.583592] Rcv Wcid(1) AddBAReq
[ 1183.590051] Start Seq = 00000000
[ 1543.361741] MacTableInsertEntry, Entry Idx:2 is inuse, overlap
[ 1958.849029] MacTableInsertEntry, Entry Idx:2 is inuse, overlap
[ 1964.087068] MacTableInsertEntry, Entry Idx:2 is inuse, overlap
[ 2343.171231] MacTableInsertEntry, Entry Idx:1 is inuse, overlap
[ 2354.335040] MacTableInsertEntry, Entry Idx:1 is inuse, overlap
[ 2366.930285] MacTableInsertEntry, Entry Idx:1 is inuse, overlap
[ 2379.419084] MacTableInsertEntry, Entry Idx:1 is inuse, overlap
[ 2392.015996] MacTableInsertEntry, Entry Idx:1 is inuse, overlap
[ 2393.200968] MacTableInsertEntry, Entry Idx:2 is inuse, overlap
[ 2397.977785] MacTableInsertEntry, Entry Idx:1 is inuse, overlap
[ 2405.333780] MacTableInsertEntry, Entry Idx:1 is inuse, overlap
[ 2414.150006] MacTableInsertEntry, Entry Idx:1 is inuse, overlap
root@OpenWrt:/bin# dmesg
-ash: can't fork: Out of memory
```

出现out of memory时，系统内核信息：
```
[   53.779752] Rcv Wcid(1) AddBAReq
[   53.786225] Start Seq = 00000000
[   74.826772] Rcv Wcid(2) AddBAReq
[   74.833447] Start Seq = 00000015
[   77.510010] 68, flush one!
[   77.614666] 6a, flush one!
[  142.689192] fd0, flush one!
[  172.041525] 2ad, flush one!
[  178.141772] c9a, flush one!
[  184.283610] 383, flush one!
[  196.031305] d87, flush one!
[  202.480946] 4ed, flush one!
[  208.020789] 912, flush one!
[  214.509158] 496, flush one!
[  214.724934] 537, flush one!
[  238.090712] 7e6, flush one!
[  274.567871] e53, flush one!
[  304.361298] 2d7, flush one!
[  304.377082] 968, flush one!
[  304.578354] 379, flush one!
[  310.761364] b1d, flush one!
[  328.361498] 594, flush one!
[  340.116851] b95, flush one!
[  357.866592] 9f8, flush one!
[  370.345273] dc1, flush one!
[  370.779746] f88, flush one!
[  388.584023] a6a, flush one!
[  406.381424] e46, flush one!
[  406.601515] f3e, flush one!
[  406.821454] 28, flush one!
[  436.394789] 909, flush one!
[  555.183266] Rcv Wcid(1) AddBAReq
[  555.189895] Start Seq = 00000000
```
以及这段时间使用free -h监控的内存使用情况：
```
root@OpenWrt:~# free -h
             total       used       free     shared    buffers     cached
Mem:        123728     102660      21068        524        872       2268
-/+ buffers/cache:      99520      24208
Swap:            0          0          0
root@OpenWrt:~# free -h
-ash: can't fork: Out of memory
root@OpenWrt:~# free -h
-ash: can't fork: Out of memory
root@OpenWrt:~# free -h
-ash: can't fork: Out of memory
root@OpenWrt:~# free -h
             total       used       free     shared    buffers     cached
Mem:        123728     104744      18984        524       1788       3124
-/+ buffers/cache:      99832      23896
Swap:            0          0          0
```
正常的内存使用（低速率“预热”之后高速率传输时内存占用量也是差不多是这样）：
```
root@OpenWrt:~# free -h
             total       used       free     shared    buffers     cached
Mem:        123728      71236      52492        448       2172       3428
-/+ buffers/cache:      65636      58092
Swap:            0          0          0
```

Nossiac, 2018-06-04:
@haswelliris  哇，非常详尽的日志，非常专业的分析。

抱歉，最近在忙新项目，没有太多精力来复现和debug这里的问题。先纸上谈兵，说说我的看法：

首先，wifi driver在高速场景下，会缓存很多个网络数据包，放在一个队列里。这个队列可能增长到很大，导致系统可用内存变少。一旦数据包被消化，内存又会慢慢还给系统。为了避免无限增长导致系统崩溃，这个缓存队列长度是有极限的，超过极限值，数据包就会被丢弃。这个值是经过多种测试得出的经验值，跟硬件内存容量是匹配的。

所以理论上，wifi driver不会无限制的吃内存。除非，出现了内存泄露。

有两个信息可以辅助分析这种内存问题。
```
cat /proc/meminfo # 系统内存可用状况统计
cat /proc/slabinfo # 查看内核slab分析器的统计信息，一般的网络包都是从这里取内存的。
```

假设是wifi driver（也包括内核中的其它模块）内存泄露，当它申请不到内存时，kernel api是会有反应的，一般有alloc failure这种信息信息印出来，同时driver一定会表现异常。

从你的log看，driver还没有出现异常日志，但是应用层先出事了。所以**不能排除应用层持续大量分配内存导致的问题**。

下面是我曾经用过的一个小程序，它在死循环里fork子进程，直到耗光应用层可分配的内存空间。这时执行任何shell命令都会报 cant fork。注意最后的日志，一般可以看到内核其实还有可用内存，driver不会报错。

```
#include <unistd.h>
#include <stdio.h>
int main () {
    pid_t fpid;
    int count = 0;
    FILE * fp = NULL;
    char buffer[1024];
    while (1) {
        fpid = fork();
        if (fpid < 0) {
            printf("error in fork!\n");
            break;
        } else if (fpid == 0) {
            count++;
        } else {
            sleep(100);
            return 0;
        }
    }
    printf("forked %d\n", count);
    fp = fopen("/proc/meminfo", "rb");
    while (fgets(buffer, sizeof(buffer), fp))
        printf("%s", buffer);
    fclose(fp);
    return 0;
}

```

Nossiac, 2018-06-04:
好了，走运了，我在一块开发板上跑起了K2P的软件。
做了一些简单的测试后，确实发现了内存慢慢变少的问题~

值得一查，不过不要太指望很快有结果，看缘分，[笑脸]。

haswelliris, 2018-06-04:
@Nossiac 十分感谢您抽空来分析这个问题。
如果是应用层导致的话，我这边先测试下不同配置选项编译的固件。有可能是编译的模块和额外组件导致的问题。毕竟之前我自己勾选的配置编译出来的固件是不能正确使用驱动加载出mac地址的。

Pentiumluyu, 2018-06-04:
@Nossiac @haswelliris 确认不使用MULTI_INF_SUPPORT,RT_SECOND_CARD,RT_THIRD_CARD选项依旧会有内存变少的问题。其实只要执行ifconfig ra0 up或者ifconfig rax0后，不执行任何操作，free显示每秒大概下降64kb（或32kb，很稳定），因为加载mt7615.ko后可用内存大概为60～70mb，所以大概1200s后会oom，尝试`MEM_ALLOC_INFO_SUPPORT`，dmesg没有发现memory leak；尝试使用kmemleak，没有发现于mt7615有关的memory leak；查看slabinfo，内存从60m降到20m，并没有发现有大量增加的项目；查看meminfo，MemFree及MemAvailable不断减少但其他项目没有增加。只加载mt7615并不会导致内存减少。

liyiwu, 2018-06-07:
我开了一个 k2p 的分支，删除了7615以外的驱动，将原厂的配置文件和eeprom移植过来了。
修改 feeds 源
src-git mtk https://github.com/liyiwu/mtk-openwrt-feeds;k2p

测试成功。
不知道怎么推送分支，希望能合到nossiac的库中。

haswelliris, 2018-06-09:
现在长时间使用也会出现无线断开的问题，具体表现是搜得到SSID，但尝试连接被拒绝。
内核报错如下：
```
[71145.844853] RtmpPCIMgmtKickOut: Not available element in FastPathTxFreeQue
[71145.958771] RtmpPCIMgmtKickOut: Not available element in FastPathTxFreeQue
[71146.119559] RtmpPCIMgmtKickOut: Not available element in FastPathTxFreeQue
```
搜索后发现潘多拉固件也是无线有问题并且也是该报错。。。
（看上去是跟fastpath软件加速冲突了。。。）

liyiwu, 2018-06-19:
死机问题找到原因了，刷固件后，要恢复出厂设置一次，目前没出现死机现象。继续观察中。
@haswelliris 可以尝试一下

thisdk, 2018-06-20:
我每天都打开这个issues一次,就是等众大神解决问题.然后自己编译纯净版的openwrt18.06.辛苦各位了.
@liyiwu 是不是我把 src-git mtk https://github.com/liyiwu/mtk-openwrt-feeds;k2p 加到openwrt18.06去编译就可以?我想试试效果如何

liyiwu, 2018-06-20:
是的，用 make menuconfig 选择好组件就可以了。目前编译会有问题，18.06最新内核的驱动还没放出来呢。我只是做了一些设置工作，核心还是大神的驱动。目前使用发现wifi会假死，过一会又会恢复。

Nossiac, 2018-06-20:
Sorry, busy days. Hopefully I'm getting back on this next month.
BTW, K2P is more popular than I expected! Why exactly do you guys prefer it?

helloray, 2018-06-21:
@Nossiac K2P is completely free. You buy it and get your money back after three month.

hcym, 2018-06-22:
wifi假死，其他正常

etnperlong, 2018-06-24:
@liyiwu 使用了您的源编译成功了！WIFI正常可用。问题就是不定期WIFI会无法连接，突然处于假死状态，LuCI界面也无法进入。

liyiwu, 2018-06-29:
这个问题我也解决不了，要靠大神了

Pentiumluyu, 2018-06-30:
有几个发现:
1. 将SKB_BUF_MINIMUN_SIZE增大到2048可以解决处理skb时导致的kernel panic（开启slub_debug后显示redzone overwritten，overwrite了32byte的0xBB） 的问题
2.out of memory 的问题与DBDC有关。仅执行ifconfig ra0 up不会出现可用内存减少的现象，但只要执行ifconfig rax0 up, 无论之前是否执行了ifconfig ra0 up，均会导致可用内存减少，而且即使在执行了ifconfig rax0 down后，可用内存依然会不断减少
3.同样的编译选项及配置文件，Openwrt 15.05 (Linux kernel 3.18)不会出现内存不断减少的现象
4.保持网络中有持续的流量（正常上网即可）无论是无线或有线的wan<->lan, lan<->lan, 可用内存都会不断地在下降后回升

@Nossiac 可以提供一下debug的思路吗？

huanghaizhuang, 2018-07-03:
我觉得ＭＴＫ的芯片大部分存在这问题。提到是数据内存丢失，是队列排队太多，毛失前面数据，。ＷＩＦＩ丢失前半发送数据内存。直接没法再接收后面数据，ＷＩＦＩ死机！要改闭源驱动里面一项文件。减少数据瞬间发送量。要分割内存版块数据，适应在ＤＤＲ2内存上面。内存量和ＣＰＵ处理数据不同步

hcym, 2018-07-03:
看来大神们暂时没有办法，只有k2p不能openwrt，也不急需

liyiwu, 2018-07-11:
大神更新驱动了，不过还没有新版内核的  lede17 的4.4.139  openwrt18 的 4.14.52

hcym, 2018-07-13:
哦，看看效果

omencathay, 2018-07-25:
我测试了下，kernel version 4.14.54   ,
mt7615-for-mt7621-linux-4.14.44.ko  

2.4G和5G能起来，但是用着用着就会自动重启，ttl显示里面并无任何有价值的信息

liyiwu, 2018-08-29:
编译个新版的
 架构	MediaTek MT7621 ver:1 eco:3
固件版本	OpenWrt 18.06-SNAPSHOT r7275-e302272765 / LuCI openwrt-18.06 branch (git-18.235.62437-6503756)
内核版本	4.14.66
跑2个小时左右重启了，比假死好，重启后服务可以恢复。

huanghaizhuang, 2018-08-29:
跑2个小时 重启。要看怎么跑，如果是一直大流量 测试下载测试，这种重启是正常。linux再生kill机制。如果普通用着用着。假死机。这种是内存泄露！是要修改内核文件 ！USB和PCI服务有关的。包括7612E等这类都有这现象，，在用内核4.9以上会解决 内存泄露问题！

liyiwu, 2018-08-30:
编译的只是基本系统，没有其他服务。没做压力测试，只是模拟正常使用，开个网页视频，正常上网浏览，一个小时1.2G流量，全部用5G wifi跑的。结果可以接受，放家里用了。以后慢慢加上其他功能，跑一段时间再回来反馈。

hcym, 2018-08-30:
现在分支是67了，还是再看看吧，两个小时左右不能接受

hcym, 2018-09-12:
没有人折腾了，估计还是没戏

lintel, 2018-10-09:
> 有几个发现:
> 
> 1. 将SKB_BUF_MINIMUN_SIZE增大到2048可以解决处理skb时导致的kernel panic（开启slub_debug后显示redzone overwritten，overwrite了32byte的0xBB） 的问题
> 2. out of memory 的问题与DBDC有关。仅执行ifconfig ra0 up不会出现可用内存减少的现象，但只要执行ifconfig rax0 up, 无论之前是否执行了ifconfig ra0 up，均会导致可用内存减少，而且即使在执行了ifconfig rax0 down后，可用内存依然会不断减少
> 3. 同样的编译选项及配置文件，Openwrt 15.05 (Linux kernel 3.18)不会出现内存不断减少的现象
> 4. 保持网络中有持续的流量（正常上网即可）无论是无线或有线的wan<->lan, lan<->lan有流量, 可用内存都会不断地在下降后回升
> 
> @Nossiac 可以提供一下debug的思路吗？

你的提示解决了我另一个困扰许久的问题 :-)

hanwckf, 2018-10-09:
> 你的提示解决了我另一个困扰许久的问题 :-)

看来潘多拉7615驱动有希望了，l大这么晚还没睡

hcym, 2018-10-11:
死心塌地了，还是用老毛子，除了没有v2ray，其他都不错的

lincchenl, 2018-10-17:
> 死心塌地了，还是用老毛子，除了没有v2ray，其他都不错的

我现在用的无灯的固件，v2ray完全可以脚本外置的。把v2ray所有文件放到外置的/opt/v2ray文件夹。然后开机脚本加入v2ray（dokodemo-door),最后在防火墙脚本里加入一个透明代理的规则就是了。我现在用的很爽。

hcym, 2018-10-18:
也试试看看，无灯的无线是稳定，就是web页面偶尔失效

haswelliris, 2018-10-18:
其实用lede主要原因是用ip route工具，内核态的ipip和udp tun，这个在openwrt15，老毛子，潘多拉都没有编译进去，3.x的内核暴力强装也不行

hanwckf, 2018-11-13:
> 有几个发现:
>
> 1. 将SKB_BUF_MINIMUN_SIZE增大到2048可以解决处理skb时导致的kernel panic（开启slub_debug后显示redzone overwritten，overwrite了32byte的0xBB） 的问题
> 2. out of memory 的问题与DBDC有关。仅执行ifconfig ra0 [...]

看来这个Bug在4.x内核容易出现，目前用mtk的V4.4.2.1泄露驱动在Padavan上面可以正常工作，大家测试暂未发现有你所说的内存异常的情况。

hcym, 2018-11-14:
专心padavan做ap用吧，这个7615

Pentiumluyu, 2018-12-05:
@lintel @hanwckf 内存逐渐变少的问题确定是与有线驱动有关，在这个commit之后就完全解决了。https://github.com/openwrt/openwrt/commit/0c285bd081da55bd63da18f7596e7107a46bb798
redzone overwritten的问题不知道改SKB_BUF_MINIMUN_SIZE这样解决是不是正确？
因为CC的有线驱动有独立的page_frag_cache，CC没有这个问题。
另，貌似4 4 2 1 的2G和5G速度均比 4 4 1 2略慢？而且4 4 2 1貌似还没打上krack的patch？

Pentiumluyu, 2018-12-06:
@hanwckf 试过是这样，不知道是什么原因。另，mu-mimo貌似没有作用？

liyiwu, 2018-12-24:
我用master分支4.14.66内核编译一个新固件，用了一天没重启，稳了

thisdk, 2018-12-24:
牛逼(破音)~~~

MeIsReallyBa, 2018-12-26:
> @Nossiac @haswelliris 确认不使用`MULTI_INF_SUPPORT`,`RT_SECOND_CARD`,`RT_THIRD_CARD`选项依旧会有内存变少的问题。其实只要执行ifconfig ra0 up或者ifconfig [...]

这里有个7615驱动，不知道是否有用
https://drive.google.com/open?id=1Ck7QHCUxFkSropZ4pyVP70HGW_yoGQSg

haswelliris, 2018-12-27:
https://github.com/Nossiac/mtk-openwrt-feeds/issues/55#issuecomment-449948940
@lintel @hanwckf

[53 more replies]