Nossiac/mtk-openwrt-feeds #57: MT7628 kernel 4.14.37 support

agusalex, 2018-05-02:
I tried using kernel version 4.14.34 to see if it would work on a WR841ND v13 router (only supported on Lede's Master branch for now) and when I executed the command "modprobe mt7628" it would throw "failed to mmap /lib/modules/4.14.37/mt7628.ko". 
So I downloaded the latest version from your page (http://nossiac.com/download/mtk-wifi-ko/mt7628-for-mt76x8-linux-4.14.34.ko) and then retried with that one, the wifi light went on for a split second and then it rebooted.

agusalex, 2018-05-02:
Update: I then tried building with kernel 4.14.34 from a previous lede commit and had the same result.

Nossiac, 2018-05-07:
I remember that I made a firmware last month for wrt841nv13, you can try it: [http://nossiac.com/download/tmp/](http://nossiac.com/download/tmp/)

agusalex, 2018-05-07:
My mistake I know realised 841v13 has 7603 wifi chip not 7628.  That helps a ton! Thank you very much!

agusalex, 2018-05-08:
Hi, I know tried with your firmware and by default it had no driver installed as  so I downloaded the appropriate one for my kernel mt7603-for-mt76x8-linux-4.14.32.ko  saved into /lib/modules/4.14.32/mt7603.ko and did modprobe mt7603. The driver run without a problem
 `root@OpenWrt:~# lsmod | grep mt
mt7603               1421820  0
`
But when I do ifconfig -a no wireless interface shows up neither ra0 nor wlan. And naturally ifconfig ra0 up throws `ifconfig: SIOCGIFFLAGS: No such device
`

Nossiac, 2018-05-09:
@agusalex I guess there must be some misunderstanding. MT7628 is a SoC with wifi chip integrated inside, there's no reason to add another mt7603 to it. You should check the device again.

The reason I built 7603 with 76x8 is only because the automation script was simply written like that.