Nossiac/mtk-openwrt-feeds #58: LUCI pages are broken on current openwrt master

nekromant, 2018-05-05:
When trying to access MTK luci tab the following error shows up. Proprietary kernel modules for 7603 & 7612 are loaded and I can access rai0 and ra0 via commandline. 

``/usr/lib/lua/luci/dispatcher.lua:413: Failed to execute function dispatcher target for entry '/admin/mtk'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:310: invalid value (table) at index 1 in table for 'concat'
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:413: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:99: in function </usr/lib/lua/luci/dispatcher.lua:98>
``

kennylam777, 2018-05-06:
I encountered this too.... What theme are you using?

I tired to change following 2 parameters and then it worked, but I still don't know which is the "key" change

1. Change the theme from Legacy OpenWRT to bootstrap/material
2. Enable luci-json-rpc

As I only started to play with this library just 1 day, I am not familiar with it still.

nekromant, 2018-05-06:
Thanks for the tip. I was using legacy openwrt theme, luci-json-rpc is active. Changing to bootstrap seems to have fixed the problem.

Nossiac, 2018-05-07:
How did you get the error message? Which exactly version of LuCI were you using?

This is the version I'm tried (installed by scripts/feeds):
> Powered by LuCI Master (git-18.125.76267-055b621) / OpenWrt SNAPSHOT r6835-e495a05

nekromant, 2018-05-07:
At the time when I checked. 
> Powered by LuCI Master (git-18.123.23126-b0f3946) / OpenWrt SNAPSHOT r6808-3a456683e5

Nossiac, 2018-05-07:
My luci version tag was from openwrt master branch, and I used "scripts/feeds update“ to make sure luci components get updated before building the firmware.

nekromant, 2018-05-07:
Same here, I'm using openwrt master and update luci before every build.

Nossiac, 2018-05-09:
@nekromant how did you get that error? I tried it using master firmware, found nothing wrong.

nekromant, 2018-05-09:
Okay, found it. Steps to reproduce:
 
- Switch to legacy openwrt theme
- MTK tab will have no dropdown menu
- Click the MTK tab

And there it shows up.

Nossiac, 2018-05-10:
OK, the latest luci I installed comes with bootstrap theme only. I'll have a try on openwrt theme later.

Nossiac, 2018-05-10:
@nekromant @kennylam777  I found the issue, and pushed a fix: 
https://github.com/Nossiac/mtk-openwrt-feeds/commit/5de9aae8a045c767dc3d5eab1c807c8193e79be3

The openwrt theme looks terrible, I'm sticking to bootstrap theme.

nekromant, 2018-05-10:
@Nossiac Great, thanks!

closed 2018-05-10 by nekromant