Nossiac/mtk-openwrt-feeds #59: MT7623A driver

BGFWDev, 2018-05-08:
Is there a 4.14 driver now for the MT7623A? With MT7615N and E support? Currently have a 4.14 build but it appears to default to the MT7623N and not the A?

Nossiac, 2018-05-25:
7623 is quite rare on the market, but sure, I'll give it a try next week.

BGFWDev, 2018-05-26:
Any help is appreciated

anonimou0, 2018-12-05:
@BGFWDev were you able to get it running?

Thanks

BGFWDev, 2018-12-05:
Barely, still  having issues getting AP mode  working at  5GHz (AC)  with the MT7612  and MT7615  cards

anonimou0, 2018-12-05:
I've a Unielec U7623 running with MT7623A too.

Did you compile the main openwrt from its github or how did you the kernel 4.14.23?

BGFWDev, 2018-12-05:
We are producing a build of of the  U7623 build  posted on Github as well.

anonimou0, 2018-12-05:
Would you mind sharing this build with the MT7615 driver on it?

Thanks

BGFWDev, 2018-12-05:
We are leveraging what is already public, it is not working in AP mode.