Nossiac/mtk-openwrt-feeds #61: Kernel panic under high traffic load(mt7615-for-mt7621)

Pentiumluyu, 2018-05-09:
Router: Phicomm K2P(mt7621 + one mt7615)
Linux version: 4.14.37(OpenWrt trunk branch)
Module: mt7615-for-mt7621-linux-4.14.34.ko
I have extracted the "Factory" partition into /lib/firmware/mt7615.1.eeprom.bin, and copied both "mt7615-sku.dat" and "mt7615-sku-bf.dat" from stock firmware into /etc/wireless/mt7615.
Thanks for providing these drivers. The driver mt7615-for-mt7621 actually works like a charm! Mac addresses of interface ra0 and rax0 are correct. Both 5G and 2.4G signal are strong. WiFi latency is low.
But under relatively high traffic load, like doing 5G bandwidth testing with iperf, the driver will cause kernel panic. It's kind of easy to reproduce, as it usually only takes two minutes of bandwidth testing(about 200Mbit/s) to cause kernel panic. I tried to change the content of /proc/irq/$ra0 interrupt_number/smp_affinity, and found it would cause the ksoftirqd process to change. For example, if irq of ra0 is restrict to cpu1, ksoftirqd/1 will cause kernel panic. Or maybe the trunk branch of OpenWrt is  unstable under high traffic load? I'm not sure how to pinpoint the causes.....
Here is the content of one crashlog(from /sys/kernel/debug/crashlog).

> Modules: pppoe@87378000+2290	ppp_async@8734c000+1be0	pppox@8735f000+58a	ppp_generic@87388000+5aa2	nf_conntrack_ipv6@87328000+15a0	iptable_nat@8737f000+310	ipt_REJECT@8737c000+410	ipt_MASQUERADE@87374000+2f0	xt_time@8736b000+730	xt_tcpudp@87367000+750	xt_tcpmss@87364000+470	xt_statistic@8735e000+390	xt_state@8735a000+310	xt_nat@87353000+650	xt_multiport@8734f000+550	xt_mark@87344000+2f0	xt_mac@87341000+2b0	xt_limit@8733a000+4f0	xt_length@87337000+310	xt_hl@8732c000+370	xt_ecn@87322000+590	xt_dscp@8731e000+430	xt_conntrack@87314000+950	xt_comment@87310000+230	xt_TPROXY@8730e000+1210	xt_TCPMSS@87305000+b30	xt_REDIRECT@872fc000+310	xt_LOG@872f9000+350	xt_HL@872f1000+590	xt_FLOWOFFLOAD@872ef000+a20	xt_DSCP@872e2000+630	xt_CLASSIFY@872e0000+290	slhc@872e4000+149b	nf_reject_ipv4@872d6000+883	nf_nat_redirect@872d8000+5fb	nf_nat_masquerade_ipv4@872dc000+67c	nf_conntrack_ipv4@8724c000+1410	nf_nat_ipv4@872b6000+dd1	nf_nat@86d7c000+381a	nf_log_ipv4@8727f000+dd0	nf_flow_table_hw@87134000+920	nf_flow_table@87268000+34bc	nf_defrag_ipv6@872a8000+342e	nf_defrag_ipv4@87299000+436	nf_conntrack_rtcache@86d6a000+af0	nf_conntrack_netlink@872b0000+5750	nf_conntrack@872c0000+10ce1	iptable_mangle@86fd5000+3d0	iptable_filter@8728a000+2d0	ipt_ECN@87212000+5d0	ip_tables@8729c000+290d	crc_ccitt@87223000+42b	xt_set@87296000+1b70	ip_set_list_set@8728e000+1c50	ip_set_hash_netiface@87280000+6290	ip_set_hash_netport@87278000+6250	ip_set_hash_netnet@87270000+6790	ip_set_hash_net@87260000+5c70	ip_set_hash_netportnet@87118000+6d30	ip_set_hash_mac@87230000+2930	ip_set_hash_ipportnet@87110000+6830	ip_set_hash_ipportip@87108000+54f0	ip_set_hash_ipport@87100000+5170	ip_set_hash_ipmark@87258000+4e90	ip_set_hash_ip@87250000+4db0	ip_set_bitmap_port@87236000+1830	ip_set_bitmap_ipmac@8713a000+1bd0	ip_set_bitmap_ip@86d66000+1b30	ip_set@87128000+590a	nfnetlink@87122000+1277	ip6t_REJECT@86d65000+450	nf_reject_ipv6@86d56000+a08	nf_log_ipv6@87200000+1070	nf_log_common@86d58000+acf	ip6table_mangle@87140000+4d0	ip6table_filter@86d69000+2d0	ip6_tables@86d78000+2801	x_tables@86d70000+374f	mt7615@86000000+3c3b34	leds_gpio@87706000+bb0	gpio_button_hotplug@87708000+1930
<4>[11315.048145]         ...
<4>[11315.053002] Call Trace:
<4>[11315.057889] [<80010498>] show_stack+0x58/0x100
<4>[11315.066750] [<80429a2c>] dump_stack+0x9c/0xe0
<4>[11315.075431] [<80050604>] __schedule_bug+0x60/0x7c
<4>[11315.084791] [<804421d4>] __schedule+0x6c/0x854
<4>[11315.093623] [<80442a28>] schedule+0x6c/0x84
<4>[11315.101947] [<8044580c>] schedule_timeout+0x38/0x1b8
<4>[11315.111842] [<800658f4>] wait_woken+0x60/0x90
<4>[11315.120514] [<8030c3c8>] sk_wait_data+0xc4/0x140
<4>[11315.129702] [<80379c30>] tcp_recvmsg+0x45c/0x8ec
<4>[11315.138887] [<803a5db8>] inet_recvmsg+0xa8/0xc4
<4>[11315.147928] [<80309b2c>] SyS_recvfrom+0xc0/0x138
<4>[11315.157126] [<80019578>] syscall_common+0x34/0x58
<3>[11319.393048] softirq: huh, entered softirq 0 HI 80032260 with preempt_count 00000101, exited with fffffe00?
<3>[11324.407206] BUG: scheduling while atomic: iperf/4034/0xfffffa00
<4>[11324.419006] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_conntrack xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CLASSIFY slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_netlink nf_conntrack iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet
<4>[11324.561226]  ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables mt7615(P) leds_gpio gpio_button_hotplug
<4>[11324.614738] CPU: 0 PID: 4034 Comm: iperf Tainted: P        W       4.14.37 #0
<4>[11324.628924] Stack : 00000000 00000000 00000000 00000000 805b7ad2 00000041 00000000 00000000
<4>[11324.645555]         857e9634 805508a7 804e19cc 00000000 00000fc2 00000001 852d1de0 532616c4
<4>[11324.662185]         00000000 00000000 805b0000 000090e0 00000000 00000ae4 00000007 00000000
<4>[11324.678815]         00000000 80550000 00096152 00000000 00000000 00000000 80570000 805a9d80
<4>[11324.695446]         00000000 00000000 77e46000 00000000 00000000 8027cfc8 001c5eaf 001bc0af
<4>[11324.712075]         ...
<4>[11324.716930] Call Trace:
<4>[11324.721819] [<80010498>] show_stack+0x58/0x100
<4>[11324.730673] [<80429a2c>] dump_stack+0x9c/0xe0
<4>[11324.739353] [<80050604>] __schedule_bug+0x60/0x7c
<4>[11324.748706] [<804421d4>] __schedule+0x6c/0x854
<4>[11324.757542] [<80442a28>] schedule+0x6c/0x84
<4>[11324.765855] [<8000b19c>] work_resched+0x8/0x2c
<3>[11327.016592] BUG: scheduling while atomic: iperf/4034/0xfffffe00
<4>[11327.028403] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_conntrack xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CLASSIFY slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_netlink nf_conntrack iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet
<4>[11327.170596]  ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables mt7615(P) leds_gpio gpio_button_hotplug
<4>[11327.224138] CPU: 3 PID: 4034 Comm: iperf Tainted: P        W       4.14.37 #0
<4>[11327.238327] Stack : 00000000 00000000 00000000 00000000 805b7ad2 00000041 00000000 00000000
<4>[11327.254959]         857e9634 805508a7 804e19cc 00000003 00000fc2 00000001 852d1b78 532616c4
<4>[11327.271589]         00000000 00000000 805b0000 00009958 00000000 00000af5 00000007 00000000
<4>[11327.288218]         00000000 80550000 00036b8a 00000000 00000000 00000000 80570000 805a9d80
<4>[11327.304850]         845df81c 00000000 00000000 00000000 00000000 8027cfc8 0000000c 805b000c
<4>[11327.321481]         ...
<4>[11327.326341] Call Trace:
<4>[11327.331225] [<80010498>] show_stack+0x58/0x100
<4>[11327.340077] [<80429a2c>] dump_stack+0x9c/0xe0
<4>[11327.348759] [<80050604>] __schedule_bug+0x60/0x7c
<4>[11327.358119] [<804421d4>] __schedule+0x6c/0x854
<4>[11327.366955] [<80442a28>] schedule+0x6c/0x84
<4>[11327.375277] [<8044580c>] schedule_timeout+0x38/0x1b8
<4>[11327.385177] [<800658f4>] wait_woken+0x60/0x90
[... cut ...]
<4>[11489.779557] [<8032b580>] net_rx_action+0x150/0x30c
<4>[11489.789101] [<80446c08>] __do_softirq+0x128/0x2ec
<4>[11489.798466] [<80032638>] run_ksoftirqd+0x38/0x6c
<4>[11489.807656] [<8004eb18>] smpboot_thread_fn+0x1a8/0x1d8
<4>[11489.817883] [<8004b358>] kthread+0x130/0x144
<4>[11489.826378] [<8000b098>] ret_from_kernel_thread+0x14/0x1c
<4>[11489.837109] Code: 10000002  2496e8e4  8e020014 <8c4300b0> 02a01025  ae030074  8f83000c  00031880  02831821 
<4>[11489.856513] 
<4>[11489.859801] ---[ end trace 05f1e5ca7e641858 ]---

==============

haoyixin, 2018-05-11:
@Pentiumluyu Try run iperf c/s on wifi client, it's making no sense for wireless performance benchmark that run it on the router, and it make the stack more complex for debugging.

haswelliris, 2018-05-14:
请问你的mt7615.1.eeprom.bin和mt7615-sku.dat是从哪里提取出来的。我试过用官方固件以及我刷过的openwrt15.05cc解包提取的都不行，用ifconfig -a 查看mac地址都是0

haoyixin, 2018-05-16:
@haswelliris eeprom在factory分区中，sku在官方固件里，只改mac的话，eeprom 5-10 bytes就是，直接改就行

Pentiumluyu, 2018-05-18:
While iperf was not running on the router, the same issue happened. Here is one crashlog.
```Time: 1525974417.239920
Modules:	pppoe@86d84000+2290	ppp_async@86d5e000+1be0	pppox@86d93000+58a	ppp_generic@86da8000+5aa2	nf_conntrack_ipv6@86d96000+15a0	iptable_nat@8723b000+310	ipt_REJECT@86d9f000+410	ipt_MASQUERADE@86d9c000+2f0	xt_time@86d94000+730	xt_tcpudp@86d92000+750	xt_tcpmss@86d82000+470	xt_string@86d7f000+310	xt_statistic@86d7c000+390	xt_state@86d74000+310	xt_quota@86d72000+350	xt_pkttype@86d68000+290	xt_owner@86d65000+3b0	xt_nat@86d5b000+650	xt_multiport@86d59000+550	xt_mark@86d52000+2f0	xt_mac@86d4e000+2b0	xt_limit@86d47000+4f0	xt_length@86d43000+310	xt_hl@86d33000+370	xt_ecn@86ca1000+590	xt_dscp@86d2d000+430	xt_conntrack@86d27000+950	xt_comment@86d22000+230	xt_addrtype@86d10000+970	xt_TPROXY@86d16000+1210	xt_TCPMSS@86d0a000+b30	xt_REDIRECT@86d03000+310	xt_LOG@8721a000+350	xt_HL@87217000+590	xt_FLOWOFFLOAD@87214000+a20	xt_DSCP@8720c000+630	xt_CLASSIFY@87203000+290	ts_kmp@87200000+570	ts_fsm@87226000+ab0	ts_bm@86cdb000+5d0	slhc@87220000+149b	nf_reject_ipv4@87239000+883	nf_nat_redirect@86c80000+5fb	nf_nat_masquerade_ipv4@86cb9000+67c	nf_conntrack_ipv4@86c86000+1410	nf_nat_ipv4@86c8b000+dd1	nf_nat@86cb4000+381a	nf_log_ipv4@86f21000+dd0	nf_flow_table_hw@86c95000+920	nf_flow_table@8723c000+34bc	nf_defrag_ipv6@86cbc000+342e	nf_defrag_ipv4@873fb000+436	nf_conntrack_rtcache@86ca5000+af0	nf_conntrack_netlink@86ce0000+5750	nf_conntrack@86cc0000+10ce1	mt7615@86000000+3c3b34	iptable_mangle@877ae000+3d0	iptable_filter@86e32000+2d0	ipt_ECN@86e0f000+5d0	ip_tables@86e34000+290d	crc_ccitt@877af000+42b	xt_set@86e2a000+1b70	ip_set_list_set@86e1e000+1c50	ip_set_hash_netiface@86e20000+6290	ip_set_hash_netport@86e10000+6250	ip_set_hash_netnet@86e00000+6790	ip_set_hash_net@877f8000+5c70	ip_set_hash_netportnet@877f0000+6d30	ip_set_hash_mac@877e8000+2930	ip_set_hash_ipportnet@877e0000+6830	ip_set_hash_ipportip@877d8000+54f0	ip_set_hash_ipport@877c8000+5170	ip_set_hash_ipmark@877c0000+4e90	ip_set_hash_ip@877b8000+4db0	ip_set_bitmap_port@877aa000+1830	ip_set_bitmap_ipmac@877a6000+1bd0	ip_set_bitmap_ip@877a2000+1b30	ip_set@87798000+590a	nfnetlink@8778c000+1277	ip6t_REJECT@87756000+450	nf_reject_ipv6@86fe2000+a08	nf_log_ipv6@86fee000+1070	nf_log_common@86fd2000+acf	ip6table_mangle@87781000+4d0	ip6table_filter@86feb000+2d0	ip6_tables@86ff8000+2801	x_tables@86fdc000+374f	leds_gpio@876fb000+bb0	gpio_button_hotplug@876fe000+1930
<4>[   37.262911] 9a
<4>[   37.266235] 2a
<4>[   37.269534] 16
<4>[   37.272830] d3
<4>[   37.276152] fc
<4>[   37.279451] 90
<4>[   37.282744] 2d
<4>[   37.286062] 
<4>[   37.445700] Rcv Wcid(2) AddBAReq
<4>[   37.452134] Start Seq = 00000000
<4>[   37.465799] AP SETKEYS DONE - AKMMap=WPA2PSK, PairwiseCipher=AES, GroupCipher=AES, wcid=2 from 70:14:A6:BF:8E:B6
<4>[   37.465799] 
<4>[   37.489003] PTK:
<4>[   37.489011] db
<4>[   37.492649] 32
<4>[   37.495986] 6a
<4>[   37.499285] fa
<4>[   37.502578] 52
<4>[   37.505901] a6
<4>[   37.509206] 06
<4>[   37.512499] 9b
<4>[   37.515823] 72
<4>[   37.519130] c0
<4>[   37.522461] db
<4>[   37.525892] 39
<4>[   37.529216] a9
<4>[   37.532510] f7
<4>[   37.535838] 69
<4>[   37.539137] 73
<4>[   37.542441] 81
<4>[   37.545826] f5
<4>[   37.549139] 43
<4>[   37.552434] d1
<4>[   37.556168] ae
<4>[   37.559465] 51
<4>[   37.562756] 5c
<4>[   37.566084] d2
<4>[   37.569383] a0
<4>[   37.572680] 37
<4>[   37.576022] 29
<4>[   37.579327] b5
<4>[   37.582621] d1
<4>[   37.585953] 4a
<4>[   37.589254] 54
<4>[   37.592548] d9
<4>[   37.595871] 5b
<4>[   37.599170] 47
<4>[   37.602463] 55
<4>[   37.605792] a1
<4>[   37.609093] f2
<4>[   37.612386] 5a
<4>[   37.615711] db
<4>[   37.619010] 05
<4>[   37.622303] ab
<4>[   37.625631] e0
<4>[   37.628932] 24
<4>[   37.632225] 77
<4>[   37.635550] 65
<4>[   37.638853] 62
<4>[   37.642147] c5
<4>[   37.645470] 56
<4>[   37.648770] fd
<4>[   37.652063] 57
<4>[   37.655384] 15
<4>[   37.658683] 75
<4>[   37.661976] 51
<4>[   37.665385] 42
[... 391 lines cut ...]
<4>[  353.895053] PrId  : 0001992f (MIPS 1004Kc)
<4>[  353.903184] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_quota xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_conntrack xt_comment xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CLASSIFY ts_kmp ts_fsm ts_bm slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_netlink nf_conntrack mt7615(P) iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet
<4>[  354.045980]  ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables leds_gpio gpio_button_hotplug
<4>[  354.111045] Process swapper/0 (pid: 0, threadinfo=8053e000, task=805509a0, tls=00000000)
<4>[  354.127126] Stack : c081f1f4 00000004 00000002 00000000 80570000 80511ae8 80550000 80515188
[... cut ...]
<4>[  354.247191] [<80239d8c>] plat_irq_dispatch+0xfc/0x138
<4>[  354.257237] [<8000b5a8>] except_vec_vi_end+0xb8/0xc4
<4>[  354.267107] [<8000cf70>] r4k_wait_irqoff+0x1c/0x24
<4>[  354.276653] [<8006624c>] do_idle+0xe4/0x168
<4>[  354.284965] [<800664c8>] cpu_startup_entry+0x24/0x2c
<4>[  354.294836] [<80577bf0>] start_kernel+0x484/0x4a4
<4>[  354.304184] Code: 00000000  8e020004  8e030000 <ac620004> ac430000  ae100000  8e110008  ae100004  32310001 
<4>[  354.323584] 
<4>[  354.326881] ---[ end trace bc0af7320af0f9c0 ]---

===============```

Pentiumluyu, 2018-05-27:
Do not happen after upgrading the kernel to 4.14.43. This issue may be related to bug in the kernel.
#67

closed 2018-05-27 by Pentiumluyu