Nossiac/mtk-openwrt-feeds #63: mt7602

MaFeSa, 2018-05-09:
Hi,
I can not find mt7602, in the wiki there is written supported ... where am I wrong?

Thanks

Nossiac, 2018-05-10:
7602 shares driver with 7612. They are actually unified as mt76x2 in MTK's source code.

MaFeSa, 2018-06-01:
I tried to compile lede-1701, now it does not start but the wifi does not transmit.
I extracted mt7612e.eeprom.bin correctly which contains all mac address but ra0 HW address is always 00:00:00:00:00:00

```
root@LEDE:~# ifconfig ra0
ra0       Link encap:Ethernet  HWaddr 00:00:00:00:00:00
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:31
```