Nossiac/mtk-openwrt-feeds #67: The Linux kernel of Openwrt snapshot have been upgraded to 4.14.41

Pentiumluyu, 2018-05-18:
The module mt7615-for-mt7621-linux-4.14.34.ko doesn't work on kernel version 4.14.41. It doesn't load mac address from /lib/firmware/xx.bin. Bring up a wireless interface will cause kernel panic. Could you please update all the modules to support kernel version 4.14.41? Thanks a lot.

Pentiumluyu, 2018-05-19:
But the module may cause memory leaks.
```Time: 1526716697.173463
Modules:	pppoe@8737c000+22b0	ppp_async@8718c000+1be0	pppox@87381000+58a	ppp_generic@873a0000+5aa2	nf_conntrack_ipv6@87372000+15a0	iptable_nat@87398000+310	ipt_REJECT@8738e000+410	ipt_MASQUERADE@87387000+2f0	xt_time@87383000+730	xt_tcpudp@87380000+750	xt_tcpmss@87379000+470	xt_string@87376000+310	xt_statistic@8736f000+390	xt_state@8736b000+310	xt_quota@87360000+350	xt_pkttype@8735d000+290	xt_owner@87356000+3b0	xt_nat@87352000+650	xt_multiport@8734b000+550	xt_mark@87348000+2f0	xt_mac@8733f000+2b0	xt_limit@8733c000+4f0	xt_length@87338000+310	xt_hl@8732d000+370	xt_ecn@87322000+590	xt_dscp@8719a000+430	xt_conntrack@87198000+950	xt_comment@8718a000+230	xt_addrtype@87188000+970	xt_TPROXY@8718e000+1210	xt_TCPMSS@87180000+b30	xt_REDIRECT@8731c000+310	xt_LOG@87313000+350	xt_HL@87310000+590	xt_FLOWOFFLOAD@872df000+a20	xt_DSCP@872dc000+630	xt_CLASSIFY@871b2000+290	ts_kmp@871af000+570	ts_fsm@86dff000+ab0	ts_bm@86dfb000+5d0	slhc@871a8000+149b	nf_reject_ipv4@871a6000+883	nf_nat_redirect@872d6000+5fb	nf_nat_masquerade_ipv4@86f9f000+67c	nf_conntrack_ipv4@8730a000+1410	nf_nat_ipv4@872ff000+dd1	nf_nat@87290000+381a	nf_log_ipv4@872b7000+dd0	nf_flow_table_hw@86df6000+920	nf_flow_table@871b8000+34bc	nf_defrag_ipv6@872e8000+342e	nf_defrag_ipv4@87732000+436	nf_conntrack_rtcache@87299000+af0	nf_conntrack_netlink@872b0000+5750	nf_conntrack@872c0000+10ce1	mt7615@86000000+3c3b34	iptable_mangle@877a5000+3d0	iptable_filter@86e0f000+2d0	ipt_ECN@877f2000+5d0	ip_tables@86e10000+290d	crc_ccitt@8778c000+42b	xt_set@86e06000+1b70	ip_set_list_set@86e02000+1c50	ip_set_hash_netiface@877f8000+6290	ip_set_hash_netport@877e8000+6250	ip_set_hash_netnet@877e0000+6790	ip_set_hash_net@877d8000+5c70	ip_set_hash_netportnet@877d0000+6d30	ip_set_hash_mac@877bc000+2930	ip_set_hash_ipportnet@877c0000+6830	ip_set_hash_ipportip@877b0000+54f0	ip_set_hash_ipport@877a8000+5170	ip_set_hash_ipmark@877a0000+4e90	ip_set_hash_ip@87798000+4db0	ip_set_bitmap_port@87786000+1830	ip_set_bitmap_ipmac@87782000+1bd0	ip_set_bitmap_ip@86ff6000+1b30	ip_set@86ff8000+590a	nfnetlink@86fea000+1277	ip6t_REJECT@8776d000+450	nf_reject_ipv6@87763000+a08	nf_log_ipv6@86f2a000+1070	nf_log_common@8774f000+acf	ip6table_mangle@8776e000+4d0	ip6table_filter@87766000+2d0	ip6_tables@86fe4000+2801	x_tables@86fe0000+374f	leds_gpio@876e3000+bb0	gpio_button_hotplug@876e4000+1930
<4>[ 1740.314954] 79
<4>[ 1740.318264] 89
<4>[ 1740.321558] 36
<4>[ 1740.324849] 74
<4>[ 1740.328161] ae
<4>[ 1740.331454] eb
<4>[ 1740.334745] 26
<4>[ 1740.338054] dd
<4>[ 1740.341351] 
<4>[ 1740.347650] Rcv Wcid(6) AddBAReq
<4>[ 1740.354064] Start Seq = 00000000
<4>[ 1740.444893] Rcv Wcid(6) AddBAReq
<4>[ 1740.451341] Start Seq = 00000003
<6>[ 1761.938836] mtk_soc_eth 1e100000.ethernet eth0: port 1 link down
<6>[ 1764.785489] mtk_soc_eth 1e100000.ethernet eth0: port 1 link up
<4>[ 1769.757143] dropbear invoked oom-killer: gfp_mask=0x14201ca(GFP_HIGHUSER_MOVABLE|__GFP_COLD), nodemask=(null),  order=0, oom_score_adj=0
<4>[ 1769.781594] CPU: 3 PID: 1272 Comm: dropbear Tainted: P                4.14.41 #0
<4>[ 1769.796297] Stack : 00000000 00000000 00000000 00000000 805b7ad2 00000044 00000000 00000000
<4>[ 1769.812925]         87d59634 805508a7 804e0a5c 00000003 000004f8 00000001 8509db20 532616c8
<4>[ 1769.829552]         00000000 00000000 805b0000 00018040 00000000 00000757 00000007 00000000
<4>[ 1769.846179]         00000000 80550000 000bed1a 00000000 80000000 00000000 80570000 804ea83c
<4>[ 1769.862805]         00000000 00200000 ffffffff 0000000a 00000003 8027cfa8 0000000c 805b000c
<4>[ 1769.879432]         ...
<4>[ 1769.884288] Call Trace:
<4>[ 1769.889170] [<80010498>] show_stack+0x58/0x100
<4>[ 1769.898019] [<804299cc>] dump_stack+0x9c/0xe0
<4>[ 1769.906679] [<800c42b4>] dump_header.isra.6+0xa4/0x200
<4>[ 1769.916896] [<800c343c>] oom_kill_process+0xf0/0x544
<4>[ 1769.926762] [<800c4138>] out_of_memory+0x3cc/0x440
<4>[ 1769.936290] [<800c87dc>] __alloc_pages_nodemask+0x944/0xc4c
<4>[ 1769.947396] [<800c0d04>] filemap_fault+0x458/0x658
<4>[ 1769.956923] [<800e8810>] __do_fault+0x24/0xa0
<4>[ 1769.965584] [<800ed1c4>] handle_mm_fault+0x694/0xb08
<4>[ 1769.975471] [<8001c3b0>] __do_page_fault+0x120/0x478
<4>[ 1769.985342] [<80021fb8>] tlb_do_page_fault_0+0x118/0x120
<4>[ 1769.996080] Mem-Info:
<4>[ 1770.000629] active_anon:69 inactive_anon:127 isolated_anon:0
<4>[ 1770.000629]  active_file:1 inactive_file:5 isolated_file:0
<4>[ 1770.000629]  unevictable:0 dirty:0 writeback:0 unstable:0
<4>[ 1770.000629]  slab_reclaimable:355 slab_unreclaimable:2154
<4>[ 1770.000629]  mapped:14 shmem:137 pagetables:12 bounce:0
<4>[ 1770.000629]  free:3556 free_pcp:0 free_cma:0
<4>[ 1770.063098] Node 0 active_anon:276kB inactive_anon:508kB active_file:0kB inactive_file:16kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:56kB dirty:0kB writeback:0kB shmem:548kB writeback_tmp:0kB unstable:0kB all_unreclaimable? no
<4>[ 1770.106532] Normal free:14288kB min:16384kB low:20480kB high:24576kB active_anon:276kB inactive_anon:508kB active_file:176kB inactive_file:16kB unevictable:0kB writepending:0kB present:131072kB managed:123752kB mlocked:0kB kernel_stack:480kB pagetables:48kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB
<4>[ 1770.160667] lowmem_reserve[]: 0 0 0
<4>[ 1770.167625] Normal: 191*4kB (UM) 175*8kB (UM) 98*16kB (UM) 42*32kB (UM) 29*64kB (UM) 16*128kB (UM) 11*256kB (UM) 5*512kB (UM) 0*1024kB 0*2048kB 0*4096kB = 14356kB
[... cut ...]
<4>[ 1772.046904] 156 total pagecache pages
<4>[ 1772.054175] 0 pages in swap cache
<4>[ 1772.060817] Swap cache stats: add 0, delete 0, find 0/0
<4>[ 1772.071220] Free swap  = 0kB
<4>[ 1772.076947] Total swap = 0kB
<4>[ 1772.082665] 32768 pages RAM
<4>[ 1772.088220] 0 pages HighMem/MovableOnly
<4>[ 1772.095841] 1830 pages reserved
<6>[ 1772.102094] [ pid ]   uid  tgid total_vm      rss nr_ptes nr_pmds swapents oom_score_adj name
<0>[ 1772.119092] Kernel panic - not syncing: Out of memory and no killable processes...
<0>[ 1772.119092]

Pentiumluyu, 2018-05-20:
The fuction `GetPacketFromRxRing` also caused some memory error, as these lines were in the crashlog:
`CPU 0 Unable to handle kernel paging request at virtual address 00000007, epc == 81d41138, ra == 81d41054`, 
`epc   : 81d41138 GetPacketFromRxRing+0x1e8/0x145c [mt7615]`, 
`ra    : 81d41054 GetPacketFromRxRing+0x104/0x145c [mt7615]`.

<details>
<summary>crashlog</summary>

```
Time: 1526799795.572095
Modules:	mt7615@81c00000+3c3b34	pppoe@87388000+22b0	ppp_async@87374000+1be0	pppox@87378000+58a	ppp_generic@873a8000+5aa2	nf_conntrack_ipv6@872ba000+15a0	iptable_nat@8739a000+310	ipt_REJECT@87395000+410	ipt_MASQUERADE@87390000+2f0	xt_time@87385000+730	xt_tcpudp@87382000+750	xt_tcpmss@8737b000+470	xt_string@87377000+310	xt_statistic@87370000+390	xt_state@8736d000+310	xt_quota@8736a000+350	xt_pkttype@87361000+290	xt_owner@87359000+3b0	xt_nat@87356000+650	xt_multiport@87354000+550	xt_mark@8734a000+2f0	xt_mac@87347000+2b0	xt_limit@8733f000+4f0	xt_length@8733c000+310	xt_hl@87333000+370	xt_ecn@87325000+590	xt_dscp@87322000+430	xt_conntrack@8719b000+950	xt_comment@8718d000+230	xt_addrtype@8718a000+970	xt_TPROXY@87192000+1210	xt_TCPMSS@87183000+b30	xt_REDIRECT@8731e000+310	xt_LOG@871ad000+350	xt_HL@871a9000+590	xt_FLOWOFFLOAD@872dd000+a20	xt_DSCP@872d8000+630	xt_CLASSIFY@871a5000+290	ts_kmp@872d6000+570	ts_fsm@872d1000+ab0	ts_bm@872ff000+5d0	slhc@872b8000+149b	nf_reject_ipv4@8773a000+883	nf_nat_redirect@872fe000+5fb	nf_nat_masquerade_ipv4@87706000+67c	nf_conntrack_ipv4@87312000+1410	nf_nat_ipv4@871b7000+dd1	nf_nat@872ac000+381a	nf_log_ipv4@871b2000+dd0	nf_flow_table_hw@872f0000+920	nf_flow_table@87304000+34bc	nf_defrag_ipv6@872f8000+342e	nf_defrag_ipv4@871bc000+436	nf_conntrack_rtcache@872a1000+af0	nf_conntrack_netlink@872e0000+5750	nf_conntrack@872c0000+10ce1	iptable_mangle@87795000+3d0	iptable_filter@86e19000+2d0	ipt_ECN@86e14000+5d0	ip_tables@86e1c000+290d	crc_ccitt@87796000+42b	xt_set@86e0e000+1b70	ip_set_list_set@86e08000+1c50	ip_set_hash_netiface@86e00000+6290	ip_set_hash_netport@877f8000+6250	ip_set_hash_netnet@877f0000+6790	ip_set_hash_net@877e0000+5c70	ip_set_hash_netportnet@877d8000+6d30	ip_set_hash_mac@877cc000+2930	ip_set_hash_ipportnet@877c0000+6830	ip_set_hash_ipportip@877b8000+54f0	ip_set_hash_ipport@877b0000+5170	ip_set_hash_ipmark@877a8000+4e90	ip_set_hash_ip@87798000+4db0	ip_set_bitmap_port@8778c000+1830	ip_set_bitmap_ipmac@87788000+1bd0	ip_set_bitmap_ip@86ffc000+1b30	ip_set@87780000+590a	nfnetlink@86ff0000+1277	ip6t_REJECT@8777f000+450	nf_reject_ipv6@87721000+a08	nf_log_ipv6@86f16000+1070	nf_log_common@86fd5000+acf	ip6table_mangle@86fdf000+4d0	ip6table_filter@86f07000+2d0	ip6_tables@86fe4000+2801	x_tables@86fbc000+374f	leds_gpio@876e3000+bb0	gpio_button_hotplug@876e0000+1930
<4>[ 1034.139655] RcRadioInit(): pRadioCtrl=8508153c,Band=1,rfcap=2,channel=36,PhyMode=1
<4>[ 1034.154950] MtCmdSetDbdcCtrl:(ret = 0)
<4>[ 1040.474777] Band Rf: 1, Phy Mode: 2
<4>[ 1040.481754] Band Rf: 2, Phy Mode: 1
<4>[ 1040.488717] AntCfgInit(2787): Not support for HIF_MT yet!
<4>[ 1040.499486] MtSingleSkuLoadParam: RF_LOCKDOWN Feature OFF !!!
<4>[ 1040.514541] MtBfBackOffLoadTable: RF_LOCKDOWN Feature OFF !!!
<4>[ 1040.526498] EEPROM Init Done!
<4>[ 1040.532429] mt_mac_init()-->
<4>[ 1040.538179] mt_mac_pse_init(2787): Don't Support this now!
<4>[ 1040.549103] mt7615_init_mac_cr()-->
<4>[ 1040.556041] mt7615_init_mac_cr(): TMAC_TRCR0=0x82783c8c
<4>[ 1040.566451] mt7615_init_mac_cr(): TMAC_TRCR1=0x82783c8c
<4>[ 1040.576889] MtAsicSetMacMaxLen(1313): Not finish Yet!
<4>[ 1040.586952] <--mt_mac_init()
<4>[ 1040.596619] CmdRxHdrTransBLUpdateRsp::EventExtCmdResult.u4Status = 0x0
<4>[ 1040.609896] CmdRxHdrTransBLUpdateRsp::EventExtCmdResult.u4Status = 0x0
<4>[ 1040.623162] CmdRxHdrTransBLUpdateRsp::EventExtCmdResult.u4Status = 0x0
<4>[ 1040.636234] MAC Init Done!
<4>[ 1040.641637] MT7615BBPInit():BBP Initialization.....
<4>[ 1040.651361] 	Band 0: valid=1, isDBDC=0, Band=2, CBW=1, CentCh/PrimCh=1/1, prim_ch_idx=0, txStream=2
<4>[ 1040.669388] 	Band 1: valid=0, isDBDC=0, Band=0, CBW=0, CentCh/PrimCh=0/0, prim_ch_idx=0, txStream=0
<4>[ 1040.687402] MT7615BBPInit() todo 
<4>[ 1040.693981] PHY Init Done!
<4>[ 1040.700948] tx_pwr_comp_init():NotSupportYet!
<4>[ 1040.710062] MtCmdSetMacTxRx:(ret = 0)
<4>[ 1040.717661] MtCmdSetMacTxRx:(ret = 0)
<4>[ 1040.725128] CountryCode(2.4G/5G)=5/7, RFIC=25, PHY mode(2.4G/5G)=14/49, support 38 channels
<4>[ 1040.741811] WifiSysOpen(), wdev idx = 0
<4>[ 1040.749469] wdev_attr_update(): wdevId0 = fc:7c:02:74:ba:d6
<4>[ 1040.760901] MtCmdSetDbdcCtrl:(ret = 0)
<4>[ 1040.769186] ApAutoChannelAtBootUp----------------->
<4>[ 1040.778911] ApAutoChannelAtBootUp: AutoChannelBootup = 0, AutoChannelFlag = 0
<4>[ 1040.793116] ApAutoChannelAtBootUp<-----------------
<4>[ 1040.802833] Current Channel is 157. DfsZeroWaitSupport=0
<4>[ 1040.813448] MtAsicSetChBusyStat(865): Not support for HIF_MT yet!
<4>[ 1040.825584] MtAsicSetChBusyStat(865): Not support for HIF_MT yet!
<4>[ 1040.837765] [PMF]APPMFInit:: apidx=0, MFPC=0, MFPR=0, SHA256=0
<4>[ 1040.849403] [PMF]WPAMakeRsnIeCap: RSNIE Capability MFPC=0, MFPR=0
<4>[ 1040.861552] HcUpdatePhyMode(): Update PhyMode for all wdev for this band PhyMode:49,Channel=157
<4>[ 1040.879019] CountryCode(2.4G/5G)=5/7, RFIC=25, PHY mode(2.4G/5G)=14/49, support 38 channels
<4>[ 1040.895746] wtc_acquire_groupkey_wcid: Found a non-occupied wtbl_idx:127 for WDEV_TYPE:1
<4>[ 1040.895746]  LinkToOmacIdx = 0, LinkToWdevType = 1
[... cut ...]
<4>[ 1043.491347] [<8000b5a8>] except_vec_vi_end+0xb8/0xc4
<4>[ 1043.501225] [<8000cf70>] r4k_wait_irqoff+0x1c/0x24
<4>[ 1043.510781] [<800662cc>] do_idle+0xe4/0x168
<4>[ 1043.519104] [<80066548>] cpu_startup_entry+0x24/0x2c
<4>[ 1043.528985] [<80577bf0>] start_kernel+0x484/0x4a4
<4>[ 1043.538341] Code: 02021021  00a21021  8c5440e0 <92820007> 7c023420  04c00029  3c060003  3c0281f2  aea00000 
<4>[ 1043.557765] 
<4>[ 1043.561140] ---[ end trace d8e3b967fc39f293 ]---

====================
```

</details>

Nossiac, 2018-05-25:
automation server restarted. the latest snapshot uses kernel 4.14.43 already.

as for out-of-memory issue, it seems your system had just started for ten minutes. if there were memleaks somewhere, it must be accumulated rapidly (normally the wifi driver won't take a lot mem at startup, it will take more and more during high throughput usage). you may do some periodical regular check first. for example:
```
cat /proc/meminfo
cat /proc/slabinfo
```

another debugging method was to remove the specious module/component at a time and try it again, which distinguishes the bad citizen from the good  ones.

Pentiumluyu, 2018-05-27:
Thanks a lot. I had enabled CONFIG_SLUB_DEBUG in the kernel config and rebuilt the firmware. Unfortunately, after flashing the new built firmware, this memory issue happened again. It's weird that this issue now only happened after long idle. While I constantly checking /proc/meminfo and /proc/slabinfo, nothing happened. Here is the most recent crashlog.
<details>
<summary>crashlog</summary>

```
Time: 1527365333.738444
Modules:	pppoe@86b40000+22b0	ppp_async@86b18000+1be0	pppox@86b33000+58a	ppp_generic@86b58000+5aa2	iptable_nat@86aa8000+310	ipt_REJECT@86ee0000+410	ipt_MASQUERADE@86a48000+2f0	xt_time@86b35000+730	xt_tcpudp@86b32000+750	xt_tcpmss@86b21000+470	xt_string@86b1f000+310	xt_statistic@86b1c000+390	xt_state@86b1a000+310	xt_quota@86b0b000+350	xt_pkttype@86b08000+290	xt_owner@86af8000+3b0	xt_nat@86af5000+650	xt_multiport@86af2000+550	xt_mark@86ae4000+2f0	xt_mac@86ae1000+2b0	xt_limit@86ad1000+4f0	xt_length@86ace000+310	xt_hl@86ac1000+370	xt_ecn@86aaf000+590	xt_dscp@86aad000+430	xt_conntrack@86aaa000+950	xt_comment@86a95000+230	xt_addrtype@86a96000+970	xt_TPROXY@86a9a000+1210	xt_TCPMSS@86a8b000+b30	xt_REDIRECT@86a88000+310	xt_LOG@86a7b000+350	xt_HL@86a76000+590	xt_FLOWOFFLOAD@86a77000+a20	xt_DSCP@86a63000+630	xt_CLASSIFY@86a61000+290	ts_kmp@86c8f000+570	ts_fsm@86c8d000+ab0	ts_bm@86c8a000+5d0	slhc@86c88000+149b	nf_reject_ipv4@86a54000+883	nf_nat_redirect@873d5000+5fb	nf_nat_masquerade_ipv4@86a53000+67c	nf_conntrack_ipv4@86cb4000+1410	nf_nat_ipv4@86cb9000+dd1	nf_nat@873c0000+381a	nf_log_ipv4@86cc0000+dd0	nf_log_common@86edc000+acf	nf_flow_table_hw@868a4000+920	nf_flow_table@86a38000+34bc	nf_defrag_ipv6@86ca8000+342e	nf_defrag_ipv4@86eec000+436	nf_conntrack_rtcache@873de000+af0	nf_conntrack_netlink@86a30000+5750	nf_conntrack@86a00000+10ce1	mt7615@85c00000+3c3b34	iptable_mangle@8709f000+3d0	iptable_filter@86fc2000+2d0	ipt_ECN@8706f000+5d0	ip_tables@8709c000+290d	crc_ccitt@86fd0000+42b	xt_set@8708c000+1b70	x_tables@87084000+374f	ip_set_list_set@86fba000+1c50	ip_set_hash_netiface@86fb0000+6290	ip_set_hash_netport@86fa8000+6250	ip_set_hash_netnet@86f98000+6790	ip_set_hash_net@86f88000+5c70	ip_set_hash_netportnet@868f8000+6d30	ip_set_hash_mac@868f0000+2930	ip_set_hash_ipportnet@868e8000+6830	ip_set_hash_ipportip@868d8000+54f0	ip_set_hash_ipport@868c8000+5170	ip_set_hash_ipmark@868c0000+4e90	ip_set_hash_ip@87070000+4db0	ip_set_bitmap_port@87062000+1830	ip_set_bitmap_ipmac@87054000+1bd0	ip_set_bitmap_ip@87034000+1b30	ip_set@87058000+590a	nfnetlink@86fdc000+1277	leds_gpio@874cc000+bb0	gpio_button_hotplug@877d0000+1930
<4>[ 1681.402524]         00000000 00200000 ffffffff 0000000a 00000002 80550000 00000004 805c0004
<4>[ 1681.402553]         ...
<4>[ 1681.402562] Call Trace:
<4>[ 1681.402598] [<80010498>] show_stack+0x58/0x100
<4>[ 1681.402632] [<8042f07c>] dump_stack+0x9c/0xe0
<4>[ 1681.402646] [<800c4644>] dump_header.isra.6+0xa4/0x200
<4>[ 1681.402655] [<800c37cc>] oom_kill_process+0xf0/0x544
<4>[ 1681.402663] [<800c44c8>] out_of_memory+0x3cc/0x440
<4>[ 1681.402682] [<800c8b6c>] __alloc_pages_nodemask+0x944/0xc4c
<4>[ 1681.402718] [<800c0d64>] filemap_fault+0x458/0x658
<4>[ 1681.402731] [<800e8d80>] __do_fault+0x24/0xa0
<4>[ 1681.402745] [<800ed734>] handle_mm_fault+0x694/0xb08
<4>[ 1681.402773] [<8001c3b0>] __do_page_fault+0x120/0x478
<4>[ 1681.402785] [<80021fb8>] tlb_do_page_fault_0+0x118/0x120
<4>[ 1681.402796] Mem-Info:
<4>[ 1681.402819] active_anon:69 inactive_anon:11 isolated_anon:0
<4>[ 1681.402819]  active_file:25 inactive_file:59 isolated_file:0
<4>[ 1681.402819]  unevictable:0 dirty:0 writeback:0 unstable:0
<4>[ 1681.402819]  slab_reclaimable:168 slab_unreclaimable:3076
<4>[ 1681.402819]  mapped:6 shmem:20 pagetables:15 bounce:0
<4>[ 1681.402819]  free:4066 free_pcp:12 free_cma:0
<4>[ 1681.402836] Node 0 active_anon:276kB inactive_anon:44kB active_file:100kB inactive_file:236kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:24kB dirty:0kB writeback:0kB shmem:80kB writeback_tmp:0kB unstable:0kB all_unreclaimable? yes
<4>[ 1681.402859] Normal free:16264kB min:16384kB low:20480kB high:24576kB active_anon:276kB inactive_anon:44kB active_file:136kB inactive_file:468kB unevictable:0kB writepending:0kB present:131072kB managed:123720kB mlocked:0kB kernel_stack:496kB pagetables:60kB bounce:0kB free_pcp:48kB local_pcp:0kB free_cma:0kB
<4>[ 1681.402862] lowmem_reserve[]: 0 0 0
<4>[ 1681.402876] Normal: 189*4kB (UME) 166*8kB (UME) 92*16kB (UME) 40*32kB (UME) 29*64kB (ME) 19*128kB (ME) 13*256kB (ME) 6*512kB (ME) 1*1024kB (E) 0*2048kB 0*4096kB = 16548kB
<4>[ 1681.402947] 123 total pagecache pages
<4>[ 1681.402952] 0 pages in swap cache
<4>[ 1681.402957] Swap cache stats: add 0, delete 0, find 0/0
<4>[ 1681.402961] Free swap  = 0kB
<4>[ 1681.402965] Total swap = 0kB
<4>[ 1681.402970] 32768 pages RAM
<4>[ 1681.402974] 0 pages HighMem/MovableOnly
<4>[ 1681.402978] 1838 pages reserved
<6>[ 1681.402982] [ pid ]   uid  tgid total_vm      rss nr_ptes nr_pmds swapents oom_score_adj name
<6>[ 1681.402999] [  509]     0   509      226        7       3       0        0             0 askfirst
<6>[ 1681.403009] [ 1217]     0  1217      267        7       3       0        0             0 dropbear
<6>[ 1681.403019] [ 1598]     0  1598      302        0       3       0        0             0 ntpd
<6>[ 1681.403029] [ 1615]     0  1615      228        7       3       0        0             0 chndns
<3>[ 1681.403036] Out of memory: Kill process 509 (askfirst) score 0 or sacrifice child
[... cut ...]
<4>[ 3117.627296] Total swap = 0kB
<4>[ 3117.627301] 32768 pages RAM
<4>[ 3117.627305] 0 pages HighMem/MovableOnly
<4>[ 3117.627308] 1838 pages reserved
<6>[ 3117.627313] [ pid ]   uid  tgid total_vm      rss nr_ptes nr_pmds swapents oom_score_adj name
<0>[ 3117.627327] Kernel panic - not syncing: Out of memory and no killable processes...
<0>[ 3117.627327] 

</details>

After this kernel panic, the router had run for 6+ hours without any problem. I will try to do some iperf testing and report later.

Pentiumluyu, 2018-05-27:
Bandwidth testing with iperf caused no issue! Only some bug reports show up on dmesg. Seems like the memory issue may relate to packet processing inside kernel.

<details>
<summary> Bug reports from dmesg while doing bandwidth testing with iperf </summary>

```
[ 4932.664367] FIX skbuff_head_cache: Object at 0x82b90020 not freed
[ 5136.502359] =============================================================================
[ 5136.502385] BUG skbuff_fclone_cache (Tainted: P    B          ): Redzone overwritten
[ 5136.502389] -----------------------------------------------------------------------------
[ 5136.502389] 
[ 5136.502403] INFO: 0x83918000-0x8391801f. First byte 0xbb instead of 0xcc
[ 5136.502447] INFO: Allocated in __alloc_skb+0x74/0x180 age=0 cpu=2 pid=1632
[ 5136.502481] INFO: Freed in tcp_ack+0x8d8/0x1178 age=0 cpu=2 pid=1632
[ 5136.502489] INFO: Slab 0x81075300 objects=18 used=3 fp=0x83918720 flags=0x8101
[ 5136.502495] INFO: Object 0x83918020 @offset=32 fp=0x839181e0
[ 5136.502495] 
[ 5136.502508] Redzone 83918000: bb bb bb bb bb bb bb bb bb bb bb bb bb bb bb bb  ................
[ 5136.502515] Redzone 83918010: bb bb bb bb bb bb bb bb bb bb bb bb bb bb bb bb  ................
[ 5136.502522] Object 83918020: 00 00 00 00 00 00 00 00 9d c5 28 32 01 00 00 00  ..........(2....
[ 5136.502529] Object 83918030: 00 00 00 00 00 00 00 00 e3 3c 6a e1 2b 59 6a e1  .........<j.+Yj.
[ 5136.502537] Object 83918040: 05 00 a8 05 00 00 00 00 10 00 00 00 00 00 00 00  ................
[ 5136.502545] Object 83918050: 78 30 00 00 c9 06 00 00 46 b9 28 32 01 00 00 00  x0......F.(2....
[ 5136.502552] Object 83918060: 61 be 28 32 01 00 00 00 00 00 00 00 00 00 00 00  a.(2............
[ 5136.502559] Object 83918070: 00 00 00 00 48 1c 00 00 48 1c 00 00 00 00 00 00  ....H...H.......
[ 5136.502566] Object 83918080: 00 00 07 00 c0 00 00 00 00 00 00 00 00 00 00 00  ................
[ 5136.502572] Object 83918090: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 5136.502579] Object 839180a0: 00 00 00 00 40 00 00 00 00 00 00 00 00 00 00 00  ....@...........
[ 5136.502586] Object 839180b0: 00 00 ff ff 00 00 ff ff 80 d9 f5 85 c0 d9 f5 85  ................
[ 5136.502593] Object 839180c0: 80 d8 f5 85 80 d9 f5 85 08 1f 00 00 01 00 00 00  ................
[ 5136.502600] Object 839180d0: 00 00 00 00 d0 80 91 83 00 00 00 00 00 00 00 00  ................
[ 5136.502607] Object 839180e0: 20 80 62 82 80 24 c4 87 92 1d 00 00 00 00 00 00   .b..$..........
[ 5136.502613] Object 839180f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 5136.502620] Object 83918100: 78 30 00 00 00 00 00 00 00 00 00 00 00 00 00 00  x0..............
[ 5136.502627] Object 83918110: 00 00 00 00 00 00 00 00 00 00 00 00 90 af 38 80  ..............8.
[ 5136.502633] Object 83918120: 03 68 c1 84 8a 1c 00 00 48 1c 00 00 00 00 00 01  .h......H.......
[ 5136.502640] Object 83918130: 00 00 09 6b c0 03 00 00 00 00 00 00 e0 00 10 00  ...k............
[ 5136.502647] Object 83918140: 00 00 00 00 00 00 00 00 44 ea e3 9b 81 00 01 10  ........D.......
[ 5136.502653] Object 83918150: 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 5136.502660] Object 83918160: 08 00 e0 00 cc 00 be 00 80 d9 f5 85 c0 d9 f5 85  ................
[ 5136.502666] Object 83918170: 80 d8 f5 85 3e d9 f5 85 08 1f 00 00 01 00 00 00  ....>...........
[ 5136.502673] Object 83918180: 00 00 00 00 6b 6b 6b a5                          ....kkk.
[ 5136.502679] Redzone 83918188: cc cc cc cc                                      ....
[ 5136.502687] Padding 839181b0: 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a  ZZZZZZZZZZZZZZZZ
[ 5136.502698] CPU: 0 PID: 7 Comm: ksoftirqd/0 Tainted: P    B           4.14.43 #0
[ 5136.502705] Stack : 00000000 00000000 00000000 00000000 805c7ad2 00000044 00000000 00000000
[ 5136.502735]         87c51774 805588a7 804e76bc 00000000 00000007 00000001 87c71b00 532616ca
[ 5136.502764]         00000000 00000000 805c0000 0001b4d0 00000000 0000086c 00000001 00000000
[ 5136.502792]         00000000 80560000 0000086b 805f0000 00000000 00000000 80580000 81075300
[ 5136.502820]         83918020 804f47a8 8391801f 804f4b10 00000000 80550000 00000000 805c0000
[ 5136.502848]         ...
[ 5136.502858] Call Trace:
[ 5136.502886] [<80010498>] show_stack+0x58/0x100
[ 5136.502922] [<8042f07c>] dump_stack+0x9c/0xe0
[ 5136.502951] [<80106a58>] check_bytes_and_report+0xc4/0x10c
[ 5136.502961] [<80106b04>] check_object+0x64/0x2e4
[ 5136.502975] [<80109a30>] free_debug_processing+0x1c0/0x3b4
[ 5136.502985] [<80109ca0>] __slab_free+0x7c/0x3fc
[ 5136.502995] [<8010a180>] kmem_cache_free+0x160/0x174
[ 5136.503013] [<802ec480>] fe_txd_unmap+0xd4/0xec
[ 5136.503023] [<802ee8d8>] fe_poll+0x130/0x810
[ 5136.503043] [<80330a44>] net_rx_action+0x150/0x30c
[ 5136.503077] [<8044c248>] __do_softirq+0x128/0x2ec
[ 5136.503097] [<80032638>] run_ksoftirqd+0x38/0x6c
[ 5136.503119] [<8004eb78>] smpboot_thread_fn+0x1a8/0x1d8
[ 5136.503141] [<8004b3b8>] kthread+0x130/0x144
[ 5136.503151] [<8000b098>] ret_from_kernel_thread+0x14/0x1c
[ 5136.503165] FIX skbuff_fclone_cache: Restoring 0x83918000-0x8391801f=0xcc
[ 5136.503165] 
[ 5136.503172] FIX skbuff_fclone_cache: Object at 0x83918020 not freed
[... 62 lines cut ...]
[ 5171.218810]         83468080 804f47a8 8346801f 804f4b10 00000000 80550000 00000000 805c0000
[ 5171.218839]         ...
[ 5171.218848] Call Trace:
[ 5171.218877] [<80010498>] show_stack+0x58/0x100
[ 5171.218915] [<8042f07c>] dump_stack+0x9c/0xe0
[... cut ...]
[ 5171.219032] [<802ee8d8>] fe_poll+0x130/0x810
[ 5171.219052] [<80330a44>] net_rx_action+0x150/0x30c
[ 5171.219088] [<8044c248>] __do_softirq+0x128/0x2ec
[ 5171.219109] [<80032638>] run_ksoftirqd+0x38/0x6c
[ 5171.219131] [<8004eb78>] smpboot_thread_fn+0x1a8/0x1d8
[ 5171.219152] [<8004b3b8>] kthread+0x130/0x144
[ 5171.219163] [<8000b098>] ret_from_kernel_thread+0x14/0x1c
[ 5171.219177] FIX kmalloc-512: Restoring 0x83468000-0x8346801f=0xcc
[ 5171.219177] 
[ 5171.219184] FIX kmalloc-512: Object at 0x83468080 not freed
```

</details>

Pentiumluyu, 2018-05-27:
Another bug reports from dmesg while doing bandwidth testing with iperf.

<details>
<summary>Bug reports from dmesg while doing bandwidth testing with iperf</summary>

```
[  672.570685] =============================================================================
[  672.570713] BUG skbuff_head_cache (Tainted: P    B          ): Redzone overwritten
[  672.570718] -----------------------------------------------------------------------------
[  672.570718] 
[  672.570732] INFO: 0x82ce0000-0x82ce001f. First byte 0xbb instead of 0xcc
[  672.570776] INFO: Allocated in __build_skb+0x4c/0xd0 age=0 cpu=0 pid=7
[  672.571637] INFO: Freed in RxRing1DoneInterruptHandle+0xe0/0x120 [mt7615] age=0 cpu=0 pid=7
[  672.571656] INFO: Slab 0x8105cc00 objects=16 used=16 fp=0x  (null) flags=0x0101
[  672.571663] INFO: Object 0x82ce0020 @offset=32 fp=0x82ce0520
[  672.571663] 
[  672.571676] Redzone 82ce0000: bb bb bb bb bb bb bb bb bb bb bb bb bb bb bb bb  ................
[  672.571682] Redzone 82ce0010: bb bb bb bb bb bb bb bb bb bb bb bb bb bb bb bb  ................
[  672.571691] Object 82ce0020: 00 00 00 00 20 00 ce 82 35 da b1 c6 68 6d 32 15  .... ...5...hm2.
[  672.571698] Object 82ce0030: 00 00 00 00 80 24 c4 87 ea 05 00 00 00 00 00 00  .....$..........
[  672.571706] Object 82ce0040: 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00  ................
[  672.571714] Object 82ce0050: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  672.571721] Object 82ce0060: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  672.571728] Object 82ce0070: 00 00 00 00 ea 05 00 00 00 00 00 00 0e 00 00 00  ................
[  672.571735] Object 82ce0080: 00 00 20 00 03 00 00 02 00 00 00 00 00 00 00 00  .. .............
[  672.571742] Object 82ce0090: 00 00 00 00 03 00 00 00 00 00 00 00 81 00 01 10  ................
[  672.571749] Object 82ce00a0: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  672.571755] Object 82ce00b0: 08 00 c0 00 c0 00 b2 00 5c 92 46 83 c0 92 46 83  ........\.F...F.
[  672.571762] Object 82ce00c0: c0 8b 46 83 72 8c 46 83 80 08 00 00 01 00 00 00  ..F.r.F.........
[  672.571769] Redzone 82ce00d0: cc cc cc cc                                      ....
[  672.571776] Padding 82ce00f8: 5a 5a 5a 5a 5a 5a 5a 5a                          ZZZZZZZZ
[  672.571788] CPU: 0 PID: 7 Comm: ksoftirqd/0 Tainted: P    B           4.14.43 #0
[  672.571795] Stack : 00000000 00000000 00000000 00000000 805c7ad2 00000044 00000000 00000000
[  672.571829]         87c51774 805588a7 804e76bc 00000000 00000007 00000001 87c71b00 532616ca
[  672.571858]         00000000 00000000 805c0000 00011410 00000000 00000568 00000001 00000000
[  672.571886]         00000000 80560000 00000567 805f0000 00000000 00000000 80580000 8105cc00
[  672.571915]         82ce0020 804f47a8 82ce001f 804f4b10 00000000 80550000 00000000 805c0000
[  672.571944]         ...
[  672.571954] Call Trace:
[  672.571987] [<80010498>] show_stack+0x58/0x100
[  672.572024] [<8042f07c>] dump_stack+0x9c/0xe0
[  672.572055] [<80106a58>] check_bytes_and_report+0xc4/0x10c
[  672.572067] [<80106b04>] check_object+0x64/0x2e4
[  672.572080] [<80109a30>] free_debug_processing+0x1c0/0x3b4
[  672.572091] [<80109ca0>] __slab_free+0x7c/0x3fc
[  672.572101] [<8010a180>] kmem_cache_free+0x160/0x174
[  672.572120] [<802ec480>] fe_txd_unmap+0xd4/0xec
[  672.572130] [<802ee8d8>] fe_poll+0x130/0x810
[  672.572150] [<80330a44>] net_rx_action+0x150/0x30c
[  672.572185] [<8044c248>] __do_softirq+0x128/0x2ec
[  672.572205] [<80032638>] run_ksoftirqd+0x38/0x6c
[  672.572227] [<8004eb78>] smpboot_thread_fn+0x1a8/0x1d8
[  672.572249] [<8004b3b8>] kthread+0x130/0x144
[  672.572259] [<8000b098>] ret_from_kernel_thread+0x14/0x1c
[  672.572273] FIX skbuff_head_cache: Restoring 0x82ce0000-0x82ce001f=0xcc
[  672.572273] 
[  672.572404] FIX skbuff_head_cache: Object at 0x82ce0020 not freed
[  769.328938] =============================================================================
[  769.328964] BUG skbuff_head_cache (Tainted: P    B          ): Redzone overwritten
[  769.328970] -----------------------------------------------------------------------------
[  769.328970] 
[  769.328984] INFO: 0x822f0000-0x822f001f. First byte 0xbb instead of 0xcc
[  769.329030] INFO: Allocated in __build_skb+0x4c/0xd0 age=1 cpu=0 pid=7
[  769.329048] INFO: Freed in fe_txd_unmap+0xd4/0xec age=2 cpu=0 pid=7
[  769.329058] INFO: Slab 0x81048e00 objects=16 used=16 fp=0x  (null) flags=0x0101
[  769.329064] INFO: Object 0x822f0020 @offset=32 fp=0x822f0220
[  769.329064] 
[  769.329077] Redzone 822f0000: bb bb bb bb bb bb bb bb bb bb bb bb bb bb bb bb  ................
[  769.329084] Redzone 822f0010: bb bb bb bb bb bb bb bb bb bb bb bb bb bb bb bb  ................
[  769.329092] Object 822f0020: 00 00 00 00 20 00 2f 82 3d a3 77 4d 7f 6d 32 15  .... ./.=.wM.m2.
[... 363 lines cut ...]
[  873.683882]         00000000 80560000 000006ec 805f0000 00000000 00000000 80580000 8109b800
[  873.683911]         84c40020 804f47a8 84c4001f 804f4b10 00000000 80550000 00000000 805c0000
[  873.683939]         ...
[  873.683949] Call Trace:
[  873.683979] [<80010498>] show_stack+0x58/0x100
[  873.684016] [<8042f07c>] dump_stack+0x9c/0xe0
[  873.684045] [<80106a58>] check_bytes_and_report+0xc4/0x10c
[  873.684055] [<80106b04>] check_object+0x64/0x2e4
[  873.684069] [<80109a30>] free_debug_processing+0x1c0/0x3b4
[  873.684079] [<80109ca0>] __slab_free+0x7c/0x3fc
[... cut ...]
[  873.684128] [<80330a44>] net_rx_action+0x150/0x30c
[  873.684163] [<8044c248>] __do_softirq+0x128/0x2ec
[  873.684184] [<80032638>] run_ksoftirqd+0x38/0x6c
[  873.684206] [<8004eb78>] smpboot_thread_fn+0x1a8/0x1d8
[  873.684228] [<8004b3b8>] kthread+0x130/0x144
[  873.684239] [<8000b098>] ret_from_kernel_thread+0x14/0x1c
[  873.684253] FIX skbuff_head_cache: Restoring 0x84c40000-0x84c4001f=0xcc
[  873.684253] 
[  873.684277] FIX skbuff_head_cache: Object at 0x84c40020 not freed
```

</details>

Pentiumluyu, 2018-05-27:
It seems that the memory issue is related to `BUG skbuff_head_cache (Tainted: P               ): Redzone overwritten`. As this bug happened repeatedly even when load was low, and `MemFree` of `/proc/meminfo` went down. But `OBJS` and `OBJ SIZE` of items in `/proc/slabinfo` even went down while `MemFree` of `/proc/meminfo` continued to go down. After unloading the module mt7615, `MemFree` remained pretty much the same(about 80M, and my router got 128M RAM totally), and this bug would not happen even under high load on the network. But after loading the module mt7615, `MemFree` went down to about 60M and continued to go down, and this bug happened repeatedly. Does the driver have anything to do with the memory issue? How could I pinpoint the problem?

Pentiumluyu, 2018-05-27:
Most recently, loading the module mt7615 caused kernel panic(only happened once after using kernel 4.14.43).

<details>
<summary>crashlog</summary>

```
Time: 1527436369.810451
Modules:	mt7615@81400000+3c3b34	pppoe@86b98000+22b0	ppp_async@86b14000+1be0	pppox@86b40000+58a	ppp_generic@86b90000+5aa2	nf_conntrack_ipv6@86b76000+15a0	iptable_nat@86b82000+310	ipt_REJECT@86b71000+410	ipt_MASQUERADE@86b6f000+2f0	xt_time@86b64000+730	xt_tcpudp@86b60000+750	xt_tcpmss@86b5e000+470	xt_string@86b41000+310	xt_statistic@86b3f000+390	xt_state@86b34000+310	xt_quota@86b31000+350	xt_pkttype@86b2f000+290	xt_owner@86b1e000+3b0	xt_nat@86b1a000+650	xt_multiport@86b10000+550	xt_mark@86b0d000+2f0	xt_mac@86b0b000+2b0	xt_limit@86af8000+4f0	xt_length@86af5000+310	xt_hl@86ae5000+370	xt_ecn@86adb000+590	xt_dscp@86ad6000+430	xt_conntrack@86ad7000+950	xt_comment@86abf000+230	xt_addrtype@86abd000+970	xt_TPROXY@86ac6000+1210	xt_TCPMSS@86aaf000+b30	xt_REDIRECT@86aab000+310	xt_LOG@86d1e000+350	xt_HL@86d1b000+590	xt_FLOWOFFLOAD@86d19000+a20	xt_DSCP@86d0a000+630	xt_CLASSIFY@86d07000+290	ts_kmp@86a02000+570	ts_fsm@86a03000+ab0	ts_bm@86a3f000+5d0	slhc@86a06000+149b	nf_reject_ipv4@86d3b000+883	nf_nat_redirect@86a91000+5fb	nf_nat_masquerade_ipv4@86d42000+67c	nf_conntrack_ipv4@86a8c000+1410	nf_nat_ipv4@86a27000+dd1	nf_nat@86d28000+381a	nf_log_ipv4@86a51000+dd0	nf_flow_table_hw@86a0e000+920	nf_flow_table@86a54000+34bc	nf_defrag_ipv6@87360000+342e	nf_defrag_ipv4@86d40000+436	nf_conntrack_rtcache@86a14000+af0	nf_conntrack_netlink@86a38000+5750	nf_conntrack@86a40000+10ce1	iptable_mangle@868d4000+3d0	iptable_filter@86f9d000+2d0	ipt_ECN@86f82000+5d0	ip_tables@870e8000+290d	crc_ccitt@868d3000+42b	xt_set@870d8000+1b70	ip_set_list_set@870d2000+1c50	ip_set_hash_netiface@870c0000+6290	ip_set_hash_netport@870b8000+6250	ip_set_hash_netnet@870a8000+6790	ip_set_hash_net@870a0000+5c70	ip_set_hash_netportnet@87090000+6d30	ip_set_hash_mac@87088000+2930	ip_set_hash_ipportnet@86fb8000+6830	ip_set_hash_ipportip@86fb0000+54f0	ip_set_hash_ipport@86fa0000+5170	ip_set_hash_ipmark@86f98000+4e90	ip_set_hash_ip@86f88000+4db0	ip_set_bitmap_port@868f4000+1830	ip_set_bitmap_ipmac@868ec000+1bd0	ip_set_bitmap_ip@868e6000+1b30	ip_set@868d8000+590a	nfnetlink@868ca000+1277	ip6t_REJECT@86fdc000+450	nf_reject_ipv6@86fe2000+a08	nf_log_ipv6@87066000+1070	nf_log_common@87053000+acf	ip6table_mangle@8702e000+4d0	ip6table_filter@87068000+2d0	ip6_tables@86fc8000+2801	x_tables@86fc4000+374f	leds_gpio@87de0000+bb0	gpio_button_hotplug@87ec6000+1930
<4>[ 9372.419837] CalCacheApply = 0 
<4>[ 9372.442826] APEdca0
<4>[ 9372.447674] Valid=1
<4>[ 9372.451920] APAifsn[0]=3
<4>[ 9372.457009] APAifsn[1]=7
<4>[ 9372.462036] APAifsn[2]=1
<4>[ 9372.467100] APAifsn[3]=1
<4>[ 9372.472154] APEdca1
<4>[ 9372.476732] Valid=1
<4>[ 9372.480931] APAifsn[0]=3
<4>[ 9372.485971] APAifsn[1]=7
<4>[ 9372.491051] APAifsn[2]=1
<4>[ 9372.496133] APAifsn[3]=1
<4>[ 9372.501197] APEdca2
<4>[ 9372.506039] APEdca3
<4>[ 9372.515584] BSSAifsn[0]=3
<4>[ 9372.520850] BSSAifsn[1]=7
<4>[ 9372.526068] BSSAifsn[2]=2
<4>[ 9372.531316] BSSAifsn[3]=2
<4>[ 9372.537190] BSSAifsn[0]=3
<4>[ 9372.542410] BSSAifsn[1]=7
<4>[ 9372.547667] BSSAifsn[2]=2
<4>[ 9372.552880] BSSAifsn[3]=2
<4>[ 9372.558684] APSDCapable[0]=0
<4>[ 9372.564429] APSDCapable[1]=0
<4>[ 9372.570281] default ApCliAPSDCapable[0]=0
<4>[ 9372.578358] default ApCliAPSDCapable[1]=0
<4>[ 9372.589469] DfsZeroWait Support=0/0 
<4>[ 9372.597413] DfsZeroWaitCacTime=255/255 
<4>[ 9372.644717] [PMF]Set_PMFMFPC_Proc:: apidx=0, Desired MFPC=0
<4>[ 9372.655905] [PMF]Set_PMFMFPC_Proc:: apidx=1, Desired MFPC=0
<4>[ 9372.667522] [PMF]Set_PMFMFPR_Proc:: apidx=0, Desired MFPR=0
<4>[ 9372.678650] [PMF]Set_PMFMFPR_Proc:: apidx=1, Desired MFPR=0
<4>[ 9372.690420] [PMF]Set_PMFSHA256_Proc:: apidx=0, Desired PMFSHA256=0
<4>[ 9372.702792] [PMF]Set_PMFSHA256_Proc:: apidx=1, Desired PMFSHA256=0
<4>[ 9372.722764] cfg_mode=14
<4>[ 9372.727726] cfg_mode=9
<4>[ 9372.738219] AndesSendCmdMsg: Could not send in band command due to diablefRTMP_ADAPTER_MCU_SEND_IN_BAND_CMD
<4>[ 9372.763575] HT: Ext Channel = ABOVE
<4>[ 9372.770544] HT: Ext Channel = BELOW
<4>[ 9372.779397] HT: greenap_cap = 0
<4>[ 9372.826508] IcapMode = 0
<4>[ 9372.840655] WtcSetMaxStaNum: MaxStaNum:90, BssidNum:2, WdsNum:0, ApcliNum:2, MaxNumChipRept:32, MinMcastWcid:124
<4>[ 9372.861211] Top Init Done!
<4>[ 9372.866650] Use dev_alloc_skb
<4>[ 9372.873702] RX[0] DESC a56d2000 size = 8192
<4>[ 9372.882803] RX[1] DESC a56d4000 size = 8192
<4>[ 9372.894541] Hif Init Done!
<4>[ 9372.899976] ctl->txq = c17f0b28
<4>[ 9372.906249] ctl->rxq = c17f0b34
<4>[ 9372.912483] ctl->ackq = c17f0b40
<4>[ 9372.918911] ctl->kickq = c17f0b4c
<4>[ 9372.925490] ctl->tx_doneq = c17f0b58
<4>[ 9372.932613] ctl->rx_doneq = c17f0b64
<4>[ 9372.940802] mt7615_fw_prepare():FW(8a10), HW(8a10), CHIPID(7615))
<4>[ 9372.952932] mt7615_fw_prepare(2752): MT7615_E3, USE E3 patch and ram code binary image
<4>[ 9372.968698] AndesMTLoadRomMethodFwDlRing(1036), cap->rom_patch_len(11150)
<4>[ 9372.982219] AndesRestartCheck: Current TOP_MISC2(0x1)
<4>[ 9372.992274] AndesRestartCheck: (TOP_MISC2 = 1), ready to continue...RET(0)
<4>[ 9373.005961] 2
<4>[ 9373.005967] 0
<4>[ 9373.009107] 1
[... 191 lines cut ...]
<4>[ 9381.081707] Cause : c0800008 (ExcCode 02)
<4>[ 9381.089672] BadVA : 00000005
<4>[ 9381.095388] PrId  : 0001992f (MIPS 1004Kc)
[... cut ...]
<4>[ 9381.414786]         ...
<4>[ 9381.419648] Call Trace:
<4>[ 9381.424990] [<814bebac>] wlan_operate_get_ht_bw+0x4/0xc [mt7615]
<4>[ 9381.437462] [<814e9cf8>] N_ChannelCheck+0xa0/0x280 [mt7615]
<4>[ 9381.449062] [<8140018c>] 0x8140018c
<4>[ 9381.456249] Code: 03e00008  ac650008  8c82090c <03e00008> 90420005  8c82090c  03e00008  90420004  8c82090c 
<4>[ 9381.475654] 
<4>[ 9381.479019] ---[ end trace 46e0f1cd7958aeaf ]---

===================================
Time: 1527436369.829985
```

</details>

Pentiumluyu, 2018-05-31:
Self-compiled module without MULTI_INF_SUPPORT,RT_SECOND_CARD,RT_THIRD_CARD works fine under kernel version 4.14.44. Maybe the memory issue is related to MULTI_INF_SUPPORT?

closed 2018-05-31 by Pentiumluyu