Nossiac/mtk-openwrt-feeds #69: mt7603 begins to refuse connections after a while

venustrg, 2018-05-22:
i use mt7621ap/mt7603/mt7612/64Mb router.

sometimes i get a lot of such errors:

`[16279.898390] /home/shello/workspace/lede-1701/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7621/tm7603/src/rlt_wifi_ap/../rlt_wifi/ap/ap_data.c:1785 assert (pTxBlk->TxPacketList.Number > 1)failed
`
after that android clients are unable to connect to 2.4GHz mt7603 wifi.
android reports "connecting -> saved"
instead of "connecting -> obtaining ip address -> connected"

there are no dmesg messages about client connections, just a lot of error messages:

```
[27265.680538] [PMF]PMF_RsnCapableValidation: PMF Connection
[27265.691583] MtAsicACQueue: Write CR:21510, Value=10421
[27265.701821] MtAsicACQueue: Write CR:21500, Value=10421
[27267.677097] [PMF]PMF_PerformRxFrameAction: ERROR FRAME
[27287.526421] [PMF]PMF_RsnCapableValidation: PMF Connection
[27287.537538] MtAsicACQueue: Write CR:21510, Value=10421
[27287.547815] MtAsicACQueue: Write CR:21500, Value=10421
[27289.523834] [PMF]PMF_PerformRxFrameAction: ERROR FRAME

```

at the same time mt7612 works.
it is enough to connect to luci, and make "save/apply" to correct situation.
mt7603 can work 20hrs under constantly load before this trouble
and can drop after a hour. 
i can't find any regularity.
tried to change almost all options, nothing helped.

venustrg, 2018-06-04:
right now we have following modules:
mt76x3 v4.0.1.0rev2.P3
mt76x2 v3.0.4.0.P3.20161201

i've compiled
mt76x3 v4.1.0.0.P48
mt76x2 v3.0.5.0.P48
mt7610 v3.0.0.9.P48
and at first glance, they work well, at least with 3.4 kernel.

do you need sources?

BrowserZz, 2023-11-15:
> right now we have following modules: mt76x3 v4.0.1.0rev2.P3 mt76x2 v3.0.4.0.P3.20161201
> 
> i've compiled mt76x3 v4.1.0.0.P48 mt76x2 v3.0.5.0.P48 mt7610 v3.0.0.9.P48 and at first glance, they work well, at least with 3.4 and 2.6 kernels. didn't tried to build for 4.x
> 
> do you need sources?

> right now we have following modules: mt76x3 v4.0.1.0rev2.P3 mt76x2 v3.0.4.0.P3.20161201
> 
> i've compiled mt76x3 v4.1.0.0.P48 mt76x2 v3.0.5.0.P48 mt7610 v3.0.0.9.P48 and at first glance, they work well, at least with 3.4 and 2.6 kernels. didn't tried to build for 4.x
> 
> do you need sources?

> right now we have following modules: mt76x3 v4.0.1.0rev2.P3 mt76x2 v3.0.4.0.P3.20161201
> 
> i've compiled mt76x3 v4.1.0.0.P48 mt76x2 v3.0.5.0.P48 mt7610 v3.0.0.9.P48 and at first glance, they work well, at least with 3.4 and 2.6 kernels. didn't tried to build for 4.x
> 
> do you need sources?

May I ask if you solved it? This same error occurs, this is when using WPA2/WPA3 authentication type, and it is not possible to connect to WIFI using WPS
Linux LEDE 4.4.198  mtk7603
[image]