Nossiac/mtk-openwrt-feeds #74: Collected errors: (Compiling)

gioreva, 2018-06-07:
Hi Nossiac
Thank you for this.

I have compiled, without your feeds, and everything works fine. Also image on device.
When I add your feeds I have these errors.

**This before feeds:**

make menuconfig
	Target  "Mediatek Ralink MIPS"
	Subtarghet  "RT5350"
	Targhet profile "A5-V11"
	Base system:
		Block-mount  - block device mounting and checking
		Blockd - Block device automounting
		Bridge - ethernet bridging configuration utility
	Kernel module ---->
		Wireless Driver ---->
			kmod-mt7601u
	Network ---->
		Web Server ---->
			apache
			Iperf
			Iperf3
			Iperf3-ssl
			Ppp-mod-pppol2tp
	Luci ---->
		Collection ---->
			Luci
		Application ----> 
			Luci-app-samb
		Themes ---->
			Material
	Kernel module --> 
		Usb Support --> 
			kmod-usb2
			kmod-usb-uhci
			kmod-usb-ohci
			kmod-usb-storage
			kmod-usb-storage-extras
			kmod-usb-storage-uas
			kmod-usb-wdm
	Utility -->
		 usbutils
		 usb-modeswitch

**same** 
```
scripts/feeds update -f mtk
scripts/feeds install -a -p mtk
```
**or**
```
./scripts/feeds update -a
./scripts/feeds install -a
```

**Collected errors:**
```
 * check_data_file_clashes: Package bridge wants to install file /home/gevagiorgio/openwrt_sources/build_dir/target-mipsel_24kc_musl/root-ramips/usr/sbin/brctl
        But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package bridge.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: * [package/install] Error 255
make[2]: Leaving directory '/home/gevagiorgio/openwrt_sources'
package/Makefile:104: recipe for target '/home/gevagiorgio/openwrt_sources/staging_dir/target-mipsel_24kc_musl/stamp/.package_install' failed
make[1]: * [/home/gevagiorgio/openwrt_sources/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/gevagiorgio/openwrt_sources'
/home/gevagiorgio/openwrt_sources/include/toplevel.mk:216: recipe for target 'world' failed
make: * [world] Error 2
```

LGA1150, 2018-06-10:
> Bridge - ethernet bridging configuration utility

Do not select `Bridge`, as it's already provided by busybox

gioreva, 2018-06-12:
Ok, doing this the compilation is successful but I have the same problems as before.
Problems that I did not have without your feeds.

lsusb
unable to initialize libusb: -99

iwlist
Usage: iwlist [interface] scanning [essid NNN] [last]
              [interface] frequency
              [interface] channel
              [interface] bitrate
              [interface] rate
              [interface] encryption
              [interface] keys
              [interface] power
              [interface] txpower

iwinfo
(nothing)

The old 2.5 Ghz WiFi network has disappeared on luci.

On Interface LAN  Physical Settings -  No wireless interface are present
On MTK - WiFi - Enable --->   Please restart manually to create this interface.

gioreva, 2018-10-01:
I have sorces thath work well.
Your code is installed but not enabled.
Led work well.
If enabled, the MT7610 driver only, the LEDs no longer work.
The folder that was present before is no longer present.
/sys/devices/platform/gpio-leds/leds/LedBlue/brightness
Only exist:
/sys/devices/platform/gpio-leds