Nossiac/mtk-openwrt-feeds #77: failed to mmap mt7620

Killbrum, 2018-06-21:
Good time of the day!
Want to improve wi-fi performance of my nexx 3020 mini-router. I took latest sources from git (lede). Took your changes and made a basic package. But I have issues with modprobe. So..

`modprobe mt7620.ko`

shows me the next message
`failed to mmap /lib/modules/4.14.50/mt7620.ko`

And the size of mt7620.ko is zero. Whats wrong in my configuration? How can I solve this?

I'm using kernel 4.14.50. 

thanks!

WuSiYu, 2018-07-10:
The kernel modules are not contain in the source code, it will be download from http://nossiac.com/download/mtk-wifi-ko/ during the compile. And there is no kernel module for the 4.14.50 kernel on that website.So wget download failed and produced a zero size file.

The good new is, the module for kernel 4.14.44 on that site may works properly on your kernel, it works fine on my kernel 4.14.53, so you can modify the `/drivers/mt7603/Makefile` and force it to download the 4.14.44 module.

anonimou0, 2018-12-05:
@WuSiYu 

How can I compile the module for my kernel and OpenWrt version? With the source code, obviously.

I need Linux 4.14.82 for MT7615 on a MT7623 device.

WuSiYu, 2018-12-12:
@anonimou0 I think it is not an open-source driver, so we can't compile the module itself(the .ko file) by ourselves.

anonimou0, 2018-12-12:
@WuSiYu , the manufacturer provided me the code for both MT7615 and MT7603 but I'm having a hard time trying to compile it.