Nossiac/mtk-openwrt-feeds #80: How to get a list of the connected wifi clients?

WuSiYu, 2018-07-10:
Thanks for your works, it worls fine on my WR1200JS.
And now I have some question about the driver's detail
Is it possible to get a list of the connected wifi clients under this driver ? And how?

thanks!

adron-s, 2018-07-27:
iwpriv apcli0 show stainfo

WuSiYu, 2018-08-02:
@adron-s it doesn't output anything,and i mean i need a list of the other devices(like a phone or a laptop) which connected to the router's AP

adron-s, 2018-08-06:
root@LEDE:/# iwpriv ra0 show stainfo
[25773.607989] RTMPAPPrivIoctlShow(): Before check, this_char=stainfo
[25773.614278] RTMPAPPrivIoctlShow(): after check, this_char=stainfo, value=
[25773.621241] Show_MacTable_Proc(): arg=
[25773.625080] Dump MacTable entries info, EntType=0x20001
[25773.630399] 
[25773.631911] HT Operating Mode : 0
[25773.635267] 
[25773.636800] 
[25773.636800] MAC                MODE AID  BSS PSM WMM MIMOPS RSSI0/1/2 PhMd  BW  MCS   SGI   STBC  Idle   Rate  
[25773.648636] A4:DB:30:4B:67:62  200011    0   0   1   3      -40/0  /0  HTMIX 20M 7     0   0    285    65     %
[25773.661057]                                                  MaxCap:HTMIX     20M   7     1   0    
[25773.666557] 

The result of the command is visible in dmesg