Nossiac/mtk-openwrt-feeds #82: Download eeprom for mt7688

Microdrive, 2018-07-30:
Please help me!

I'm using the mt7628-for-mt7628-linux-4.4.61.ko driver. Chip mt7688.
My driver uses the default settings for eeprom and does not load the MAC address and other parameters.

When the command is executed

`ifconfig ra0 up`

Outputs  (driver in debug mode):

```
[ 3486.514098] efuse_probe: efuse = 10000012
[ 3486.522034] RtmpChipOpsEepromHook::e2p_type=2, inf_Type=4
[ 3486.532724] RtmpEepromGetDefault::e2p_dafault=2
[ 3486.541693] RtmpChipOpsEepromHook: E2P type(2), E2pAccessMode = 2, E2P default = 2
[ 3486.556679] NVM is FLASH mode
[ 3486.562554] Allocate 8192 memory for BA reordering
[ 3486.572094] 1. Phy Mode = 14
[ 3486.577786] NICReadEEPROMParameters()-->
[ 3486.585550] --> rtmp_nv_init
[ 3486.591384] net ra0: Direct firmware load for mt7620.eeprom.bin failed with error -2
[ 3486.606738] net ra0: Falling back to user helper
[ 3486.651167] firmware mt7620.eeprom.bin: firmware_loading_store: map pages failed
[ 3486.667229] os_load_code_from_bin:fw not available(/lib/firmware/mt7620.eeprom.bin)
[ 3486.849665] AndesMTRxProcessEvent (seq=4)
[ 3486.857639] Initialize MAC Address from E2PROM
[ 3486.866628] E2PROM MAC: =00:00:00:00:00:00
[ 3486.874741] Use the MAC address what is assigned from EEPROM.

```
I add the file /lib/firmware/mt7620.eeprom.bin

`dd if=/dev/mtd2 of=/lib/firmware/mt7620.eeprom.bin bs=512 count=1`
`ifconfig ra0 up`

```
 RtmpChipOpsEepromHook: E2P type(2), E2pAccessMode = 2, E2P default = 2
[ 3909.396916] NVM is FLASH mode
[ 3909.402791] Allocate 8192 memory for BA reordering
[ 3909.412334] 1. Phy Mode = 14
[ 3909.418026] NICReadEEPROMParameters()-->
[ 3909.425788] --> rtmp_nv_init
[ 3909.431979] RtmpOSFileOpen(): Error 2 opening /etc_ro/wlan/MT7628_EEPROM_20140317.bin
[ 3909.447537] --> Error opening file /etc_ro/wlan/MT7628_EEPROM_20140317.bin
[ 3909.461162] Load EEPROM Buffer from default BIN.
[ 3909.470388] Cal Free IC!!
[ 3909.475565] mt7628_cal_free_data_get
[ 3909.482709] The EEPROM in Flash is wrong, use default
[ 3909.492716] proprietary driver not support flash write, will write on ated.
[ 3909.673761] AndesMTRxProcessEvent (seq=4)
[ 3909.681755] Initialize MAC Address from E2PROM
[ 3909.690740] E2PROM MAC: =00:00:00:00:00:f0
```
It looks like the file mt7620.eeprom.bin is ignored! The driver loads the eeprom from the "factory" section.

The analysis of the source code showed:
1) The driver loads the eeprom into FLASH mode ([3909.396916] NVM is FLASH mode)
2) The validFlashEepromID (pAd) function reads the first 2 bytes of the "factory" section and if they equal 0x7628 returns "true".But my validFlashEepromID returns false !!! Why??? My "factory" section looks like this:
```
00000000  28 76 00 02 1c 88 79 54  61 c4 00 00 00 00 00 00  |(v....yTa.......|
00000010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000020  00 00 00 00 20 00 00 00  1c 88 79 54 61 c4 1c 88  |.... .....yTa...|
00000030  79 54 61 c5 22 34 00 20  ff ff 00 01 00 00 00 00  |yTa."4. ........|
00000040  00 00 22 00 00 00 00 00  30 00 00 00 00 00 00 00  |..".....0.......|
00000050  82 00 00 94 40 ba c0 ca  20 80 80 c3 40 ca 20 00  |....@... ...@. .|
00000060  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000000a0  c6 c6 c4 c4 c4 c0 c0 c4  c4 c4 c4 c4 c0 c0 00 00  |................|
000000b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000c0  00 00 00 00 00 00 00 00  00 1a 22 2a 31 35 01 35  |.........."*15.5|
000000d0  39 40 46 4d 7f 7f 7f 00  00 00 00 00 00 00 00 00  |9@FM............|
000000e0  11 1d 11 1d 1c 35 1c 35  1e 35 1e 35 17 19 17 19  |.....5.5.5.5....|
000000f0  02 00 00 00 98 80 80 88  0a 00 00 00 00 00 00 00  |................|
00000100  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00000120  00 00 00 00 00 00 00 00  00 00 00 00 00 00 77 00  |..............w.|
00000130  11 1d 11 1d 15 7f 15 7f  17 7f 17 7f 10 3b 10 3b  |.............;.;|
00000140  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00000200  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000400  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*

```
3) The eeprom buffer is loaded from the MT7628_EEPROM_20140317.bin file.  MT7628_EEPROM_20140317.bin - default settings.

Nossiac, please see the source code. What could be the problem?

Thank you!

jhw520, 2020-03-27:
Did you solve the problem? I also have this problem...

sanitariu, 2020-03-27:
Mac is at the wrong address. Use latest openwrt i think they solved that.