Nossiac/mtk-openwrt-feeds #86: MT7610e driver for kernel 4.14.54(OpenWrt 18.06.0)

MiltonLai, 2018-08-20:
Hi Nossiac,

I noticed that the ko files hadn't been updated since 9th Jul, as I am working on a customized build on OpenWrt 18.06.0, the driver for kernel 4.14.54 is required, could you please spare some time to release the latest ko files? 

Many thanks,
Milton

bracki19, 2018-08-23:
Also for  4.14.63(18.06.1) would be nice.

antikies, 2018-08-26:
Also for MT7603 (4.14.63 - 18.06.01) would be greet.

nicefile, 2018-09-06:
just use snapshot and get 4.14.66 kernel version of OpenWRT https://git.openwrt.org/?p=openwrt/openwrt.git;a=snapshot;h=5fde931d6071225e4383b33aaaa2131d49f55eee;sf=tgz
I've build it for mi router 3g but mt7612  driver is fails to load. So no 5Ghz radio