Nossiac/mtk-openwrt-feeds #89: Xiaomi R3G (mt7603) kernel panic on ifconfig ra0 up

djwlindenaar, 2018-09-14:
Hi,
I've been trying to get the mt7603 module working on my mir3g setup, but no luck so far. After getting the required files in place, I get a kernel panic when issueing `ifconfig ra0 up`.

Anything I'm missing?

```
[ 8823.357555] mt7603: module license 'MediaTek' taints kernel.
[ 8823.363248] Disabling lock debugging due to kernel taint
[ 8823.406674] register mt7603e
[ 8823.409662] bus=0x1, slot = 0x0, irq=0xff
[ 8823.418495] 
[ 8823.418495] 
[ 8823.418495] === pAd = c0801000, size = 1302152 ===
[ 8823.418495] 
[ 8823.427999] <-- RTMPAllocTxRxRingMemory, Status=0, ErrorValue=0x
[ 8823.434860] <-- RTMPAllocAdapterBlock, Status=0
[ 8823.439383] pAd->CSRBaseAddress =0xc0700000, csr_addr=0xc0700000!
[ 8823.445521] device_id =0x7603
[ 8823.448505] RtmpChipOpsHook(474): Not support for HIF_MT yet!
[ 8823.454275] mt7603_init()-->
[ 8823.457164] mt_bcn_buf_init(224): Not support for HIF_MT yet!
[ 8823.462941] <--mt7603_init()
[ 8834.469872] TX_BCN DESC ae141000 size = 320
[ 8834.474300] RX[0] DESC ae153000 size = 2048
[ 8834.478970] RX[1] DESC ae154000 size = 2048
[ 8834.488346] E2pAccessMode=2
[ 8834.491718] cfg_mode=9
[ 8834.494282] cfg_mode=9
[ 8834.496658] wmode_band_equal(): Band Equal!
[ 8834.502968] APSDCapable[0]=0
[ 8834.505844] APSDCapable[1]=0
[ 8834.508838] APSDCapable[2]=0
[ 8834.511774] APSDCapable[3]=0
[ 8834.514643] APSDCapable[4]=0
[ 8834.517612] APSDCapable[5]=0
[ 8834.520489] APSDCapable[6]=0
[ 8834.523494] APSDCapable[7]=0
[ 8834.526371] APSDCapable[8]=0
[ 8834.529364] APSDCapable[9]=0
[ 8834.532287] APSDCapable[10]=0
[ 8834.535257] APSDCapable[11]=0
[ 8834.538337] APSDCapable[12]=0
[ 8834.541353] APSDCapable[13]=0
[ 8834.544418] APSDCapable[14]=0
[ 8834.547380] APSDCapable[15]=0
[ 8834.550333] default ApCliAPSDCapable[0]=0
[ 8834.670727] Key1Str is Invalid key length(0) or Type(0)
[ 8834.676172] Key2Str is Invalid key length(0) or Type(0)
[ 8834.681704] Key3Str is Invalid key length(0) or Type(0)
[ 8834.687109] Key4Str is Invalid key length(0) or Type(0)
[ 8834.719742] [PMF]Set_PMFMFPC_Proc:: apidx=0, Desired MFPC=0
[ 8834.725455] [PMF]Set_PMFMFPR_Proc:: apidx=0, Desired MFPR=0
[ 8834.731087] [PMF]Set_PMFSHA256_Proc:: apidx=0, Desired PMFSHA256=0
[ 8834.739578] FW Version:
[ 8834.739594] C
[ 8834.742059] L
[ 8834.743624] 1
[ 8834.745187] 7
[ 8834.746836] 3
[ 8834.748416] 7
[ 8834.749999] 6
[ 8834.751614] 6
[ 8834.753194] -
[ 8834.754898] A
[ 8834.756474] 
[ 8834.759519] FW Build Date:
[ 8834.759523] 2
[ 8834.762345] 0
[ 8834.763926] 1
[ 8834.765508] 7
[ 8834.767076] 0
[ 8834.768756] 9
[ 8834.770335] 2
[ 8834.771959] 5
[ 8834.773541] 1
[ 8834.775113] 1
[ 8834.776812] 4
[ 8834.778389] 9
[ 8834.779959] 0
[ 8834.781579] 6
[ 8834.783272] 
[ 8834.784847] 
[ 8834.787930] Kernel bug detected[#1]:
[ 8834.791495] CPU: 2 PID: 1685 Comm: ifconfig Tainted: P                4.14.66 #0
[ 8834.798856] task: 8fcf8c80 task.stack: 8e136000
[ 8834.803360] $ 0   : 00000000 00000001 00000001 00000001
[ 8834.808570] $ 4   : 8d98ef60 00000000 00000f60 00000000
[ 8834.813779] $ 8   : 8048e2d0 81003000 00000000 00000000
[ 8834.818990] $12   : 00000000 805a0000 000bf9cf 00000000
[ 8834.824198] $16   : 8d98ef60 00000000 00000000 00000001
[ 8834.829407] $20   : 8f19d9c0 c0811000 00000000 80630000
[ 8834.834616] $24   : 00000000 8029c878                  
[ 8834.839825] $28   : 8e136000 8e137a40 00000000 8001b894
[ 8834.845035] Hi    : 00000124
[ 8834.847897] Lo    : 74e58000
[ 8834.850789] epc   : 80026c68 r4k_dma_cache_wback_inv+0x1c/0x26c
[ 8834.856699] ra    : 8001b894 mips_dma_map_page+0x194/0x218
[ 8834.862157] Status: 11007c02	KERNEL EXL 
[ 8834.866068] Cause : 50800034 (ExcCode 0d)
[ 8834.870053] PrId  : 0001992f (MIPS 1004Kc)
[ 8834.874125] Modules linked in: mt7603(P) pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
[ 8834.939409] Process ifconfig (pid: 1685, threadinfo=8e136000, task=8fcf8c80, tls=77f2ada8)
[ 8834.947630] Stack : 00000000 806076c8 8060dfa8 00000f60 811b61c0 00000f60 811b61c0 8001b894
[ 8834.955965]         00000001 802a0830 00000030 00000003 00000001 80600000 80600000 80600000
[ 8834.964299]         c093e9f0 c0801000 ae140000 8f0e9900 8f19d9c0 c0811000 c08130ec c0811000
[ 8834.972651]         8f19d9c0 8d8c55a4 00000000 811b61c0 00000f60 1cc282a3 00000001 00000000
[ 8834.980987]         00000000 8d8be36c 80630000 00000000 8e137b0c 8d8f9b9c 0000ff00 00000000
[ 8834.989318]         ...
[ 8834.991758] Call Trace:
[ 8834.994202] [<80026c68>] r4k_dma_cache_wback_inv+0x1c/0x26c
[ 8834.999758] [<8001b894>] mips_dma_map_page+0x194/0x218
[ 8835.005313] [<8d8c55a4>] linux_pci_map_single+0x90/0x9c [mt7603]
[ 8835.011600] [<8d8be36c>] RtmpDrvPciMapSingle+0x4c/0xc4 [mt7603]
[ 8835.017789] [<8d8aa5f0>] AndesMTPciKickOutCmdMsg+0x160/0x2ac [mt7603]
[ 8835.024499] [<8d8a92b8>] AndesSendCmdMsg+0x2ec/0x14c4 [mt7603]
[ 8835.030601] [<8d8a9bc8>] AndesSendCmdMsg+0xbfc/0x14c4 [mt7603]
[ 8835.036554] Code: afbf001c  00a08825  00808025 <00020336> 3c02805a  2442fce0  8c430008  7c630440  10600043 
[ 8835.046282] 
[ 8835.049130] ---[ end trace 9dfef710f95ab33c ]---
[ 8835.055812] Kernel panic - not syncing: Fatal exception
[ 8835.062728] Rebooting in 3 seconds..
```

nicefile, 2018-09-19:
this is working for mt7603 2.4Ghz radio:
dd if=/dev/mtd3 of=/etc/wireless/mt7603/mt7603.eeprom.bin bs=1 count=200
modprobe mt7603
ifconfig ra0 up
but don't get to excited  . 5Ghz radio is useless unless proven wrong  .

djwlindenaar, 2018-09-29:
@nicefile Thanks for the tip, but sadly it did not resolve the issue. 

The exact same oops and call trace. 

Any other ideas?

BTW I notice the chip is called mt7603e in the open source driver. Is the e significant in this?

closed 2019-02-23 by djwlindenaar