Nossiac/mtk-openwrt-feeds #93: mt7603 on a Xiaomi 3g

tkso1997, 2018-10-18:
Is there any news for mt7603 on a Xiaomi 3g? With openwrt drivers the 2.4ghz 7603 is basically not usable. the 5ghz works pretty good! Can we mix both drivers, for instance mtk7603 for 2.4ghz and the opensource mt7012 from openwrt  for 5ghz?

nicefile, 2018-12-09:
Yes we can  . Take look here for my custom 4.14.66 build  with opensource 5Ghz and close source 2.4: https://drive.google.com/drive/folders/1_WRgzpJyFr3GqQvtHelz2i_4nhI9KMyB

tkso1997, 2018-12-10:
Cool! 
Is there anything I need to be prepared when installing it? Can I just do a sysupgrade or do I have to flash the partitions directly?
Are you using it on your router, how is it working?

Thanks 

Thorsten

nicefile, 2018-12-10:
I've did flash from breed to "OpenWRT-v18.06.1-MiR3G-ForBreed_Full.bin" . now included in g drive . Then sysupgrade to my firmware (pass is 1234567890 afair). No intensive testing done with my build but it worked with few devices connected and 2.4 Ghz was fast as it should .

nicefile, 2018-12-10:
https://drive.google.com/drive/folders/10cJSdaFMHwEjphrtaMpVfhQoO1AqksxI  same with breed flashed openwrt

julianocss, 2018-12-11:
Hello @nicefile.

Can you tell me how you mixed 2.4 and 5 ghz drivers?
Another doubt: This repo is a kernel dependent? I cannot clone him and build a new firmware using latest openwrt snapshot?

Thanks.

julianocss, 2018-12-12:
Hello @brainfog0.

I made the first two steps, the firmware was built without errors. 
But i am afraid to flash and brick the router. 

Can you help-me in these steps?
"added script to fetch eeprom from flash, and added module to /etc/modules.d for autobooting."

Thanks. 
Juliano

nicefile, 2018-12-12:
@julianocss 
easy: https://git.openwrt.org/?p=openwrt/openwrt.git;a=shortlog;h=refs/heads/openwrt-18.06;pg=1
look for commit just before update to 4.14.67 and that will be "kernel: bump 4.9 to 4.9.124" click tree next to it then snapshot  to download tgz

julianocss, 2018-12-12:
Thanks @nicefile, it works.

And what is these steps that @brainfog0 said: "added script to fetch eeprom from flash, and added module to /etc/modules.d for autobooting." 
My router is an Archer C50 V3 (MT7628AN + MT7612E). Can you help-me how to mix the drivers?

Juliano

nicefile, 2018-12-12:
@julianocss 
Additional script is not needed .  You need extract 512b eeprom from 'factory' (mtd3 in my case) like 
dd if=/dev/mtd3 of=/etc/wireless/mt7603/mt7603.eeprom.bin bs=1 count=512 .Should work from start . Try build Openwrt with nossiac modules first to see if they work . Then if not  post the issue  in new thread and we will see .

brainfog0, 2018-12-13:
@julianocss
sorry, was no clear enough. i meant adding this command mentioned by nicefile above to uci-defaults, so you don't need to use it manually after installing firmware.
autobooting not needed for tests, i just made it later for convenience. first you want try to load driver manually with "insmod mt7603" command.
if you have doubts about bricking router, try to find usb uart adapter or flash breed u-boot to your router (be careful with flashing u-boot though)

julianocss, 2018-12-13:
Hello @nicefile and @brainfog0.

So, the steps will be:
1) Build a firmware with 4.14.66 kernel without mt76 drivers.
2) Download the file " mt7612-for-mt76x8-linux-4.14.66.ko "  (this is the correct file for my router?).
3) Install manually using isnmod ( the command is only isnmod 'downloaded_file.ko' )?

Thanks.
Juliano

brainfog0, 2018-12-13:
@julianocss
instead of downloading manually would be easier to add feeds from this repository (look readme.md for feeds instructions).
then enable needed modules in menuconfig and enable plugin for luci for easier configuration, and build firmware.
yes, to load module into a kernel, use command insmod 'module_name'

julianocss, 2018-12-13:
Ok, thanks @brainfog0 .

Do you know which modules will be necessary for my router?
Just mt7612? Or mt7612 + mt7628 ?

Regards.
Juliano

brainfog0, 2018-12-13:
you may try both. on xiaomi r3g mt7603 worked fine, but mt7612 crashed router.
maybe on your hardware it will work better

julianocss, 2018-12-13:
@brainfog0 

Both radios (2.4 and 5 Ghz) works with only MT7603 driver?

Juliano

brainfog0, 2018-12-13:
@julianocss 
nope. mt7603 - 2.4 GHz module, mt7612 - 5 GHz module

julianocss, 2018-12-15:
Hello @nicefile and @brainfog0 

I tried to compile the firmware with 4.14.66 snapshot build, but image builder do not completed and gave a lot of erros. I think that could be a problem with feeds, like luci-app-adblock, luci-app-sqm, etc.
Do you have any suggestions? I thought about creating an image of stable version 18.06.1, but the Kernel version is different...

Juliano

julianocss, 2018-12-17:
Hello @nicefile and @brainfog0

I was able to compile the firmare.
The driver did not load automatically. After you manually load it and start the interface, the router restarts.

Do you have any idea what would be the problem?

I use this Openwrt commit:

[https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=e3022727658166e736198529582a46abf2397ea4](url)

Regards,
Juliano

tkso1997, 2019-01-30:
Hello @nicefile 
> https://drive.google.com/drive/folders/10cJSdaFMHwEjphrtaMpVfhQoO1AqksxI same with breed flashed openwrt

Is it working okay for you and could you share it again (link is down)?

One other thing: How did you manage to prepare the firmware that it is flashable with breed, would be very interested in it! What are the commands to turn a openwrt image into a breed compatible file?

Regards
Thorsten

tkso1997, 2019-01-30:
Hella @nicefile @all,

I compiled OpenWrt with kernel-4.14.93 your newest 4.14.93-mt7603 and the open source 5GHz mt76 driver. So far it works quite nice. Any tips on how to make the 2.4GHz a bit more responsive? Throughput is okay but its a bit laggy. Any recommendations are very welcome.

So far the combination of your closed-source 2.4GHz driver plus the open source 5GHz one seems to be the best for Xiaomi 3g. 

If someone wants to try my version feel free:

[https://www.dropbox.com/s/ehbnvj94z5ecb9w/openwrt-ramips-mt7621-mir3g-squashfs-sysupgrade.bin?dl=0](url)

nicefile, 2019-02-08:
Hello @ALL 
kernel 4.14.93 from me  with mt76 from 31.01 and shed_cake 15.01
I've added script  and firmware do flash it under breed :
https://drive.google.com/drive/folders/1P9tsRaSeHpEyqMZBZ0HwNzIhYJgb6OxG?usp=sharing

pataha, 2019-02-12:
@tkso1997 can I flash your firmware through luci upgrade page?

tkso1997, 2019-02-19:
@pataha yes you can

nicefile, 2019-04-12:
@ALL yet another update with mt version 2019-03-27 and 4.14.93
https://drive.google.com/open?id=1eYKT40xQY4m-wt-TaExBR_bL7-7gBdsO
It supposed to have 1100Mhz overclock (patch based ) but cannot confirm that on my device . Just don't see it

tr4usut, 2019-06-03:
Hope you can share build usually when have new build. tks you @nicefile :)

jinglei207, 2020-06-20:
> 
> 
> @ALL yet another update with mt version 2019-03-27 and 4.14.93
> https://drive.google.com/open?id=1eYKT40xQY4m-wt-TaExBR_bL7-7gBdsO
> It supposed to have 1100Mhz overclock (patch based ) but cannot confirm that on my device . Just don't see it

Hi how did you add the 5ghz driver? i saw that it as the interface name `rai0`, but under luci, it shows `ra0`, i can't enable it.
https://github.com/Nossiac/mtk-openwrt-feeds/issues/116

9000000, 2020-07-08:
> @ALL yet another update with mt version 2019-03-27 and 4.14.93
> https://drive.google.com/open?id=1eYKT40xQY4m-wt-TaExBR_bL7-7gBdsO
> It supposed to have 1100Mhz overclock (patch based ) but cannot confirm that on my device . Just don't see it

I cannot download it

nicefile, 2020-07-08:
> > @ALL yet another update with mt version 2019-03-27 and 4.14.93
> > https://drive.google.com/open?id=1eYKT40xQY4m-wt-TaExBR_bL7-7gBdsO
> > It supposed to have 1100Mhz overclock (patch based ) but cannot confirm that on my device . Just don't see it
> 
> I cannot download it

Sorry. I've trashed this project  .I'ts possible to build it manually but that version will be outdated .

9000000, 2020-07-08:
> Sorry. I've trashed this project .I'ts possible to build it manually but that version will be outdated .

Thank you I have found some similar new projects but there is so much unnecessary software for me, I just need to be simple.
https://github.com/coolsnowwolf/lede
https://github.com/Lienol/openwrt