Nossiac/mtk-openwrt-feeds #94: mt7628-for-mt7628-linux-4.9.87.ko is missing mt7628-for-mt7628-linux-4.9.87.ko

dibg, 2018-10-18:
Hello and thanks for this nice initiative. The open source driver for the mt7628 is unstable so i resorted as a last hope to your repository for getting a usable device once again. I try to build the latest openwrt with kernel 4.9.87 for the xiaomi nano which uses the mt7628 chip and i noticed the corresponded driver is missing for this combination mt7628-for-mt7628-linux-4.9.87.ko. I tried the 4.9.* but as soon i load this the system reboots. Care if you provide us the mt7628-for-mt7628-linux-4.9.87.ko ? :)

dibg, 2018-10-18:
I considered changing the kernel to a version that matches your drivers yet non of my option are compatible :/ 
This is the kernel that are available to build openwrt with:  
LINUX_VERSION-3.18 = .71
LINUX_VERSION-4.14 = .29
LINUX_VERSION-4.4 = .121
LINUX_VERSION-4.9 = .87

And this is the binaries you provide in your repository:
4.4.(108,  111,  112,  116,  120,  124,  126,  132,  151, 50, 61, 71, 89, 92)
4.9.*

The 4.4.120 was close yet not compatible.
any change providing the 4.4.121 or 4.9.87 ?

nicefile, 2018-10-21:
download this snapshot 18.06 (last before kernel bump)  https://git.openwrt.org/?p=openwrt/openwrt.git;a=snapshot;h=5fde931d6071225e4383b33aaaa2131d49f55eee;sf=tgz
then build it with 4.14.66 kernel .

dibg, 2018-10-21:
Im not sure how this help :? The modules that already available is for this kernels: 
4.4.(108, 111, 112, 116, 120, 124, 126, 132, 151, 50, 61, 71, 89, 92)
4.9.*
4.14.66 is not one of them. Can you elaborate how this will help ?

nicefile, 2018-10-22:
here it is ... http://nossiac.com/download/mtk-wifi-ko/mt7603-for-mt7620-linux-4.14.66.ko

dibg, 2018-10-22:
Actually you did help. It seems the driver i needed was the mt7628-for-mt76x8-linux-*.ko and not the mt7628-for-mt7628-linux-*.ko. I build based on the suggested snapshot since nossiac had the specific binary for the 4.14.66 and as we speak i have a working router.

The latest comment about the driver doesn't help further. The xiaomi nano have the mt7628an cpu not the mt7620.

dibg, 2018-10-30:
finally i moved to padavan firmware since the interface with the firmware on openwrt it was buggy. For everyone looking for a stable firmware for the xiaomi nano i recommend padavans firmware. I tested it for 24 hours with iperf3 and it was fully stable.

closed 2018-10-30 by dibg