Nossiac/mtk-openwrt-feeds #96: Anyone makes the dirver works?

szfrankhan, 2018-10-25:
Hi all,
My router has MT7621+7603e+7612e. I am using LEDE17.01. I tried Openwrt's mt76. It does not work well. I think maybe MTK's driver is better. It seems there are still some problems:

1, in MTK luci-plugin, there is only ra0. Both mt7612 and mt760e are under ra0. I can not find rai0. When I enable mt7612 (or mt7603), both wifi interfaces are enabled/disabled;
2, my cell and PC can find wifi-mt7603 and the signal is strong. But when I tried to connect with it, the cell is stuck in 'trying to get IP address'

So is there anyone make the driver work and what is her/his platform?

Thanks!
Frank

dibg, 2018-10-30:
Try padavans firmware if your device supported. Prometheus project added extra support for other devices like xiaomi mi3g and works really well. The only thing im missing is the flexibility of openwrt, installing extra packages and customization.

jinglei207, 2020-06-20:
> Hi all,
> My router has MT7621+7603e+7612e. I am using LEDE17.01. I tried Openwrt's mt76. It does not work well. I think maybe MTK's driver is better. It seems there are still some problems:
>
> 1, in [...]

did you solve the problem? i have a router with same hardware.