Nossiac/mtk-openwrt-feeds #104: Drivers for 4.14.95 (OpenWrt 18.06.2)

michallipka, 2019-03-31:
Current drivers are compiled for kernel 4.14.93 which was current at that time(Jan 22nd).  According to changelog on Jan 28th Koen Vandeputte did kernel: bump 4.14 to 4.14.95 so it is now current kernel for this stable release. Could you compile drivers for this (there shouldn't be serious differences and problems as this is not big version change)

redchenjs, 2019-03-31:
As far as I know, the 4.14.93 version works well on the 4.14.95 kernel without any problems.

michallipka, 2019-03-31:
@redchenjs thanks for info. I was trying to start them . I unloaded mt76,mt76x2_common,mt76x02_lib,mt7603e and after modprobe mt7603I got two lines in log:
 kern.warn kernel: [  750.962412] mt7603: module license 'MediaTek' taints kernel.
kern.warn kernel: [  750.968098] Disabling lock debugging due to kernel taint
so I assume it loaded. But I can't bring interface up: # ifconfig ra0 up
ifconfig: SIOCGIFFLAGS: No such device
(the same with wlan0). Any clues?

redchenjs, 2019-03-31:
Before loading the driver, be sure to put the configuration file in the correct folder:

> /etc/wireless/mt7603/mt7603.dat

This configuration file can be generated by [uci2dat](https://github.com/Nossiac/mtk-openwrt-feeds/tree/master/applications/uci2dat/src)
After that you can run the script below:
```
ifconfig wlan0 down
rmmod mt7603e
insmod ./mt7603-for-mt7621-linux-4.14.93.ko
ifconfig ra0 up
```

michallipka, 2019-03-31:
I wanted to test it before as I have troubles compiling uci2dat.
I installed gcc package and downloaded uci.h (and put in /usr/include) and created empty /usr/include/uci_config.h (I couldn't find a package with those files).
And while compiling I am getting errors:
# gcc -o uci2dat uci2dat.c 
/tmp/ccPkNikj.o: In function `uci_lookup_package':
uci2dat.c:(.text+0x40): undefined reference to `uci_lookup_next'
uci2dat.c:(.text+0x4c): undefined reference to `uci_lookup_next'
/tmp/ccPkNikj.o: In function `uci_lookup_option':
uci2dat.c:(.text+0xd4): undefined reference to `uci_lookup_next'
uci2dat.c:(.text+0xe0): undefined reference to `uci_lookup_next'
/tmp/ccPkNikj.o: In function `main':
uci2dat.c:(.text+0xcfd0): undefined reference to `uci_alloc_context'
uci2dat.c:(.text+0xcfdc): undefined reference to `uci_alloc_context'
uci2dat.c:(.text+0xd058): undefined reference to `uci_unload'
uci2dat.c:(.text+0xd064): undefined reference to `uci_unload'
uci2dat.c:(.text+0xd08c): undefined reference to `uci_load'
uci2dat.c:(.text+0xd098): undefined reference to `uci_load'
uci2dat.c:(.text+0xd3d4): undefined reference to `uci_unload'
uci2dat.c:(.text+0xd3e0): undefined reference to `uci_unload'
collect2: error: ld returned 1 exit status

I suppose I am missing some libraries but I have no idea how to track them

redchenjs, 2019-03-31:
I updated my comment, you can try the new example file.

michallipka, 2019-03-31:
Thank you for this manual end example config file - I will write it myself instead of using uci2dat.
module loads:
kern.warn kernel: [  180.349505] mt7603: module license 'MediaTek' taints kernel.
kern.warn kernel: [  180.355234] Disabling lock debugging due to kernel taint
kern.warn kernel: [  180.394928] register mt7603e
kern.warn kernel: [  180.397924] bus=0x1, slot = 0x0, irq=0x16
kern.warn kernel: [  180.406999]
kern.warn kernel: [  180.406999]
 kern.warn kernel: [  180.406999] === pAd = c0b81000, size = 1302152 ===
kern.warn kernel: [  180.406999]
kern.warn kernel: [  180.416692] <-- RTMPAllocTxRxRingMemory, Status=0, ErrorValue=0x
kern.warn kernel: [  180.423707] <-- RTMPAllocAdapterBlock, Status=0
kern.warn kernel: [  180.428263] pAd->CSRBaseAddress =0xc0a80000, csr_addr=0xc0a80000!
kern.warn kernel: [  180.434446] device_id =0x7603
kern.warn kernel: [  180.437441] RtmpChipOpsHook(474): Not support for HIF_MT yet!
kern.warn kernel: [  180.443276] mt7603_init()-->
kern.warn kernel: [  180.446200] mt_bcn_buf_init(224): Not support for HIF_MT yet!
kern.warn kernel: [  180.452025] <--mt7603_init()

lsmod|grep mt7
cfg80211              216416  3 mt76x02_lib,mt76,mac80211
mac80211              425472  3 mt76x2e,mt76x02_lib,mt76
mt76                   23776  3 mt76x2e,mt76x2_common,mt76x02_lib
**mt7603               1435472  0 [permanent]**
mt76x02_lib            34816  2 mt76x2e,mt76x2_common
mt76x2_common          10624  1 mt76x2e
mt76x2e                10080  0 

but router reboots when trying to bring interface ra0 up.
logs (removed timestamps and kern.warn kernel)
TX_BCN DESC adc3d000 size = 320
RX[0] DESC adc3f000 size = 2048
RX[1] DESC adc88000 size = 2048
cfg_mode=9
cfg_mode=9
wmode_band_equal(): Band Equal!
APSDCapable[0]=0
APSDCapable[1]=0
APSDCapable[2]=0
APSDCapable[3]=0
APSDCapable[4]=0
APSDCapable[5]=0
APSDCapable[6]=0
APSDCapable[7]=0
APSDCapable[8]=0
APSDCapable[9]=0
APSDCapable[10]=0
APSDCapable[11]=0
APSDCapable[12]=0
APSDCapable[13]=0
APSDCapable[14]=0
APSDCapable[15]=0
default ApCliAPSDCapable[0]=0
Key1Str is Invalid key length(1) or Type(0)
Key2Str is Invalid key length(1) or Type(0)
Key3Str is Invalid key length(1) or Type(0)
Key4Str is Invalid key length(1) or Type(0)
RtmpOSFileOpen(): Error 2 opening /etc/Wireless/RT2860AP/SingleSKU.dat
FW Version:
C
L
1
7
3
7
6
6
-
A

FW Build Date:
2
0
1
7
0
9
2
5
1
1
4
9
0
6

I suspect some errors in configuration (it uses the example you sent) and will try to configure it myself.
If you have any remarks/questions please let me know :)

redchenjs, 2019-04-01:
I'm using the OpenWrt git version and the mt7603 driver can work without any problems, maybe
you should build a new firmware using the latest OpenWrt git.

michallipka, 2019-04-01:
I flashed last OpenWrt snapshot (OpenWrt SNAPSHOT r9760-6027e49093 ) and still the same issues (with mt7603). I tried to use mt7612-for-mt7621-linux-4.14.93.ko for 5GHz wifi (wlan1, originally mt76x2e module) and it behaves a little better. I can insmod it, then ifconfig rai0 up. Unfortunately wifi doesn't show and if I do ifconfig rai0 down then router immediately reset
```
# insmod mt7612-for-mt7621-linux-4.14.93.ko
(logs)
 mt7612: module license 'MediaTek' taints kernel.
 Disabling lock debugging due to kernel taint
 register rt2860
 bus=0x2, slot = 0x1, irq=0x17

 === pAd = c0b81000, size = 2108512 ===

 <-- RTMPAllocTxRxRingMemory, Status=0
 <-- RTMPAllocAdapterBlock, Status=0
 pAd->CSRBaseAddress =0xc0a80000, csr_addr=0xc0a80000!
 device_id =0x7662
 ==>rlt_wlan_chip_onoff(): OnOff:1, Reset= 1, pAd->WlanFunCtrl:0x0, Reg-WlanFunCtrl=0x20a
 chip_id1=, chip_id2=, pAd->MACVersion=0x76623000
 chip_id1=0x0, chip_id2=0x0, pAd->MACVersion=0x76623000
 get_dev_name_prefix(): dev_idx = 1, dev_name_prefix=rai
```
```
# ifconfig rai0
rai0      Link encap:Ethernet  HWaddr 00:00:00:00:00:00  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:23
```
```
# ifconfig rai0 up
(logs)
 cfg_mode=9
 cfg_mode=9
 wmode_band_equal(): Band Equal!
 APSDCapable[0]=0
 APSDCapable[1]=0
 APSDCapable[2]=0
 APSDCapable[3]=0
 APSDCapable[4]=0
 APSDCapable[5]=0
 APSDCapable[6]=0
 APSDCapable[7]=0
 APSDCapable[8]=0
 APSDCapable[9]=0
 APSDCapable[10]=0
 APSDCapable[11]=0
 APSDCapable[12]=0
 APSDCapable[13]=0
 APSDCapable[14]=0
 APSDCapable[15]=0
 default ApCliAPSDCapable[0]=0
 Key1Str is Invalid key length(0) or Type(0)
 Key2Str is Invalid key length(0) or Type(0)
 Key3Str is Invalid key length(0) or Type(0)
 Key4Str is Invalid key length(0) or Type(0)
 RtmpChipOpsEepromHook::e2p_type=0, inf_Type=5
 RtmpEepromGetDefault::e2p_dafault=1
 NVM is EFUSE mode
 build time =
 2
 0
 1
 4
 1
 1
 1
 5
 0
 6
 0
 6
 0
 6
 a

 rom patch for E3 IC

 platform =
 A
 L
 P
 S

 hw/sw version =
 �
 �
 patch version = 
```
```
# ifconfig rai0
rai0      Link encap:Ethernet  HWaddr 00:00:00:00:00:00  
          inet6 addr: fe80::200:ff:fe00:0/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:23 
```

Any thoughts?

redchenjs, 2019-04-01:
Which device are you using? I built the firmware image from the OpenWrt git source. If you want to try, I can build a firmware image for your device.

michallipka, 2019-04-02:
sure. Why not? Thank you
I am using Xiaomi Mi Router 3G (MediaTek MT7621 ver:1 eco:3)

redchenjs, 2019-04-03:
The OpenWrt git version firmware image:
[openwrt-ramips-mt7621-xiaomi_mir3g-squashfs-sysupgrade.bin.zip](https://github.com/Nossiac/mtk-openwrt-feeds/files/3037421/openwrt-ramips-mt7621-xiaomi_mir3g-squashfs-sysupgrade.bin.zip)
For 5G wifi, you need to export the EEPROM before unloading the mt76 driver, run this command:
```
cat /sys/kernel/debug/ieee80211/phy1/mt76/eeprom > /etc/wireless/mt7612/mt7612e.eeprom.bin
```
And also you need a config file:
```
/etc/wireless/mt7612/mt7612.dat
```
Example: [mt7612.dat.zip](https://github.com/Nossiac/mtk-openwrt-feeds/files/3037440/mt7612.dat.zip)

michallipka, 2019-04-03:
Thank you
I am able to run 5G wifi on mt7612 after flashing with firmware you provided (Maybe this eeprom helped). Everything works (interface up down, bridge, I was even able to set encryption using iwpriv rai0 set .... (from documentation). My download speed using speedtest was 210Mbps (near router) so quite good.

Something strange is with mt7603. I can insmod it but nothing goes to the logs. ifconfig -a show interface ra0 but I can't bring it up (and still nothing in the logs):
```
# ifconfig ra0 up
ifconfig: SIOCSIFFLAGS: Operation not permitted
```
I was doing this while running on mt7612 if it's relevant.

Do you have any idea why can't I start ra0 interface?

redchenjs, 2019-04-04:
After flashing the new firmware, you need to recreate the config file because the "/etc/wireless" folder will be automatically deleted, check to see if this file exists.
```
/etc/wireless/mt7603/mt7603.dat
```
Example: [mt7603.dat.zip](https://github.com/Nossiac/mtk-openwrt-feeds/files/3041842/mt7603.dat.zip)

michallipka, 2019-04-04:
I know. I have /etc/wireless in sysupgrade.conf so it is preserved
```
/etc/wireless/mt7603# ls -l
-rw-r--r--    1 root     root       2078044 Apr  1 22:03 mt7603-for-mt7621-linux-4.14.93.ko
-rw-r--r--    1 root     root          1742 Apr  1 13:25 mt7603.dat
```

michallipka, 2019-04-04:
I rebooted the router (and therefore got back to open driver with 5Ghz) and I tried with mt7603 (for 2,4Ghz).
insmod went ok, logs normal:
```
Thu Apr  4 09:47:25 2019 kern.warn kernel: [ 2200.602156] mt7603: module license 'MediaTek' taints kernel.
Thu Apr  4 09:47:25 2019 kern.warn kernel: [ 2200.607857] Disabling lock debugging due to kernel taint
Thu Apr  4 09:47:25 2019 kern.warn kernel: [ 2200.655386] register mt7603e
Thu Apr  4 09:47:25 2019 kern.warn kernel: [ 2200.658431] bus=0x1, slot = 0x0, irq=0x16
Thu Apr  4 09:47:25 2019 kern.warn kernel: [ 2200.667074]
Thu Apr  4 09:47:25 2019 kern.warn kernel: [ 2200.667074]
Thu Apr  4 09:47:25 2019 kern.warn kernel: [ 2200.667074] === pAd = c0b01000, size = 1302152 ===
Thu Apr  4 09:47:25 2019 kern.warn kernel: [ 2200.667074]
Thu Apr  4 09:47:25 2019 kern.warn kernel: [ 2200.676518] <-- RTMPAllocTxRxRingMemory, Status=0, ErrorValue=0x
Thu Apr  4 09:47:25 2019 kern.warn kernel: [ 2200.683350] <-- RTMPAllocAdapterBlock, Status=0
Thu Apr  4 09:47:25 2019 kern.warn kernel: [ 2200.687944] pAd->CSRBaseAddress =0xc0a00000, csr_addr=0xc0a00000!
Thu Apr  4 09:47:25 2019 kern.warn kernel: [ 2200.694082] device_id =0x7603
Thu Apr  4 09:47:26 2019 kern.warn kernel: [ 2200.697171] RtmpChipOpsHook(474): Not support for HIF_MT yet!
Thu Apr  4 09:47:26 2019 kern.warn kernel: [ 2200.702925] mt7603_init()-->
Thu Apr  4 09:47:26 2019 kern.warn kernel: [ 2200.705866] mt_bcn_buf_init(224): Not support for HIF_MT yet!
Thu Apr  4 09:47:26 2019 kern.warn kernel: [ 2200.711637] <--mt7603_init()
```
unfortunately I can't bring the interface up (file /etc/wireless/mt7603/mt7603.dat) is present. At least there is no reboot nor kernel panic.
```
# ifconfig ra0 up
ifconfig: SIOCSIFFLAGS: Operation not permitted
```
logs
```
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.548098] TX_BCN DESC ad774000 size = 320
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.552328] RX[0] DESC ad776000 size = 2048
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.557002] RX[1] DESC ad777000 size = 2048
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.567328] cfg_mode=9
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.569700] cfg_mode=9
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.572055] wmode_band_equal(): Band Equal!
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.577835] APSDCapable[0]=0
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.580729] APSDCapable[1]=0
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.583594] APSDCapable[2]=0
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.586502] APSDCapable[3]=0
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.589389] APSDCapable[4]=0
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.592264] APSDCapable[5]=0
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.595167] APSDCapable[6]=0
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.598059] APSDCapable[7]=0
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.600934] APSDCapable[8]=0
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.603797] APSDCapable[9]=0
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.606707] APSDCapable[10]=0
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.609683] APSDCapable[11]=0
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.612643] APSDCapable[12]=0
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.615639] APSDCapable[13]=0
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.618613] APSDCapable[14]=0
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.621570] APSDCapable[15]=0
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.624558] default ApCliAPSDCapable[0]=0
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.630268] Key1Str is Invalid key length(0) or Type(0)
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.635854] Key2Str is Invalid key length(0) or Type(0)
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.641316] Key3Str is Invalid key length(0) or Type(0)
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.646785] Key4Str is Invalid key length(0) or Type(0)
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.671948] RtmpOSFileOpen(): Error 2 opening /etc/Wireless/RT2860AP/SingleSKU.dat
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.679602] FW Version:
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.679609] C
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.682040] L
Thu Apr  4 09:47:41 2019 kern.warn kernel: [ 2216.683604] 1
Thu Apr  4 09:47:42 2019 kern.warn kernel: [ 2216.685195] 7
Thu Apr  4 09:47:42 2019 kern.warn kernel: [ 2216.686788] 3
Thu Apr  4 09:47:42 2019 kern.warn kernel: [ 2216.688386] 7
Thu Apr  4 09:47:42 2019 kern.warn kernel: [ 2216.689958] 6
Thu Apr  4 09:47:42 2019 kern.warn kernel: [ 2216.691525] 6
Thu Apr  4 09:47:42 2019 kern.warn kernel: [ 2216.693092] -
Thu Apr  4 09:47:42 2019 kern.warn kernel: [ 2216.694694] A
Thu Apr  4 09:47:42 2019 kern.warn kernel: [ 2216.696284]
Thu Apr  4 09:47:42 2019 kern.warn kernel: [ 2216.699363] FW Build Date:
Thu Apr  4 09:47:42 2019 kern.warn kernel: [ 2216.699375] 2
Thu Apr  4 09:47:42 2019 kern.warn kernel: [ 2216.702065] 0
Thu Apr  4 09:47:42 2019 kern.warn kernel: [ 2216.703629] 1
Thu Apr  4 09:47:42 2019 kern.warn kernel: [ 2216.705236] 7
Thu Apr  4 09:47:42 2019 kern.warn kernel: [ 2216.706828] 0
Thu Apr  4 09:47:42 2019 kern.warn kernel: [ 2216.708428] 9
Thu Apr  4 09:47:42 2019 kern.warn kernel: [ 2216.710001] 2
Thu Apr  4 09:47:42 2019 kern.warn kernel: [ 2216.711566] 5
Thu Apr  4 09:47:42 2019 kern.warn kernel: [ 2216.713129] 1
Thu Apr  4 09:47:42 2019 kern.warn kernel: [ 2216.714730] 1
Thu Apr  4 09:47:42 2019 kern.warn kernel: [ 2216.716324] 4
Thu Apr  4 09:47:42 2019 kern.warn kernel: [ 2216.717925] 9
Thu Apr  4 09:47:42 2019 kern.warn kernel: [ 2216.719500] 0
Thu Apr  4 09:47:42 2019 kern.warn kernel: [ 2216.721066] 6
```
isn't this relevant:
``
RtmpOSFileOpen(): Error 2 opening /etc/Wireless/RT2860AP/SingleSKU.dat
``
?
And actually what did you do differently than openwrt when building image (so those drivers behave differently)? I would try to build image myself

redchenjs, 2019-04-04:
This is my OpenWrt build config file: [openwrt.config.zip](https://github.com/Nossiac/mtk-openwrt-feeds/files/3042531/openwrt.config.zip)

I don't know the difference between my config and the OpenWrt's one, I just select the model and a few packages, then build it.

The file "/etc/Wireless/RT2860AP/SingleSKU.dat" isn't relevant to this problem.
Maybe on your device it is also necessary to export the 2.4G wifi EEPROM:
```
cat /sys/kernel/debug/ieee80211/phy0/mt76/eeprom > /etc/wireless/mt7603/mt7603.eeprom.bin
```
And my WPA2-PSK config file, it works on my device (Password: 12345678) : 
[mt7603.dat.zip](https://github.com/Nossiac/mtk-openwrt-feeds/files/3042690/mt7603.dat.zip)

michallipka, 2019-04-04:
it looks like the eeprom was the key - everything works now. thank you
I will try to start this on stable release just to check if it works

nicefile, 2019-04-04:
Please check my build on mir3g . Also include additional packages like uci2dat and luci plugin . So far only 2.4Ghz can use close source driver .But my internal testing  show  5Ghz open driver is fast and stable
https://github.com/Nossiac/mtk-openwrt-feeds/issues/93#issuecomment-461721589

redchenjs, 2019-04-05:
@nicefile The MT7603 2.4G wifi is fast and stable now with latest mt76 driver, just like MT7612 5G wifi, you can try the latest version of OpenWrt.

closed 2019-04-06 by michallipka