Nossiac/mtk-openwrt-feeds #105: Wrong interface in mtk-luci-plugin

michallipka, 2019-04-06:
I have mt7612 and mt7603 loaded
mtk-luci-plugin show all interfaces as ra0 which is ok for mt7603, but mt7612 creates interface rai0.
I can configure rai0 using console and it works.
I don't see option to configure this in luci. Is there a workaround for this?
[image]

nicefile, 2019-04-12:
Is that You from OpenWrt forum ? Musimy pogadać  w sprawie https://github.com/Nossiac/mtk-openwrt-feeds/issues/93

jinglei207, 2020-06-20:
did you solve the problem?

michallipka, 2020-06-21:
I configured wifi using console and not luci (as stated in original comment)